espressif/esp-idf #11320: The ESP-IDF fails to verify correctly root certificate on HTTP connect

RilabsAutomotive, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-rc1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

EPS32S3 custom board

### Power Supply used.

USB

### What is the expected behavior?

The embedded SW app is configured as basic authentication.
Whenever , the user SW application is starting a new secured connection to a file server, the ESP-IDF has to check all of the root certificates that the HTTP server provides and compare them to the preconfigured root certificate in the client (embedded SW app).
Whenever one of the provided by the server root certificate matches the configured in the client, then the certificate verification shall pass.

### What is the actual behavior?

Instead of passing the root certificate check, it fails and this way it is not possible to configure the root certificate for the HTTP client correctly

### Steps to reproduce.

**Preconditions**
The embedded SW app is configured as basic authentication.

**Scenario 1**
1. Step : start the embedded SW application that uses a "**minor**" (lower level) root certificate
2. Step : The embedded SW app boots, connects to a MQTT server(TLS) over WiFi and subscribes to a topic
3. Step : The embedded SW app receives a notification over MQTT to connect to a HTTP file server and get one file (for SW update OTA)
4. Step : The embedded SW app connects to the HTTP file server securely (TLS) - NOK, certificate verification fails

**Scenario 1**
1. Step : start the embedded SW application that uses a "**major**" (highest level) root certificate
2. Step : The embedded SW app boots, connects to a MQTT server(TLS) over WiFi and subscribes to a topic
3. Step : The embedded SW app receives a notification over MQTT to connect to a HTTP file server and get one file (for SW update OTA)
4. Step : The embedded SW app connects to the HTTP file server securely (TLS) - OK, certificate verification passes

[image]

### Debug Logs.

```plain
I (09:07:59.312) OTA: HTTP_EVENT_ERROR
I (09:07:59.317) OTA: HTTP_EVENT_DISCONNECTED
I (09:07:59.322) OTA: HTTP_EVENT_DISCONNECTED
E (09:07:59.327) OTA: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (09:08:14.533) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (09:08:14.533) esp-tls-mbedtls: Failed to verify peer certificate!
E (09:08:14.537) esp-tls: Failed to open new connection
E (09:08:14.542) transport_base: Failed to open a new connection
E (09:08:14.550) HTTP_CLIENT: Connection failed, sock < 0
I (09:08:14.555) OTA: HTTP_EVENT_ERROR
I (09:08:14.559) OTA: HTTP_EVENT_DISCONNECTED
I (09:08:14.564) OTA: HTTP_EVENT_DISCONNECTED
E (09:08:14.569) OTA: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (09:08:29.862) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (09:08:29.863) esp-tls-mbedtls: Failed to verify peer certificate!
E (09:08:29.866) esp-tls: Failed to open new connection
E (09:08:29.872) transport_base: Failed to open a new connection
E (09:08:29.880) HTTP_CLIENT: Connection failed, sock < 0
```

RilabsAutomotive, 2023-05-03:
Here is the successful connection to the HTTP server when using the major root certificate
[image]

laukik-hase, 2023-05-08:
Hello, @RilabsAutomotive!

Thank you for sending the issue report. I was investigating this issue but unfortunately could not reproduce it with the `https_request` example on the ESP-IDF `v5.0-rc1` tag.

I was able to connect to [How's My SSL](https://www.howsmyssl.com) with the Root CA as well as the Intermediate Root certificates.

Could you please attach the following -

- A minimal example that reproduces the issue
- The `sdkconfig` file for your example
- Debug logs (with `CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y`)

RilabsAutomotive, 2023-05-10:
Hello @laukik-hase , 

here are the logs with "https_request" sample. I can't upload it here, because it is too big

Thanks

laukik-hase, 2023-05-12:
Hello, @RilabsAutomotive!

- Could you please share the endpoint you are using?
- Is the _minor_ certificate you are referring to the leaf certificate in the certificate chain? I am not sure whether this would work as the device would verify the certificate sent by the server with the public key in its certificate - the intermediate or root CA would be needed for this.

cc: @mahavirj

RilabsAutomotive, 2023-05-12:
[https_request.zip](https://github.com/espressif/esp-idf/files/11464849/https_request.zip)
Hello, I managed to strip some of the unneeded files from the sample app and now I upload it. I think all of the answers are inside.

laukik-hase, 2023-05-15:
Hello, @RilabsAutomotive!

> Is the minor certificate you are referring to the leaf certificate in the certificate chain? This may not work as the device would verify the certificate sent by the server with the public key in its certificate - the intermediate or root CA would be needed for this.

In your example, you are using the leaf certificate for server authentication. This will not work as I mentioned earlier. You need to use the intermediate or the root CA certificate for authentication.

You can use the command `openssl s_client -showcerts -connect nfz-nexus-dev.mykrone.green:443 </dev/null` for getting the certificate chain. From the sample output, the `depth 2` and `depth 1` certificates will work.

```log
depth=2 OU = Root CA, O = ABC, CN = ABC
verify return:1
depth=1 C = BE, O = Intermediate CA, CN = XYZ
verify return:1
depth=0 C = DE, ST = P, L = Q, O = BR, CN = *.abc.com
verify return:1
```

RilabsAutomotive, 2023-05-16:
Thank you @laukik-hase for the explanation and your support. Now it is clear for us. We will use the Root certificate (major).

closed 2023-05-19 by laukik-hase

AxelLin, 2023-08-22:
> Hello, @RilabsAutomotive!
>
>
> In your example, you are using the leaf certificate for server authentication. This will not work as I mentioned earlier. You need to use the intermediate or the root CA [...]

Hi @laukik-hase @mahavirj 
Just a question:
If there are multiple intermediate CA, setting any "intermediate CA" in cert_pem would work?
Or it needs to load all "intermediate CA" as a CA bundle?

Regards,
Axel

mahavirj, 2023-08-23:
>  If there are multiple intermediate CA, setting any "intermediate CA" in cert_pem would work?

Yes.

The certificate chain would be verified till the intermediate CA (provided match has been found) that has been compiled in the firmware. However, if for some reason the intermediate CA gets compromised then it can become trickier to replace the cert in the firmware (it being the only one trusted).

Hence, we recommend using the cert bundle feature, first the probability of root cert becoming compromised is significantly low, and second being that device in the worst case scenario can have some fall back option to reach to different remote endpoint (if required).