espressif/esp-idf #11321: Problem in Encrypt Image

hdpklm, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2-dev-24-gf404fe96b1-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-wroom-1u

### Power Supply used.

USB

### What is the expected behavior?

I try to encrypt Image App to download in the esp32-s3

my problem is After generate the private key "secure_boot_signing_key.pem" and run the esp32-s3 for the first time, every thing work ok.

I can upload any other images using the command --encrypt in the esptool.py write_flash

but I want to encrypt the image before upload in the esp32-s3 for download in the esp32-s3 via ota or any other method like uart or usb(I´m in Dev mode)

I find in this document "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#manual-encryption" how to encrypt the image but it use the key generate randomly "my_flash_encryption_key.bin", and its not dependly from the privatekey and I have the provate key burned in the efuse so this method its not what I need.

any other solution please.

I think aditional to the documentation, you have to add a section with sample example like howto.

thank you 

### What is the actual behavior?

I cant encrypt the image

### Steps to reproduce.

nothing

AdityaHPatwardhan, 2023-05-05:
Hi @hdpklm 
First of all, secure_boot_signing_key.pem is related to the [Secure Boot](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html) feature and the problem that you are facing seems to be related to [Flash Encryption](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#manual-encryption) feature.

In case of flash encryption, Please note that in order encrypt the image outside of the chip, you need to know the plaintext flash encryption key. By default when flash encryption is enabled through esp-idf firmware bootloader, then a random flash encryption key is generated and burned in the eFuse, this key is not made accessible to anyone. The only way to use the key is to use the `--encrypt` option as you have mentioned. This key is different from the key used for secure boot ( secure_boot_signing_key.pem)

If you want to the ability to encrypt the binary outside of the chip ( on host machine) then the private key needs to be burned externally. Please follow the steps mentioned in [Using Host generated key](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#manual-encryption) so that the key generated outside shall be burned in the eFuse and bootloader shall not generate a random key. Then you can follow the steps given in [Manual encrypting files](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#manual-encryption) in order to encrypt the files outside before flashing.

As far as the current module that you have. I think it has been already burned with random flash encryption key so there is no other way than to use `--encrypt` option. You may test the flash encryption and security features on [esp32 qemu emulator](https://github.com/espressif/qemu/wiki) as well.

Please let me know if you want be to elaborate on any point.
Thanks,
Aditya

hdpklm, 2023-05-05:
hello @AdityaHPatwardhan 

so, secure_boot_signing_key.pem is a private key RSA, and its use for secure boot. and the encrypt key its not related with the private key.

and every time I upload a plantext image, the esp32 its encrypt the image before write. but the esptool send the image as plantext.

my question is, I have 2 blocks burned in my esp32, block4 and block5.
BLOCK_KEY0 (BLOCK4)
  Purpose: SECURE_BOOT_DIGEST0
  Encryption key0 or user data
   = a1 03 ac 09 d6 11 fe 89 e0 18 88 30 29 87 2a a7 d4 f3 e6 4d 72 54 3d 18 71 aa 67 f7 d0 c3 06 e5 R/-
BLOCK_KEY1 (BLOCK5)
  Purpose: XTS_AES_128_KEY
    Encryption key1 or user data
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-

1- SECURE_BOOT_DIGEST0 is the public key from secure_boot_signing_key.pem, and XTS_AES_128_KEY is a random number generated on the first reboot after the first flash.

2- and I don't have access to this key, so if I want to post the image to a website, anyone can unmount and read the entire content of the image. this means it is not safe to hide my private data

3- and the other, if I use my own library, then I write to the partition, I have to encrypt the data before writing, or esp_partition_write automatically encrypts the data, and I have to add additional parameters or flags.

4- I think I can rewrite another XTS_AES_128_KEY in another block and change any register to use the last block instead of block5?

5- is the XTS_AES_128_KEY safe to post the image online or can anyone read the data with a brute force attack ???

AdityaHPatwardhan, 2023-05-05:
Hi @hdpklm Please find my answers below

> my question is, I have 2 blocks burned in my esp32, block4 and block5. BLOCK_KEY0 (BLOCK4) Purpose: SECURE_BOOT_DIGEST0 Encryption key0 or user data = a1 03 ac 09 d6 11 fe 89 e0 18 88 30 29 87 2a a7 [...]

SECURE_BOOT_DIGEST0 is the digest of the public key generated from secure_boot_sigining_key.pem ( private RSA key) It is related to secure boot.
XTS_AES_128_KEY is a random number generated ( Related to Flash Encryption)
 
> 2- and I don't have access to this key, so if I want to post the image to a website, anyone can unmount and read the entire content of the image. this means it is not safe to hide my private data
What do you mean by post image to a website? 

If you have enabled flash encryption for the chip then the partitions which are marked as encrypted shall be present in encrypted format. I am not sure what you mean by anyone can unmount and read the entire content of the image?
As the flash is encrypted format it is useless without first decrypting it through the key stored in the eFuse. The ability to encrypt and decrypt content using the key is only accessible to the device hardware and trusted code executing on the device. Otherwise no outside entity can use the key for any purposes.
 
> 3- and the other, if I use my own library, then I write to the partition, I have to encrypt the data before writing, or esp_partition_write automatically encrypts the data, and I have to add additional parameters or flags.

esp_partition_write API internally takes care of encryption based on the encrypted flag enabled for the respective partition in the partition table.

> 4- I think I can rewrite another XTS_AES_128_KEY in another block and change any register to use the last block instead of block5?

No that is not possible. As you can see BLOCK5 is already stored with `EFUSE_KEY_PURPOSE_XTS_AES_128_KEY`, no other key can be used for the flash encryption on the device.

> 5- is the XTS_AES_128_KEY safe to post the image online or can anyone read the data with a brute force attack ???
I am unclear about what you mean by image? Do you mean the encrypted image?

As it is encrypted image with standard XTS-AES encryption algorithm. The standard vulnerabilities that may apply to XTS-AES 256 block cipher mode shall be applicable. Please take a look at [Flash encryption algorithm](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#flash-encryption-algorithm) for more details. AFAIK it is  not very easy to decipher the encrypted cipher text without the plaintext key.

AdityaHPatwardhan, 2023-05-06:
> I want to know how can I public my app image (contain passwords and credential) in a public website and every one can download this bin file in plaintext and read everithing.

Please note that if you make your application image as public then everyone shall know the password and credentials.
It is not recommended to make the secret data such as password and credentials as public. 

> I think on first boot after create the XTS_AES_128_KEY key its esential to save a copy on host computer. and it shuld encrypt every image before send to the esp32.

Yes, you need to encrypt the image on host first before writing it to the esp32s3 module.

> other problem, if I public a encrypted image online, and the esp32 download the image via OTA. How can it write encripted image, if the esp_partition_write encrypt all data before write to flash?

Please checkout the [pre-encrypted OTA feature](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/esp_https_ota.html#ota-upgrades-with-pre-encrypted-firmware), this should satisfy your use-case.

Also, this does not seem to be an issue/bug as there is no problem with the esp-idf.
I am closing the issue here, please feel free to raise a query on [esp32 forum](https://esp32.com/index.php) for further questions. If you search related topics then you would get more information. ( e.g., [Encrypted firmware update](https://esp32.com/viewtopic.php?f=13&t=33356&p=113139&hilit=Flash+encryption+encrypted+firmware#p113139))

Please feel free to re-open the issue if you feel the above information does not solve your problem and that you think fixing some bug in esp-idf should solve it.

closed 2023-05-06 by AdityaHPatwardhan