espressif/esp-idf #11322: Bluedroid + release/v5.1 + `-O2` = build fail

KaeLL, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-350-g56123c52aa

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful build.

### What is the actual behavior?

Build fail

### Steps to reproduce.

Enable `-O2` (CONFIG_COMPILER_OPTIMIZATION_PERF) and build

### Build or installation Logs.

```plain
/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:35:34: error: 'strncpy' specified bound 33 equals destination size [-Werror=stringop-truncation]
   35 | #define BCM_STRNCPY_S(x1,x2,x3)  strncpy((x1),(x2),(x3))
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
/esp-idf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c:89:9: note: in expansion of macro 'BCM_STRNCPY_S'
   89 |         BCM_STRNCPY_S(param.get_dev_name_cmpl.name, name, BTC_MAX_LOC_BD_NAME_LEN + 1);
```
```

KaeLL, 2023-05-03:
By release/v5.1 in the title I meant it's affecting it and master. I didn't test other refs.

zhp0406, 2023-05-05:
Is the above issue encountered while using the "esp_ble_gap_get_device_name" API? I am using the latest esp-idf v5.1 and I have not been able to reproduce the problem. Could you provide more log information and the specific commit ID of the esp-idf?

zhp0406, 2023-05-05:
I am using the latest esp-idf v5.1（commit id 420ebd208ae9eb71cb71ebd22742d1175f11addc）and I have not been able to reproduce the problem.
Could you provide more log information and the specific commit ID of the esp-idf?

AxelLin, 2023-05-05:
The commit ID is mentioned in the issue report: v5.2-dev-350-g56123c52aa
BTW, I can confirm this build issue when build for esp32c3.
Make sure you have CONFIG_COMPILER_OPTIMIZATION_PERF=y && CONFIG_BT_BLUEDROID_ENABLED=y.

SlitheringMule, 2023-06-07:
Happens to me too on latest 5.1 release

zhp0406, 2023-06-10:
> Happens to me too on latest 5.1 release
Which example did you encounter this problem in, and did you make any changes to the code of the example?

SlitheringMule, 2023-06-10:
> > Happens to me too on latest 5.1 release
> Which example did you encounter this problem in, and did you make any changes to the code of the example?
Code which works fine without any optimisation, but with -O2 it failed. Generic bluedroid code I don't have anymore I deleted

zhp0406, 2023-06-11:
> The commit ID is mentioned in the issue report: v5.2-dev-350-g56123c52aa BTW, I can confirm this build issue when build for esp32c3. Make sure you have CONFIG_COMPILER_OPTIMIZATION_PERF=y && CONFIG_BT_BLUEDROID_ENABLED=y.

Thank you for reporting the bug. I have successfully reproduced it and the upcoming commit will fix the issue.

SlitheringMule, 2023-06-12:
> Thank you for reporting the bug. I have successfully reproduced it and the upcoming commit will fix the issue.

I was having the bug on the esp32, maybe it's worth checking it now works for it too

zhp0406, 2023-06-14:
> I was having the bug on the esp32, maybe it's worth checking it now works for it too

you are right,All chips utilizing Bluedroid will experience an impact.

habablaxe, 2023-07-14:
for a temporary solution:
esp-idf v5.1
in file: \esp-idf\v5.1\components\bt\CMakeLists.txt
change line 747: 
  set(jump_table_opts "-fjump-tables -Wno-error=stringop-truncation")

caidamao, 2023-08-01:
What is the approximate stage to provide a fix for this issue?
When 5.0 was updated to 5.1, almost all projects were blocked.
> for a temporary solution: esp-idf v5.1 in file: \esp-idf\v5.1\components\bt\CMakeLists.txt change line 747: set(jump_table_opts "-fjump-tables -Wno-error=stringop-truncation")

AxelLin, 2023-08-01:
> Thank you for reporting the bug. I have successfully reproduced it and the upcoming commit will fix the issue.

@zhp0406 
It is a big surprise that it takes so much time to fix a build failure .(This issue has been reported for almost 3 Months)

zhp0406, 2023-08-10:
@AxelLin 
I'm sorry, the fix for this bug has already been applied to the master branch . but due to some other reasons, the changes have not yet taken effect in version 5.1. 
You can cherry-pick it over for use (commit 0634bb904ea8fdf014d173b8f142638491453775). 
Thank you for bringing up the issue.

Alvin1Zhang, 2023-11-02:
Thanks for reporting and sorry for the very slow turnaround, the fix on master branch is available at https://github.com/espressif/esp-idf/commit/126b98d88987e3a1b2fb4b9887218e914a0ac704, and already back ported to other release branches. Feel free to reopen, thanks.

closed 2023-11-02 by Alvin1Zhang

zhp0406, 2023-11-07:
available at 62f5905be25e8f096032fb0f62dd768e7f0008de on release/v5.1 branch