espressif/esp-idf #11323: ability to explicitly turn off dead-times in mcpwm

moefear85, 2023-05-03:
### Is your feature request related to a problem?

Currently, the only way to disable dead-times in the mcpwm module after configuring them, is to call mcpwm_generator_set_dead_time() again and passing 0 ticks as its parameters. This feels awkward.

### Describe the solution you'd like.

It would be nice to have a mcpwm_generator_disable_dead_time() with a mcpwm_generator_enable_dead_time(), such that it remembers the last configuration, or at least a mcpwm_generator_remove_dead_time() function, that totally deletes the whole dead-time configuration, such that upon re-enabling, one simply reconfigures it.

suda-morris, 2023-05-04:
Sorry about that. But if you still have a strong opinion on having such a convenient function, maybe you can create a wrapper function around the `mcpwm_generator_set_dead_time` on the application layer.

moefear85, 2023-05-04:
well technically, there are small functions all over the esp-idf API, but it's not really an issue to me, that's why it's called open source. I've already implemented a wrapper.