espressif/esp-idf #11324: automatic mcpwm shutdown when system hangs

moefear85, 2023-05-03:
### Is your feature request related to a problem?

When the esp hangs, for whatever reason, I have it set up to halt rather than reboot, so I realize that something went wrong (during debugging/development). This is problematic with the mcpwm module, since then the duty no longer updates, but the module continues to output a fixed duty, which damages the power stage.

### Describe the solution you'd like.

It would be nice if the mcpwm module is configurable to turn off completely when the system halts. As far as I'm aware, the fault scenario of the module only kicks in when the fault pin is toggled, but not if the system halts.

### Describe alternatives you've considered.

Currently I use hardware safeguards to protect the power stage. It would also be possible to create a software handler to detect reboots and simply halt the system if this was due to an error, but it carries unnecessary overhead and bloats the project, so I don't use this method.

suda-morris, 2023-05-04:
@moefear85  This is an important topic. 

> As far as I'm aware, the fault scenario of the module only kicks in when the fault pin is toggled, but not if the system halts.

There's also a feature called "software fault" which is returned from `mcpwm_new_soft_fault`. And you can call `mcpwm_soft_fault_activate` in any system failure callback function.

moefear85, 2023-05-04:
And that kicks in when the esp32 hangs?

suda-morris, 2023-05-04:
> there is no way to call the activate function.

There can be if the system side can allow register a callback for the panic_handler: 
https://github.com/espressif/esp-idf/blob/master/components/esp_system/panic.c#L231

For the shutdown situation, we already have the `esp_register_shutdown_handler`.

suda-morris, 2023-05-04:
Maybe you can open a new Github issue asking for the panic_hanlder callback feature and close this one? AFAICT, the MCPWM driver side already offered the ability to control the MCPWM output under a **fault** situation.

igrr, 2023-05-04:
I wouldn't be a fan of an idea of allowing arbitrary callbacks to run in the context of the panic handler. In the panic handler we have a rather limited execution environment.

Could each peripheral define a list of registers and related values which should be written on panic? We can collect these all register/value pairs in a linker section, and then the panic handler can go over the list of registers and write pre-defined values there.

moefear85, 2023-05-04:
why close this issue? this is exactly why I opened it. sorry if you misunderstood what I was asking about.

suda-morris, 2023-05-04:
> I wouldn't be a fan of an idea of allowing arbitrary callbacks to run in the context of the panic handler. In the panic handler we have a rather limited execution environment.
>
> Could each peripheral [...]

This is also a good idea. The challenge is, how to set the register value conditionally. The condition is set by the user's configuration and saved as the driver state. Here in MCPWM, there're two different software faults we can activate, depending on how the user sets the brake mode.

suda-morris, 2023-05-04:
> why close this issue? this is exactly why I opened it. sorry if you misunderstood what I was asking about.

@moefear85 nevermind, you can keep this opened. But the focus is not on the MCPWM anymore, but have moved forward to a system feature: How to insert a user logic in the panic handler.

igrr, 2023-05-04:
Would simply resetting the MCPWM peripheral be equivalent to a "shutdown"? The panic handler already has logic to reset a few communication peripherals, this could be extended with more (or a configurable list of) peripherals.

suda-morris, 2023-05-04:
> Would simply resetting the MCPWM peripheral be equivalent to a "shutdown"? The panic handler already has logic to reset a few communication peripherals, this could be extended with more (or a configurable list of) peripherals.

1. The reset logic only exists if the panic behavior is not set to "restart". That's to say, if `CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT` is enabled (I think @moefear85 is using this feature), we still can't put the MCPWM into a predefined state.
2. Whether a reset can work depends on how the external MOSFET half-bridge treats a "safe" state. Theoretically, there can be 2**2=4 choices. (PWM_H,PWM_L) = any of [(0,0), (0,1), (1,0), (1,1)].

closed 2023-05-11 by espressif-bot