espressif/esp-idf #11327: inverting the mcpwm dead-time output is not selective per generator

moefear85, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

fatal: No names found, cannot describe anything.

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 rev1 module

### Power Supply used.

External 3.3V

### What is the expected behavior?

from the page [MCPWM Dead-Times](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html#dead-time), I feel it is correct to assume that it is possible to selectively invert the output for only one signal but not the other, by setting `dead_time_config.flags.invert_output = true;` as in the following excerpt from that page:
```
mcpwm_dead_time_config_t dead_time_config = {
        .posedge_delay_ticks = 50,
        .negedge_delay_ticks = 0
    };
    ESP_ERROR_CHECK(mcpwm_generator_set_dead_time(gena, gena, &dead_time_config));
    dead_time_config.posedge_delay_ticks = 0;
    dead_time_config.negedge_delay_ticks = 100;
    dead_time_config.flags.invert_output = true;
    ESP_ERROR_CHECK(mcpwm_generator_set_dead_time(gena, genb, &dead_time_config));
```

When I try it myself, I expect it to selectively invert only one signal but not the other.

### What is the actual behavior?

Either both signals are inverted simultaneously, or neither.

### Steps to reproduce.

This bug exists since a long time, at least many months. Only now have I been able to distill a minimal example that exposes it (it is a c++ project, in case that is relevant):

```
#include "driver/mcpwm_prelude.h"
#include "driver/gpio.h"

const uint8_t SIDE_H=1;
const uint8_t SIDE_L=0;

const uint8_t PHASES=2;
const uint8_t SIDES=2;

gpio_num_t pin_L=GPIO_NUM_23;
gpio_num_t pin_H=GPIO_NUM_18;

//uint32_t dead_ticks=0;
uint32_t mcpwm_tmr_rsltn=10000000; // 1 tick = 0.1us
uint32_t mcpwm_period=400; // 25KHz

bool actns_invrt_l=false;
bool actns_invrt_h=false;
bool deadtime_enable=true;
bool dedtme_edgnss_h=false;
bool dedtme_edgnss_l=false;
bool dedtme_invrt_l=false;
bool dedtme_invrt_h=true;

mcpwm_timer_handle_t mcpwm_timer={};
mcpwm_oper_handle_t mcpwm_operator={};
mcpwm_gen_handle_t mcpwm_generators[SIDES]={};
mcpwm_cmpr_handle_t mcpwm_comparator={};
mcpwm_cap_timer_handle_t mcpwm_capture_timer=NULL;

extern "C" void app_main(void)
{
     mcpwm_timer_config_t timer_config=
    {
        .group_id=0,
        .clk_src=MCPWM_TIMER_CLK_SRC_DEFAULT,
        .resolution_hz=mcpwm_tmr_rsltn,
        .count_mode=MCPWM_TIMER_COUNT_MODE_UP,
        .period_ticks=mcpwm_period,
    };
    ESP_ERROR_CHECK( mcpwm_new_timer(&timer_config, &mcpwm_timer) );
    

    mcpwm_operator_config_t operator_config=
    {
        .group_id=0,
        .flags=
        {
            .update_gen_action_on_tez=true,
            .update_dead_time_on_tez=true,
        },
    };
    ESP_ERROR_CHECK( mcpwm_new_operator(&operator_config, &mcpwm_operator) );
    ESP_ERROR_CHECK( mcpwm_operator_connect_timer(mcpwm_operator, mcpwm_timer) );

    mcpwm_comparator_config_t compare_config=
    {
        .flags=
        {
            .update_cmp_on_tez=true,
        },
    };
    ESP_ERROR_CHECK( mcpwm_new_comparator(mcpwm_operator, &compare_config, &mcpwm_comparator) );

    int gen_gpios[2]={pin_L,pin_H};
    mcpwm_generator_config_t gen_config={};
    for (int i=0;i<2;i++)
    {
        gen_config.gen_gpio_num = gen_gpios[i];
        ESP_ERROR_CHECK( mcpwm_new_generator(mcpwm_operator, &gen_config, &mcpwm_generators[i]) );
    }

    
    for (int i=0;i<2;i++)
    {
        ESP_ERROR_CHECK( mcpwm_generator_set_action_on_timer_event(mcpwm_generators[i],MCPWM_GEN_TIMER_EVENT_ACTION(MCPWM_TIMER_DIRECTION_UP, MCPWM_TIMER_EVENT_EMPTY, MCPWM_GEN_ACTION_HIGH)) );
        ESP_ERROR_CHECK( mcpwm_generator_set_action_on_compare_event(mcpwm_generators[i],MCPWM_GEN_COMPARE_EVENT_ACTION(MCPWM_TIMER_DIRECTION_UP, mcpwm_comparator, MCPWM_GEN_ACTION_LOW)) );
    }

    mcpwm_dead_time_config_t deadtime_config_L={.posedge_delay_ticks=1}; // necessary, otherwise deadtimes won't kick in
    deadtime_config_L.flags.invert_output=false;
    ESP_ERROR_CHECK( mcpwm_generator_set_dead_time(mcpwm_generators[0], mcpwm_generators[0], &deadtime_config_L) );
    mcpwm_dead_time_config_t deadtime_config_H={.posedge_delay_ticks=1};
    deadtime_config_H.flags.invert_output=true;
    ESP_ERROR_CHECK( mcpwm_generator_set_dead_time(mcpwm_generators[1], mcpwm_generators[1], &deadtime_config_H) );

    ESP_ERROR_CHECK( mcpwm_timer_enable(mcpwm_timer) );
    ESP_ERROR_CHECK( mcpwm_timer_start_stop(mcpwm_timer,MCPWM_TIMER_START_NO_STOP) );

    mcpwm_comparator_set_compare_value(mcpwm_comparator,mcpwm_period*0.25);
}
```

### More Information.

I know you might object that I've got the generators wrong in the call to `mcpwm_generator_set_dead_time(...)`, but not only is the documentation confusing as to what it actually means, and not only does everything else work fine when I use the combination as above, but also there are 16 possible combinations in total, and I've iterated over all of them. Neither one produces the desired output.
So if selective inversion truly is possible, and there is a combination of generators that works, then there must be a bug in the library.

suda-morris, 2023-05-04:
Hi @moefear85 the inverting is not applied to either generator, it's applied to the "deadtime path". 

[image]

```c
mcpwm_dead_time_config_t dead_time_config = {
        .posedge_delay_ticks = 50,
        .negedge_delay_ticks = 0
};
ESP_ERROR_CHECK(mcpwm_generator_set_dead_time(gena, gena, &dead_time_config));
```
This way, we're applying the dead time to the` gena` itself, we're going the `gena->gena` path, so we're configuring the switch **S2** here.

```c
    dead_time_config.posedge_delay_ticks = 0;
    dead_time_config.negedge_delay_ticks = 100;
    dead_time_config.flags.invert_output = true;
    ESP_ERROR_CHECK(mcpwm_generator_set_dead_time(gena, genb, &dead_time_config));
```
Now we're producing `genb` from the **original** `gena`, so we're going the `gena->genb` path, so we're configuring the switch **S3**. 

**S2** and **S3** are independent.

moefear85, 2023-05-04:
The documentation clearly is claiming that the inversion is done on the signal, not on the edgeness of the dead-time (whether applied to positive or negative). So if you are claiming the latter, then the documentation is not just misleading, it is outright wrong.

suda-morris, 2023-05-04:
@moefear85 
Thanks for telling me you're misled by the documentation. We will improve it ASAP. FYI, I can add more hints in this diagram: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html#active-high-complementary to highlight the where the inversion happens.

suda-morris, 2023-05-05:
Okay, this is the hardware limitation: 
**you can't set rising edge delay for both MCPWM generator 0 and 1 at the same time**.

Based on your snippet code, if we change the `deadtime_config_H` to 

```c
mcpwm_dead_time_config_t deadtime_config_H = {.negedge_delay_ticks = 1};
```

Then you can see the inverter works only for generator1, not for generator0.

[image]

We will update the programming guide to mention this limitation.

closed 2023-05-21 by espressif-bot

IhorNehrutsa, 2023-09-13:
@suda-morris 

> Hi @moefear85 the inverting is not applied to either generator, it's applied to the "deadtime path".
> 
> [image]
 
Please tell me which document this drawing is from?
Thanks