espressif/esp-idf #11328: A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

cyberumut, 2023-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've been compiling using "make" for a long time. I bought a new esp32s3 test card but I am getting the error in the title. I updated the toolchain to esp32s3.
I don't want to use idf.py when compiling it gives too many errors.
How do I make the default chip esp32s3 using "make & make flash"
[image]

igrr, 2023-05-03:
`make` based IDF build system can only be used to build projects for ESP32. Newer chips are only supported in the CMake based build system.

closed 2023-05-03 by cyberumut

vladimir-vakhter, 2023-11-11:
This URL may be useful: https://www.espboards.dev/blog/use-esp-idf-with-vscode/
It explains how to change the target device