espressif/esp-idf #11333: MQTT5 esp-idf example not working with test.mosquitto.org broker

krupis, 2023-05-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I am trying to use mqtt5 from the esp-idf examples:
esp\esp-idf\examples\protocols\mqtt5.

The default MQTT broker is: mqtt://mqtt.eclipseprojects.io

I have tried that and everything works without any issues.

I have tried to change the MQTT broker to test.mosquitto.org as shown below:
[image]

After building and flashing my ESP32 device (ESP32-Ethernet-Kit V1.2) , I monitor the debug logs:
```
I (0) cpu_start: App cpu up.
I (353) cpu_start: Pro cpu start user code
I (353) cpu_start: cpu freq: 160000000 Hz
I (353) cpu_start: Application information:
I (358) cpu_start: Project name:     mqtt5
I (363) cpu_start: App version:      v5.0.1
I (368) cpu_start: Compile time:     May  4 2023 11:33:17
I (374) cpu_start: ELF file SHA256:  da1a5438147737a8...
I (380) cpu_start: ESP-IDF:          v5.0.1
I (384) cpu_start: Min chip rev:     v0.0
I (389) cpu_start: Max chip rev:     v3.99 
I (394) cpu_start: Chip rev:         v3.0
I (399) heap_init: Initializing. RAM available for dynamic allocation:
I (406) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (412) heap_init: At 3FFB39C0 len 0002C640 (177 KiB): DRAM
I (418) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (425) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (431) heap_init: At 4008CF74 len 0001308C (76 KiB): IRAM
I (439) spi_flash: detected chip: generic
I (442) spi_flash: flash io: dio
W (446) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (460) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (470) MQTT5_EXAMPLE: [APP] Startup..
I (470) MQTT5_EXAMPLE: [APP] Free memory: 296908 bytes
I (480) MQTT5_EXAMPLE: [APP] IDF version: v5.0.1
I (540) system_api: Base MAC address is not set
I (540) system_api: read default base MAC address from EFUSE
I (560) esp_eth.netif.netif_glue: c4:dd:57:5f:29:0b
I (560) esp_eth.netif.netif_glue: ethernet attached to netif
I (2960) ethernet_connect: Waiting for IP(s).
I (2960) ethernet_connect: Ethernet Link Up
I (3960) esp_netif_handlers: example_netif_eth ip: 192.168.64.106, mask: 255.255.254.0, gw: 192.168.65.254
I (3960) ethernet_connect: Got IPv4 event: Interface "example_netif_eth" address: 192.168.64.106
I (4530) ethernet_connect: Got IPv6 event: Interface "example_netif_eth" address: fe80:0000:0000:0000:c6dd:57ff:fe5f:290b, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4530) example_common: Connected to example_netif_eth
I (4540) example_common: - IPv4 address: 192.168.64.106,
I (4540) example_common: - IPv6 address: fe80:0000:0000:0000:c6dd:57ff:fe5f:290b, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4560) MQTT5_EXAMPLE: Other event id:7
W (4660) mqtt5_client: Not authorized
I (4660) MQTT5_EXAMPLE: MQTT_EVENT_ERROR
I (4660) MQTT5_EXAMPLE: MQTT5 return code is 135
E (4660) mqtt_client: MQTT connect failed
I (4670) MQTT5_EXAMPLE: MQTT_EVENT_DISCONNECTED

```

Is there any particular reason why it would not allow me to connect to test.mosquitto.org broker?

I have tried other MQTT tools such as MQTT Explorer and I can connect to test.mosquitto.org without any issues.

ESP-YJM, 2023-05-04:
@krupis I think you can remove the https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt5/main/app_main.c#L212-L213 user and pwd part.

krupis, 2023-05-04:
@Esp-Doc 
Thanks for the response. I have commented these 2 lines out and can confirm that it works now. Thanks!

closed 2023-05-04 by krupis