espressif/esp-idf #11336: release/v4.4 install.sh venv error

KaeLL, 2023-05-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-433-g3cec3a0026

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

install.sh to work

### What is the actual behavior?

```
Detecting the Python interpreter
Checking "python" ...
Python 3.11.2
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32
Installing tools: xtensa-esp-elf-gdb, xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20221026 (already installed)
Installing Python environment and packages
pip 23.0.1 from /usr/lib/python3/dist-packages/pip (python 3.11)
Installing virtualenv
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Traceback (most recent call last):
  File "/home/xpt666/idf/esp-idf/tools/idf_tools.py", line 1567, in action_install_python_env
    import virtualenv  # noqa: F401
    ^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'virtualenv'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/xpt666/idf/esp-idf/tools/idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "/home/xpt666/idf/esp-idf/tools/idf_tools.py", line 1972, in main
    action_func(args)
  File "/home/xpt666/idf/esp-idf/tools/idf_tools.py", line 1570, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv'],
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1.
Setting IDF_PATH to '/home/xpt666/idf/esp-idf'
Detecting the Python interpreter
Checking "python" ...
Python 3.11.2
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /usr/bin/python
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
pyserial>=3.3
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
idf-component-manager~=1.2
pygdbmi<=0.9.0.2; python_version > "3.10"
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6,<4
ecdsa>=0.16.0
construct==2.10.54
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /home/xpt666/idf/tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/xpt666/idf/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/home/xpt666/idf/tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/xpt666/idf/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/xpt666/idf/esp-idf/tools:/home/xpt666/.local/bin:/home/xpt666/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
```

### Steps to reproduce.

`release/v4.4`'s `$ ./install.sh esp32`

dobairoland, 2023-05-09:
> So venv and virtualenv aren't the same, I guess. Should probably be reflected on Get-Started's list of packages.

I'm afraid we cannot do that. `python3-virtualenv` is usually old and buggy. Your case is an exception because you use a very new OS with relatively old ESP-IDF (in a sense that the initial release of v4.4 was a long time ago). On older Ubuntus, users have to uninstall it in order our installer could pass.

This has changed in v5.0. We switched to `python3-venv` and this package is in the prerequisites in the installation guide.

closed 2023-05-09 by dobairoland

creatormir, 2023-07-05:
I have a similar problem on windows.
What could be the solution?

```
C:\Users\ib\.espressif\tools\idf-python\3.11.2>C:\Users\ib\.espressif\tools\idf-python\3.11.2\**python.exe -m pip install virtualenv**
Requirement already satisfied: virtualenv in c:\users\ib\appdata\roaming\python\python311\site-packages (20.23.1)
Requirement already satisfied: distlib<1,>=0.3.6 in c:\users\ib\appdata\roaming\python\python311\site-packages (from virtualenv) (0.3.6)
Requirement already satisfied: filelock<4,>=3.12 in c:\users\ib\appdata\roaming\python\python311\site-packages (from virtualenv) (3.12.2)
Requirement already satisfied: platformdirs<4,>=3.5.1 in c:\users\ib\appdata\roaming\python\python311\site-packages (from virtualenv) (3.8.0)
```

noqnio, 2023-08-04:
Hi, I am having the same issue but on windows. I am trying to use IDF v4.4 because that is the version shipped with ADF (I tried v5+patches to no avail).  

The setup of python throws the same error:

`C:\Users\--\.espressif\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "c:\Users\--\.espressif\esp-adf\esp-idf\tools\idf_tools.py", line 1976, in <module>
...
subprocess.CalledProcessError: Command '['C:\\Users\\--\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'C:\\Users\\--\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'C:\\Users\\--\\.espressif\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.`

while if I manually try to run virtual env, it seems to be there:

`--@DESKTOP-Q34LDFO MINGW64 ~/.espressif/tools/idf-python/3.11.2
$ ./python.exe -m virtualenv --version
virtualenv 20.24.2 from C:\Users\--\AppData\Roaming\Python\Python311\site-packages\virtualenv\__init__.py`

I have installed the prerequisites, tried installing IDF v4.4 on its own through vscode addon configure, but nothing.  

Any ideas?

dobairoland, 2023-08-04:
@noqnio https://github.com/espressif/vscode-esp-idf-extension/issues/1015

wizmo2, 2023-09-06:
> I'm afraid we cannot do that. `python3-virtualenv` is usually old and buggy. Your case is an exception because you use a very new OS with relatively old ESP-IDF (in a sense that the initial release [...]

Can I recommend you update the vscode extension/setup tool to remove the option to select older versions of the IDF that require virtualenv (<py3.9) to install.  Simple change that would prevent a lot of grief by all parties!

For others that land here, you can simply downgrade the extension from the extension home tab using `Uninstall/Install Another Version`.   I selected version 1.6.1.  You need to remove any failed `esp` user folder, but then I was able to successfully setup older versions using the `ESP-IDF: Configure ISP-IDF extension` function