espressif/esp-idf #11338: [Crash][NimBLE][ESP32-S3] BLE Indication crashes due to Interrupt WDT

chipweinberger, 2023-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

No panic.

### What is the actual behavior?

After about ~5 indications, eventually I hit a Interrupt WDT. See backtrace.

### Steps to reproduce.

1. connect to esp32s3 over BLE
2. subscribe to indications
3. indicate about ~5 times

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x403880eb  PS      : 0x00060f34  A0      : 0x80378200  A1      : 0x3fcf3270  
0x403880eb: compare_and_set_native at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_hw_support/include/soc/spinlock.h:104
 (inlined by) xPortEnterCriticalTimeout at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/port.c:301

A2      : 0x3fca0068  A3      : 0x0000cdcd  A4      : 0x0000abab  A5      : 0xb33fffff  
A6      : 0x00060f23  A7      : 0x00060f20  A8      : 0x0000abab  A9      : 0x00ffffff  
A10     : 0x00060f23  A11     : 0x00000000  A12     : 0x00060f20  A13     : 0xffffffff  
A14     : 0xfff00000  A15     : 0x02ca0068  SAR     : 0x0000000a  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x403880e8:0x3fcf3270 0x403781fd:0x3fcf32b0 0x4200cda7:0x3fcf32e0
0x403880e8: compare_and_set_native at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_hw_support/include/soc/spinlock.h:104
 (inlined by) xPortEnterCriticalTimeout at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/port.c:301

0x403781fd: xPortEnterCriticalTimeoutSafe at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:598
 (inlined by) vPortEnterCriticalSafe at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:605
 (inlined by) timer_list_lock at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:305

0x4200cda7: timer_process_alarm at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:319 (discriminator 1)
 (inlined by) timer_task at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:386 (discriminator 1)

Core  1 register dump:
PC      : 0x4037825f  PS      : 0x00060534  A0      : 0x80378375  A1      : 0x3fccc4f0  
0x4037825f: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:232

A2      : 0x3fcd73a0  A3      : 0x00000000  A4      : 0x3fca5e44  A5      : 0x3fca5e44  
A6      : 0x501103f3  A7      : 0x00000000  A8      : 0x3fcd73a0  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3fca60f4  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x02fef7dd  A15     : 0x00060520  SAR     : 0x0000000a  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x42016c18  LEND    : 0x42016c28  LCOUNT  : 0x00000013  
0x42016c18: gap_client_get(unsigned short) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:180
 (inlined by) handle_gap_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:528

0x42016c28: gap_client_get(unsigned short) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:179
 (inlined by) handle_gap_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:528

Backtrace: 0x4037825c:0x3fccc4f0 0x40378372:0x3fccc520 0x42095a7e:0x3fccc550 0x4208aa34:0x3fccc580 0x4208af2a:0x3fccc5b0 0x4208d9e0:0x3fccc5e0 0x4208de85:0x3fccc610 0x420172ba:0x3fccc640 0x4201738f:0x3fccc680 0x42012008:0x3fccc6c0 0x4201256e:0x3fccc710 0x42056f22:0x3fccc770 0x42014487:0x3fccc7a0 0x42014c4f:0x3fccc7d0 0x42014ddd:0x3fccc800 0x42014eff:0x3fccc830 0x42011a21:0x3fccc8b0
0x4037825c: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:232

0x40378372: esp_timer_start_once at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:159

0x42095a7e: npl_freertos_callout_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:393

0x4208aa34: ble_npl_callout_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:231
 (inlined by) ble_hs_timer_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:452

0x4208af2a: ble_hs_timer_resched at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:483

0x4208d9e0: ble_gattc_process_status at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:767
 (inlined by) ble_gattc_process_status at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:758

0x4208de85: ble_gattc_indicate_custom at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4358

0x420172ba: indicate_msg(unsigned short, unsigned short, unsigned char const*, unsigned short, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:1155

0x4201738f: pd_bluetooth_indicate_client(unsigned short, pd_ble_chr_t, unsigned char const*, unsigned short, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:1203

0x42012008: _ZL27pd_bthttp_socket_sendmsg_exP18pd_bthttp_socket_t20pd_btsockmsg_delim_tS1_PKhjP10pd_error_t$isra$0 at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_bthttp_socket.cpp:112

0x4201256e: pd_bthttp_socket_resp_send_body_bytes(pd_bthttp_socket_t*, pd_bthttp_resp_t*, unsigned char const*, unsigned int, bool, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_bthttp_socket.cpp:430

0x42056f22: pd_bthttp_resp_sendstr(pd_bthttp_request_t*, char const*, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_bthttp_txrx.cpp:70

0x42014487: pd_xtrans_resp_sendstr(pd_xtrans_t*, char const*, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_xtrans.cpp:412

0x42014c4f: pd_xtrans_resp_json_success(pd_xtrans_t*, cJSON*, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_xtrans.cpp:726 (discriminator 4)

0x42014ddd: _ZL22pd_xhandler_voidToJsonP11pd_xtrans_t18pd_xhandler_args_tP10pd_error_t$isra$0 at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_xhandlers.cpp:145

0x42014eff: pd_xhandler_call_api_generic(pd_xtrans_t*, pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_xhandlers.cpp:448

0x42011a21: bthttp_task(void*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/api/xcoms/pd_bthttp_server.cpp:231
```

rahult-github, 2023-05-05:
Hi @chipweinberger ,

Two queries.

1.  Can you please share application level code at below lines ? 

0x42016c18: gap_client_get(unsigned short) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:180
 (inlined by) handle_gap_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:528

0x42016c28: gap_client_get(unsigned short) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:179
 (inlined by) handle_gap_event at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:528

2. Is it possible for you to suggest what changes you are doing that can be put on existing IDF examples , so as to reproduce issue at our end ?

chipweinberger, 2023-05-05:
> 1. Can you please share application level code at below lines in the handle_gap_event function ? 

Yes. Here it is. To be honest, I don't understand why `gap_client_get` function appears in the registers. Do you? I don't really understand this stack trace.

Note: `gap_client_get` _is_ called in `handle_gap_event`, i.e. from `ble_gap_adv_start(...,handle_gap_event, NULL);`

```
pd_bt_connected_client_t* gap_client_get(uint16_t gapConnHandle)
{
    for (int i = 0; i < MAX_BLE_CLIENTS; i++) {
        if (connectedClients[i].isValid &&
            connectedClients[i].gapConnHandle == gapConnHandle)
        {
            return &connectedClients[i];
        }
    }
    return NULL;
}
```

And here are my other BLE related functions that appear in that stackstrace

<details>
  <summary>pd_bluetooth_indicate_client</summary>

```
void pd_bluetooth_indicate_client(uint16_t gapConnHandle, 
                            pd_ble_chr_t jChr,
                          const uint8_t* msg,
                                uint16_t msgLen,
                             pd_error_t* error)
{
    pd_err_unset(error);

    {
        uint16_t attrHandle = pd_ble_chr_attr_handle(jChr);

        // only indicate if client is subscribed to indications
        for (int i = 0; i < MAX_BLE_SUBSCRIPTIONS; i++){
            if (subscribedClients[i].isValid &&
                subscribedClients[i].attrHandle == attrHandle &&
                subscribedClients[i].gapConnHandle == gapConnHandle &&
                subscribedClients[i].indicate)
            {
                uint16_t gapConnHandle = subscribedClients[i].gapConnHandle;

                eTry(,indicate_msg,(gapConnHandle, attrHandle, msg, msgLen, error));
            }
        }

        pd_err_success(error);
    }
end:
    return;
}
```

</details>

<details>
  <summary>indicate_msg</summary>

```
void indicate_msg(uint16_t gapConnHandle, 
                uint16_t attrHandle, 
          const uint8_t* msg, 
                uint16_t msgLen, 
             pd_error_t* error)
{
    pd_err_unset(error);
[... 232 lines cut ...]
                
                // call subscribe handler
                pd_bt_on_chr_subscribed_handler_t handler = get_on_chr_subscribed_handler(jChr);
                if (handler) {
                    int rv = handler(event->subscribe.conn_handle, jChr);
                    return rv;
                }
            } else {
                gap_subscription_remove(event->subscribe.conn_handle, attrHandle);

                // call unsubscribe handler
                pd_bt_on_chr_unsubscribed_handler_t handler = get_on_chr_unsubscribed_handler(jChr);
                if (handler) {
                    handler(event->subscribe.conn_handle, jChr);
                }
            }
            return 0; // success
        }
        default:
            ESP_LOGI(TAG, "GAP event: %s", ble_gap_event_str(event->type));
            return 0; // success
    }
}
```
</details>

chipweinberger, 2023-05-05:
Two things:

1. I tried commenting out all the timers I am using in my code (I had a couple timers always running every second). Still repros.

2. I hit it again with a slightly different backtrace.

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40378261  PS      : 0x00060334  A0      : 0x80378375  A1      : 0x3fcd6f30  
0x40378261: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:232

A2      : 0x3fcd7388  A3      : 0x00000000  A4      : 0x3fca5e44  A5      : 0x9039dec3  
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x3fcbad18  A9      : 0x0673b164  
A10     : 0x00000000  A11     : 0x3fca60f4  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x06a0db9b  A15     : 0x00060320  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x4037825e:0x3fcd6f30 0x40378372:0x3fcd6f60 0x420959fe:0x3fcd6f90 0x4208a9b4:0x3fcd6fc0 0x4208aad2:0x3fcd6ff0 0x4208ac1d:0x3fcd7020 0x42095f83:0x3fcd7050 0x42015c14:0x3fcd7080
0x4037825e: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:232

0x40378372: esp_timer_start_once at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:159

0x420959fe: npl_freertos_callout_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:393

0x4208a9b4: ble_npl_callout_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:231
 (inlined by) ble_hs_timer_reset at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:452

0x4208aad2: ble_hs_timer_sched at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:473

0x4208ac1d: ble_hs_timer_sched at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:462
 (inlined by) ble_hs_timer_exp at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:416
 (inlined by) ble_hs_timer_exp at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:409

0x42095f83: ble_npl_event_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x42015c14: start_nimble_port_run(void*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_common/jcommon/bluetooth/pd_bluetooth.cpp:629

Core  1 register dump:
PC      : 0x420c17b7  PS      : 0x00060134  A0      : 0x8200677d  A1      : 0x3fcf69a0  
0x420c17b7: esp_pm_impl_waiti at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_pm/pm_impl.c:832

A2      : 0x00000000  A3      : 0x9039dec3  A4      : 0x00060023  A5      : 0x9039dec3  
A6      : 0x00005dc8  A7      : 0x02ca1120  A8      : 0x3fca5e44  A9      : 0x9039dec3  
A10     : 0x00000000  A11     : 0x9039dec3  A12     : 0x9039dec3  A13     : 0x40000000  
A14     : 0x001600ba  A15     : 0x3fced470  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x420c17b4:0x3fcf69a0 0x4200677a:0x3fcf69d0 0x4038514b:0x3fcf6a00
0x420c17b4: cpu_ll_waiti at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:185
 (inlined by) esp_pm_impl_waiti at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_pm/pm_impl.c:830

0x4200677a: esp_vApplicationIdleHook at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/freertos_hooks.c:63

0x4038514b: prvIdleTask at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/tasks.c:3998
```

chipweinberger, 2023-05-06:
So I've found part of the issue. We are inserting timers...forever?

I added code to print on every iteration of LIST_FOREACH in the timers.

```
static IRAM_ATTR esp_err_t timer_insert(esp_timer_handle_t timer, bool without_update_alarm)
{
        ....
        LIST_FOREACH(it, &s_timers[dispatch_method], list_entry) {
            esp_rom_printf("timer %u\n", t);
            t++;
            ....
        }
}
```

After indication logs:
```
timer 0
timer 1
timer 2
timer 3
timer 4
..... forever....
```

chipweinberger, 2023-05-06:
I also added a call to `esp_backtrace_print`  and it looks like there is stack corruptions coming from somewhere.

```
static IRAM_ATTR esp_err_t timer_insert(esp_timer_handle_t timer, bool without_update_alarm)
{
    esp_backtrace_print(12);
```

```
Backtrace: 0x40378341:0x3FCB4360 0x403784AE:0x3FCB43A0 0x40378726:0x3FCB43D0 0x40376A91:0x3FCB4400 0x421124A1:0x3FCB4430 0x40x40378341: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:228

0x403784ae: esp_timer_start_once at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:159

0x40378726: ets_timer_arm at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/ets_timer_legacy.c:91

0x40376a91: timer_arm_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:490

0x421124a1: ieee80211_hostapd_beacon_txcb at ??:?

003F58A:0x3FCB4450 |<-CORRUPTED
```

<details>
  <summary>Logs</summary>

```
I (5049) pd wifi: Entering Discovery Mode
I (5049) pd wifi: wifi_start_internal(ap)
I (5049) pd wifi: esp_netif_init()
I (5049) pd wifi: esp_event_loop_create_default()
I (5049) pd wifi: esp_netif_create_default_wifi_ap()
I (5049) pd wifi: esp_wifi_init()
I (5059) pp: pp rom version: e7ae62f
I (5059) net80211: net80211 rom version: e7ae62f
I (5079) wifi_init: rx ba win: 6
I (5079) wifi_init: tcpip mbox: 32
I (5079) wifi_init: udp mbox: 6
I (5079) wifi_init: tcp mbox: 6
I (5079) wifi_init: tcp tx win: 6500
I (5079) wifi_init: tcp rx win: 6500
I (5079) wifi_init: tcp mss: 1440
I (5079) wifi_init: WiFi/LWIP prefer SPIRAM
I (5089) pd wifi: esp_event_handler_instance_register(ESP_EVENT_ANY_ID)
I (5089) pd wifi: esp_event_handler_instance_register(IP_EVENT_STA_GOT_IP)
I (5089) pd wifi: esp_wifi_set_mode(WIFI_MODE_APSTA)
I (5089) pd wifi: AP channel 10
I (5089) pd wifi: switching to 'Create Wifi AP Hotspot' mode
I (5089) pd wifi: 
ssid:jamcorder-dbidrj 
password:
I (5099) pd wifi: esp_wifi_set_config(WIFI_IF_AP)
I (5109) pd wifi: esp_wifi_start()
I (5109) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11

Backtrace: 0x40378341:0x3FCB42F0 0x403784AE:0x3FCB4330 0x40378726:0x3FCB4360 0x40376A91:0x3FCB4390 0x421129D5:0x3FCB43C0 0x42113EAC:0x3FCB4420 0x42117729:0x3FCB4450 0x42118885:0x3FCB4470 0x4211728E:0x3FCB4490 0x4215D989:0x3FCB44B0
0x40378341: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:228

0x403784ae: esp_timer_start_once at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:159

0x40378726: ets_timer_arm at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/ets_timer_legacy.c:91

0x40376a91: timer_arm_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:490

0x421129d5: ieee80211_hostap_attach at ??:?

0x42113eac: wifi_softap_start at ??:?

0x42117729: _do_wifi_start at ??:?

0x42118885: wifi_start_process at ??:?
[... 994 lines cut ...]
timer 0
timer 1

Backtrace: 0x40378341:0x3FCF32A0 0x4200CFCD:0x3FCF32E0
0x40378341: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:228

0x4200cfcd: timer_process_alarm at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:353
 (inlined by) timer_task at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:392

timer 0
timer 1

Backtrace: 0x40378341:0x3FCB4360 0x403784AE:0x3FCB43A0 0x40378726:0x3FCB43D0 0x40376A91:0x3FCB4400 0x421124A1:0x3FCB4430 0x4003F58A:0x3FCB4450 |<-CORRUPTED
0x40378341: timer_insert at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:228

0x403784ae: esp_timer_start_once at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/esp_timer.c:159

0x40378726: ets_timer_arm at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_timer/src/ets_timer_legacy.c:91

0x40376a91: timer_arm_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:490

0x421124a1: ieee80211_hostapd_beacon_txcb at ??:?
```

</details>

chipweinberger, 2023-05-06:
**More Info:** Enabling Wifi,  Usb Host, Uart, etc, all seem to make it hit the IWDT _sooner_, but it can repro without any of that as well. Just takes longer.

chipweinberger, 2023-05-06:
**New Info:** Not too surprising, when we hit this issue, the timer list is somehow a circular loop.

```
timer 0 callback: 0x42085954 cur: 0x3fcbe700 next: 0x3fcbe700 prev: 0x3fcbe718
0x42085954: ble_npl_event_fn_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:302

timer 1 callback: 0x42085954 cur: 0x3fcbe700 next: 0x3fcbe700 prev: 0x3fcbe718
0x42085954: ble_npl_event_fn_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:302

timer 2 callback: 0x42085954 cur: 0x3fcbe700 next: 0x3fcbe700 prev: 0x3fcbe718
0x42085954: ble_npl_event_fn_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:302

timer 3 callback: 0x42085954 cur: 0x3fcbe700 next: 0x3fcbe700 prev: 0x3fcbe718
0x42085954: ble_npl_event_fn_wrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:302
```

AxelLin, 2023-05-06:
Just wondering if this is related: https://github.com/espressif/esp-idf/pull/11215 (maybe you can try this fix)

chipweinberger, 2023-05-06:
Hah, I  was just about to comment the same thing. Let me try that!

chipweinberger, 2023-05-06:
**YES THAT FIXES THE CRASH.**

chipweinberger, 2023-05-22:
I see the fix was merged in master and v5.1

https://github.com/espressif/esp-idf/commit/500b5b21036db37855caeb7dd673f62fbe8a8cb7

closed 2023-05-22 by chipweinberger