espressif/esp-idf #11339: "`noreturn` attribute ignored" warning with ESP-IDF v5.0.1

RobMeades, 2023-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1 a4afa444

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compile without warnings.

### What is the actual behavior?

I'm getting some warnings from within the ESP-IDF header files themselves about the `noreturn` attribute being ignored, see below.  Is this (a) just a "fact of life" given where version 5 is at the moment, or (b) might I be doing something wrong, or (c) should I be suppressing them, etc...?

```
In file included from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/esp_intr_alloc.h:11,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/esp_cpu.h:20,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/spinlock.h:11,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/ubxlib/workspace/ubxlib_feature_espidf_v5_rmea@2/ubxlib/port/platform/esp-idf/src/u_port.c:36:
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_common/include/esp_err.h:79:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
   79 | void _esp_error_check_failed(esp_err_t rc, const char *file, int line, const char *function, const char *expression) __attribute__((noreturn));
      | ^~~~
In file included from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:83,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/ubxlib/workspace/ubxlib_feature_espidf_v5_rmea@2/ubxlib/port/platform/esp-idf/src/u_port.c:36:
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_system/include/esp_system.h:75:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
   75 | void esp_restart(void) __attribute__ ((noreturn));
      | ^~~~
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_system/include/esp_system.h:115:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
  115 | void  __attribute__((noreturn)) esp_system_abort(const char* details);
      | ^~~~
```

### Steps to reproduce.

Include "FreeRTOS.h" in a `.c` file and compile that `.c` file using `idf.py`.

### Build or installation Logs.

```plain
In file included from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/esp_intr_alloc.h:11,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/esp_cpu.h:20,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/spinlock.h:11,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/ubxlib/workspace/ubxlib_feature_espidf_v5_rmea@2/ubxlib/port/platform/esp-idf/src/u_port.c:36:
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_common/include/esp_err.h:79:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
   79 | void _esp_error_check_failed(esp_err_t rc, const char *file, int line, const char *function, const char *expression) __attribute__((noreturn));
      | ^~~~
In file included from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:83,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/ubxlib/workspace/ubxlib_feature_espidf_v5_rmea@2/ubxlib/port/platform/esp-idf/src/u_port.c:36:
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_system/include/esp_system.h:75:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
   75 | void esp_restart(void) __attribute__ ((noreturn));
      | ^~~~
/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_system/include/esp_system.h:115:1: warning: '_Noreturn' attribute directive ignored [-Wattributes]
  115 | void  __attribute__((noreturn)) esp_system_abort(const char* details);
      | ^~~~
```

igrr, 2023-05-05:
Could you please post the complete command line which is used to compile this source file (`u_port.c`)? It should be printed above the error message. Alternatively, you can find it in build/compile_commands.json, searching for `u_port.c`.

RobMeades, 2023-05-05:
Thanks for the swift response: here's the relevant line from `compile_cmmands.json` I think:

```
  "directory": "/home/ubxlib/workspace/ubxlib_PR-876@2/_jenkins_work/10.0/build",
  "command": "/home/ubxlib/.ubxlibpkg/esp_idf_tools-v5.0.1/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -I/home/ubxlib/workspace/ubxlib_PR-876@2/_jenkins_work/10.0/build/config -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/at_client/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/error/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/assert/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/location/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/mqtt_client/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/http_client/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/security/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/sock/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/ubx_protocol/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/spartn/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/utils/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/debug_utils/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/cfg -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/type/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/network/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/device/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/short_range/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/ble/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/wifi/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/cell/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/gnss/api -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf/mcu/esp32/cfg -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/clib -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/mbedtls -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/at_client/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/assert/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/location/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/mqtt_client/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/http_client/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/security/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/sock/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/ubx_protocol/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/spartn/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/utils/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/debug_utils/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/event_queue -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/mutex_debug -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/common/log_ram -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/network/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/device/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/common/short_range/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/ble/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/wifi/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/cell/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/gnss/src -I/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf/src -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/newlib/platform_include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/esp_additions/include/freertos -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/freertos/esp_additions/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/soc -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/include/soc/esp32 -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/port/esp32/. -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_hw_support/port/esp32/private_include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/heap/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/log/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/soc/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/soc/esp32/. -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/soc/esp32/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/hal/esp32/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/hal/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/hal/platform_port/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_rom/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_rom/include/esp32 -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_rom/esp32 -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_common/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v5.0.1/components/esp_system/include -I/home/ubxlib/.ubxlibpkg/esp_idf-v [... cut ...] _APP_CLOUD_LOCATE_MQTT_USERNAME=WF592TTWUQ18512KLU6L \"-DU_WIFI_TEST_CFG_WPA2_PASSPHRASE=GqpZvmK8@r5yL#AP\" -DU_CFG_APP_CLOUD_LOCATE_MQTT_PASSWORD=LNvUrEp9efTu6F6Xy4kcxtWwfE189GoVEJXfuVkV -o esp-idf/ubxlib/CMakeFiles/__idf_ubxlib.dir/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf/src/u_port.c.obj -c /home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf/src/u_port.c",
  "file": "/home/ubxlib/workspace/ubxlib_PR-876@2/ubxlib/port/platform/esp-idf/src/u_port.c"
```

igrr, 2023-05-05:
Are you including anything else before "freertos/FreeRTOS.h"? I tried compiling a source file with only `#include "freertos/FreeRTOS.h"` with the same compilation flags as in your command, but didn't get any warning unfortunately.

RobMeades, 2023-05-05:
Aha, yes, so we do bring in a `compiler.h` file, in which we define a `U_NO_RETURN` macro for ourselves, but in doing so that is bringing in `stdnoreturn.h` for later compiler versions, which I guess your 5.0.1 tools now are:

```
# if defined(__STDC_VERSION__) && __STDC_VERSION__ >= 201112L
#  include <stdnoreturn.h>
#  define U_NO_RETURN noreturn
# else
#  define U_NO_RETURN __attribute__((noreturn))
# endif
```

The commit where this was introduced says:  "Fix "control reaches end of non-void function" GCC compile issue: when enabling `-Wreturn-type` GCC didn't understand that our assert function never returns."

Question is, what is the right thing to do?  Have we got the STDC version wrong for this, or should we just add another conditional so that `#if ESP_IDF_VERSION_MAJOR >= 5` we use the `__attribute__((noreturn))` form, which is what your code seems to be using?

igrr, 2023-05-05:
The fact that including stdnoreturn.h before freertos/FreeRTOS.h leads to this warning is still a bug in IDF, i think, so we'll have to fix this!

In the meantime, you can add an `ESP_IDF_VERSION_*` condition to your code to avoid including stdnoreturn.h, just to get your work unblocked.

RobMeades, 2023-05-05:
Excellent, I will do that, you may close this issue if you wish.  Thanks again for the remarkably swift response.

igrr, 2023-05-05:
You are welcome! We can keep this open, the issue will get closed automatically when we push a fix.

closed 2023-05-21 by espressif-bot