espressif/esp-idf #11340: Idf.py failing with AttributeError: 'HTTPResponse' object has no attribute 'strict'

jkano, 2023-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Being able to build the blink project with idf.py

### What is the actual behavior?

I followed the Standard Toolchain Setup for Linux and macOS for version 5.0.1 (https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/get-started/linux-macos-setup.html) and got to the section **Start a Project** but instead of using the hello-world I used the blink project.

Copied the blink project to **~/development/esp/projects** folder and ran

`idf.py set-target esp32`

This is what I get:

```
jose@mac blink % idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/Users/jose/development/esp/projects/blink/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /Users/jose/development/esp/projects/blink/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/jose/development/esp/projects/blink"...
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/jose/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/jose/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/jose/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
CMake Error at /Users/jose/development/esp/esp-idf/tools/cmake/build.cmake:519 (message):
  Traceback (most recent call last):

    File "/opt/homebrew/Cellar/python@3.10/3.10.11/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py", line 196, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/opt/homebrew/Cellar/python@3.10/3.10.11/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
[... 162 lines cut ...]
      data = self._raw_read(amt)
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 809, in _raw_read
      data = self._fp_read(amt) if not fp_closed else b""
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/urllib3/response.py", line 794, in _fp_read
      return self._fp.read(amt) if amt is not None else self._fp.read()
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 96, in read
      self._close()
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/filewrapper.py", line 76, in _close
      self.__callback(result)
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 353, in cache_response
      self._cache_set(cache_url, request, response, body, expires_time)
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/controller.py", line 274, in _cache_set
      self.serializer.dumps(request, response, body),
    File "/Users/jose/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/cachecontrol/serialize.py", line 54, in dumps
      u"strict": response.strict,

  AttributeError: 'HTTPResponse' object has no attribute 'strict'

Call Stack (most recent call first):
  /Users/jose/development/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /Users/jose/development/esp/projects/blink/build/log/idf_py_stderr_output_36590 and /Users/jose/development/esp/projects/blink/build/log/idf_py_stdout_output_36590
```

jkano, 2023-05-05:
Thanks for the quick reply, it worked! 

I think this should be on the documentation as a workaround until the fix is made.

At least the issue and fix are now here for visibility.

dobairoland, 2023-05-05:
The workaround is already available. It is enough to re-run the install script and ESP-IDF environment should be fixed. No need to checkout a new version of ESP-IDF (v5.0 or later versions). As for v4.4 and older versions, the fix will come soon.

dobairoland, 2023-05-09:
Fix for v4.4: 743d6a12c48134b298707475d9a54f2fd3beee89