espressif/esp-idf #11341: i2c_driver_delete(i2c_port_t i2c_num) crashes IDF 5.0.1 but works on IDF 5.0

d18-builder, 2023-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF version: v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected i2c_driver_delete(i2c_port_t i2c_num) to free the used resources (GPIO, buffers etc).

### What is the actual behavior?

Instead it just crashed my application.

### Steps to reproduce.

Start a blank project or modify an existing one.

1. Use i2c_param_config(i2c_port_t i2c_num, const i2c_config_t *i2c_conf) to configure i2c port...
2. Use i2c_driver_install(i2c_port_t i2c_num, i2c_mode_t mode, size_t slv_rx_buf_len, size_t slv_tx_buf_len, int intr_alloc_flags) to install the i2c port...
3. Lastly, use  i2c_driver_delete(i2c_port_t i2c_num) uninstall the current i2c port and free the resources used...

It will crash on step 3...

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40376c14  PS      : 0x00060033  A0      : 0x82005614  A1      : 0x3fcf0e30
0x40376c14: esp_intr_disable at C:/Users/pgdur/esp/esp-idf/components/esp_hw_support/intr_alloc.c:824

A2      : 0x3fcec4bc  A3      : 0x3fca1d1c  A4      : 0x00060020  A5      : 0x00060023
A6      : 0x00060023  A7      : 0x0000cdcd  A8      : 0x00060620  A9      : 0x3fcf0e00
A10     : 0x00000001  A11     : 0x3fca1d24  A12     : 0x00000000  A13     : 0x00060023
A14     : 0x000000fd  A15     : 0x0000cdcd  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00060620  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40376c11:0x3fcf0e30 0x42005611:0x3fcf0e50 0x420056f1:0x3fcf0e70 0x4037eaac:0x3fcf0e90 0x40386f79:0x3fcf0eb0
0x40376c11: esp_intr_disable at C:/Users/pgdur/esp/esp-idf/components/esp_hw_support/intr_alloc.c:823

0x42005611: esp_intr_free at C:/Users/pgdur/esp/esp-idf/components/esp_hw_support/intr_alloc.c:710

0x420056f1: esp_intr_free_cb at C:/Users/pgdur/esp/esp-idf/components/esp_hw_support/intr_alloc.c:690

0x4037eaac: ipc_task at C:/Users/pgdur/esp/esp-idf/components/esp_system/esp_ipc.c:87

0x40386f79: vPortTaskWrapper at C:/Users/pgdur/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

### More Information.

This was not a problem in IDF 5.0 official release. If I revert back to 5.0, the problem is gone.

mythbuster5, 2023-05-06:
Can you share your code of this part directly?

mythbuster5, 2023-05-17:
Which commit are you using? I can't reproduce the same issue locally.

in i2c_sample example, my code is like 
```
void app_main(void)
{
    uint8_t data[2];
    ESP_ERROR_CHECK(i2c_master_init());
    ESP_LOGI(TAG, "I2C initialized successfully");

    // /* Read the MPU9250 WHO_AM_I register, on power up the register should have the value 0x71 */
    // ESP_ERROR_CHECK(mpu9250_register_read(MPU9250_WHO_AM_I_REG_ADDR, data, 1));
    // ESP_LOGI(TAG, "WHO_AM_I = %X", data[0]);

    // /* Demonstrate writing by reseting the MPU9250 */
    // ESP_ERROR_CHECK(mpu9250_register_write_byte(MPU9250_PWR_MGMT_1_REG_ADDR, 1 << MPU9250_RESET_BIT));

    ESP_ERROR_CHECK(i2c_driver_delete(I2C_MASTER_NUM));
    ESP_LOGI(TAG, "I2C de-initialized successfully");
}
```

AxelLin, 2023-12-07:
@d18-builder Do you still hit the issue with latest v5.0.4?

d18-builder, 2023-12-13:
I am currently using ESP-IDF v5.1.2
<https://esp32.com/viewtopic.php?f=10&p=123464#p123464>and this issue is no
longer present in this version.

B.R.

AxelLin, 2023-12-16:
@d18-builder
Could you verify if it works with v5.0.5? (If yes, you can close this issue)

mythbuster5, 2025-01-13:
Long time no respond and issue is not exist in newer version (5.1.2) so this can be closed

closed 2025-01-13 by mythbuster5