espressif/esp-idf #11342: Getting SD-Card write errors or other random errors when writing to SD while requesting data through httpd rest endpoint

paulusTecnion, 2023-05-05:
I have an ESP32-S2 connected to another microcontroller U1 through SPI on SPI3 (1 bit mode, 20 MHz) and have an SD card card on SPI2 (1 bit mode, 20 MHz). One task is logging data from U1 to the SD-card. Furthermore, I have the http rest server from ESP running in another task. I use the USB as output console. WiFi is enabled in APSTA mode. 

I am retrieving data from U1, processing the data and writing it to a CSV on the SD card. The ESP is master and U1 slave. U1 toggles a pin HIGH to indicate data is ready to be transmitted which is detected through an interrupt, after which it gets retrieved by the ESP32-S2 using DMA.

The data chunks between the ESP32-S2 and U1 are 2328 bytes large, processed and saved in memory on the ESP32. When 4 transactions have taken please (so 4*2328) and converted the data is written to the SD card. Writing happens every 600-700 ms. 

I have a web portal on the same ESP as an UI interface, which is requesting the state of the ESP32 every 1 second through a REST endpoint. I connect my laptop to the ESP32 as client. No interaction is done with the SD card from the REST endpoint. However, after a while, which can vary from 5 minutes up to 16 hours, the SD card returns diskio cmd  0x107, 0x108 or 0x109 errors. For example, this happened after 16 hours of logging:

```
[09:37:36:612] <0x1b>[0;32mI (56998143) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:37:36:582] <0x1b>[0;32mI (60598103) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:38:37:372] <0x1b>[0;32mI (60658893) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:39:38:002] <0x1b>[0;32mI (60719523) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:40:38:333] <0x1b>[0;32mI (60779853) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:41:38:983] <0x1b>[0;32mI (60840503) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:42:39:483] <0x1b>[0;32mI (60901003) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:43:40:103] <0x1b>[0;32mI (60961623) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[10:44:34:613] <0x1b>[0;31mE (61016133) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108<0x1b>[0m␍␊
[10:44:34:623] <0x1b>[0;31mE (61016143) diskio_sdmmc: sdmmc_write_blocks failed (264)<0x1b>[0m␍␊
[10:44:34:633] <0x1b>[0;31mE (61016153) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108<0x1b>[0m␍␊
[10:44:34:643] <0x1b>[0;31mE (61016163) diskio_sdmmc: sdmmc_write_blocks failed (264)<0x1b>[0m␍␊
[10:44:34:653] <0x1b>[0;31mE (61016173) LOGGER: Error 0x00000010 occured in Logging statemachine. Stopping..<0x1b>[0m␍␊
[10:44:34:663] <0x1b>[0;31mE (61016183) LOGGER: Error 0x00000010 occured in Logging statemachine. Stopping..<0x1b>[0m␍␊
[10:44:34:713] <0x1b>[0;31mE (61016233) LOGGER: Error 0x00000010 occured in Logging statemachine. Stopping..<0x1b>[0m␍␊
```

Generally in other issues / forums such errors were due to problems with the power supply towards the SD-card. However, the power supply is stable. I also tried to enable power saving of the WiFi (enabled MIN power saving), but it didn't make a difference.

I sometimes also get random panic handler errors like the following (including start of logging):

```
[20:53:18:856] <0x1b>[0;32mI (34911) LOGGER: File seq nr: 5<0x1b>[0m␍␊
[20:53:18:867] <0x1b>[0;32mI (34911) LOGGER: Changing LOGTASK state from 1 to 2<0x1b>[0m␍␊
[20:53:57:009] <0x1b>[0;31mE (73061) esp-rest: Failed to open file : /www/js/jquery-3.6.4.min.js<0x1b>[0m␍␊
[20:53:57:019] <0x1b>[0;31mE (73071) esp-rest: Trying /www/js/jquery-3.6.4.min.js.gz<0x1b>[0m␍␊
[20:54:00:819] <0x1b>[0;31mE (76871) esp-rest: Failed to open file : /www/js/jquery-3.6.4.min.js<0x1b>[0m␍␊
[20:54:00:829] <0x1b>[0;31mE (76881) esp-rest: Trying /www/js/jquery-3.6.4.min.js.gz<0x1b>[0m␍␊
[20:54:01:448] <0x1b>[0;31mE (77491) esp-rest: Failed to open file : /www/js/plotly-basic-2.22.0.min.js<0x1b>[0m␍␊
[20:54:01:474] <0x1b>[0;31mE (77501) esp-rest: Trying /www/js/plotly-basic-2.22.0.min.js.gz<0x1b>[0m␍␊
[21:53:06:139] <0x1b>[0;32mI (3622191) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:54:06:160] <0x1b>[0;32mI (3682211) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:55:06:210] <0x1b>[0;32mI (3742261) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:56:06:500] <0x1b>[0;32mI (3802541) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:57:07:040] <0x1b>[0;32mI (3863091) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:58:07:300] <0x1b>[0;32mI (3923351) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[21:59:07:660] <0x1b>[0;32mI (3983711) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:00:08:610] <0x1b>[0;32mI (4044661) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:01:09:470] <0x1b>[0;32mI (4105521) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:02:09:590] <0x1b>[0;32mI (4165641) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:03:09:711] <0x1b>[0;32mI (4225761) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:04:10:090] <0x1b>[0;32mI (4286141) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:05:10:831] <0x1b>[0;32mI (4346881) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:06:10:971] <0x1b>[0;32mI (4407021) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:07:11:881] <0x1b>[0;32mI (4467931) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:08:11:931] <0x1b>[0;32mI (4527981) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:09:12:911] <0x1b>[0;32mI (4588961) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:10:13:741] <0x1b>[0;32mI (4649791) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:11:14:390] <0x1b>[0;32mI (4710431) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:12:15:370] <0x1b>[0;32mI (4771411) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:13:16:001] <0x1b>[0;32mI (4832051) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:14:16:120] <0x1b>[0;32mI (4892171) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:15:16:220] <0x1b>[0;32mI (4952271) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:16:16:751] <0x1b>[0;32mI (5012801) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:17:17:320] <0x1b>[0;32mI (5073371) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:18:17:350] <0x1b>[0;32mI (5133401) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:19:18:210] <0x1b>[0;32mI (5194261) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:20:18:220] <0x1b>[0;32mI (5254271) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:21:18:911] <0x1b>[0;32mI (5314961) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:22:19:530] <0x1b>[0;32mI (5375581) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[22:23:19:871] <0x1b>[0;32mI (5435921) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2<0x1b>[0m␍␊
[... 58 lines cut ...]
    if (xSemaphoreTake(sdcard_semaphore, portMAX_DELAY) != pdTRUE)
    {
        ESP_LOGE(REST_TAG, "Failed to take semaphore");
        return ESP_FAIL;
    } else {
        httpd_resp_sendstr(req, settings_json);
        xSemaphoreGive(sdcard_semaphore);
    }
```

Where on the logging side it looks like:

```

    if (xSemaphoreTake(sdcard_semaphore, 600 / portTICK_PERIOD_MS) != pdTRUE) 
    {
        // did not get the sdcard write semaphore. Something really wrong!
        ESP_LOGE(TAG_LOG, "Error getting sdcard semaphore");
        goto error;
    }
    
... rest of code processing and writing to sd card using fputs
```

I am already looking at this for more than a week now and I am feeling a little lost at this point. Any help would be great!

paulusTecnion, 2023-05-07:
Could this be related to something like this? https://stackoverflow.com/questions/71921533/esp32-hardware-isr-sometimes-not-triggered-when-wifi-is-transmitting  Although there is no such power management option in v5.0.1

pacucha42, 2023-05-07:
Hi @paulusTecnion, thank you for the report. I'm sorry to hear you have been having troubles with ESP platform, however, your environment seems to be quite complex and it's hard to trace the root cause from the description provided. Could you please share verbose logs to let us see more details? You can set the log verbosity per logging component, to avoid overflooding by detailed log message from all the IDF components (use esp_log_level_set() API, more info on https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html)

At the first glance, I'd suspect the SD card(s) in use, as all the errors 0x107, 0x108 and 0x109 can indicate the card's malfunction (especially 0x109 - invalid CRC). Have you tried replacing your SDs?

Regarding the power management question - yes, it could be the root issue as well.

idea--list, 2023-05-07:
I remember having occasional 0x107 and 0x109 errors while trying to communicate with SD cards over SPI. In my case i was wiring up the SD slot on a breadboard using 15 cm long dupont wires... have been also enabling the auto light sleep feature of FreeRTOS. After changing to a board that has a properly mounted sd slot and also configuring gpio pads not to be powered off during sleep, my issues were solved. Not sure though if any of these would apply to your use case.

pacucha42, 2023-05-08:
> I remember having occasional 0x107 and 0x109 errors while trying to communicate with SD cards over SPI. In my case i was wiring up the SD slot on a breadboard using 15 cm long dupont wires... have [...]

Yes, too long wires are general issue due to a signal distortion. But I don't expect it in a production application

paulusTecnion, 2023-05-12:
Thanks for your responses! Indeed we made a custom PCB so wire length isn't an issue. However, it looks like the issues have been resolved, but I'm not sure what was the cause. It looks like it was some kind of timing issues. What I did:

- Disable GDB stub on panic
- Compiler: optimize for performance for the ESP32 and U1
- Changed the SPI input delay for the U1
- Set SPI mode for internal flash to 4 bit mode

After this I am already logging for days without issues. I will revert these things one by one to see what was the exact cause and will post the results here. So far I tried to only put the compiler optimization for ESP32 to Debug, but it still remains logging well. Will keep you posted.

paulusTecnion, 2023-05-12:
What is also interesting to see is that when I put a semaphore around the writing to sd card section (the semaphore I don't use anywhere else) this saves me execution of time of around 20-30 ms for the writing part. Anyone an idea why this is the case? Seems like the compiler gives this then priority when using semaphores even though I am not using it anywhere else..

paulusTecnion, 2023-06-16:
Coming back to this item. There were quite some things wrong in my code. I was calling sometimes SD card or other peripheral code through the HTTP endpoints who's task priority is lower. When the SD card was writing  in the other task and waiting for it's result at the same time, this could lead to an occasional conflict leading to SD card errors due to timing issues. I rewrote my whole code that the HTTP endpoints would only poll or push for tasks to retrieve or store data, while using the other task to deal with these requests. Probably basic software architecture stuff, but I'm also learning FreeRTOS still :-) 

Thanks for the help all.

adokitkat, 2023-08-14:
Hello! Do you still experience problems or were they resolved?