espressif/esp-idf #11343: ESP32S3 Temperature sensor eFuse factory calibration status

kutukvpavel, 2023-05-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've been trying to get internal temp sensor working on ESP32S3 on IDF v4.4.4 (latest bugfix for the time being) and I'm stumped.

<details>
  <summary>Here's my code:</summary>

```c++
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "driver/temp_sensor.h"
#include "esp_check.h"
#include "esp_log.h"

#include <math.h>

static const char TAG[] = "TEMP_SENS";

namespace my_temp_sensor
{
    static SemaphoreHandle_t chip_sensor_mutex = NULL;
    static esp_err_t chip_sensor_initialized = ESP_FAIL;

    esp_err_t init()
    {
        assert(!chip_sensor_mutex); //One-time init

        chip_sensor_mutex = xSemaphoreCreateMutex();
        assert(chip_sensor_mutex);
        temp_sensor_config_t temp_sensor_cfg = TSENS_CONFIG_DEFAULT();
        temp_sensor_get_config(&temp_sensor_cfg);
        temp_sensor_cfg.dac_offset = TSENS_DAC_L2; // DEFAULT: range:-10℃ ~  80℃, error < 1℃.
        ESP_RETURN_ON_ERROR(temp_sensor_set_config(temp_sensor_cfg), TAG, "On-chip sensor config failed");
        chip_sensor_initialized = temp_sensor_start();
        ESP_LOGI(TAG, "Permanently activating temp sensor (desktop build)");
        ESP_LOGI(TAG, "Temp sensor init result: %s", esp_err_to_name(chip_sensor_initialized));
        return chip_sensor_initialized;
    }

    float get_chip_temp()
    {
        assert(chip_sensor_mutex);
        assert(chip_sensor_initialized == ESP_OK);

        if (xSemaphoreTake(chip_sensor_mutex, pdMS_TO_TICKS(5)) != pdTRUE)
        {
            ESP_LOGE(TAG, "Cross-thread access of the on-chip temperature sensor failed!");
            return NAN;
        }
        float tsens_out = NAN;
        esp_err_t ret = ESP_OK;

            uint32_t raw = 0;
            temp_sensor_read_raw(&raw);
            ESP_LOGD(TAG, "RAW chip temp sensor value: %u", raw);
            ret = temp_sensor_read_celsius(&tsens_out);
            ESP_ERROR_CHECK_WITHOUT_ABORT(ret);

        xSemaphoreGive(chip_sensor_mutex);
        return tsens_out;
    }
}
```

</details>

In brief, it's a threadsafe version of the IDF example. All core mechanics are exactly the same.

**The problem** is, this code consistently spits out 100+ Celsius and, consequently, complains about the value being out of range. The ESP32S3 I'm using is inside a WROOM module that is sitting on a regular PCB, and I can easily keep my finger on the shield as long as I want, and even taking thermal resistance into account, there's no way the chip is at 100+C.

So I started investigating. First, I noticed that (at first glance) IDF v5.0 introduced significant changes into temp sensor driver code. Porting my project to v5.0 is going to be a major pain, so I try to avoid it for the time being. Fortunately, upon closer examination I found that core calibration math hasn't changed that much.

<details>
  <summary>Here's v4.4.4 math:</summary>

```c++
static float parse_temp_sensor_raw_value(uint32_t tsens_raw, const int dac_offset)
{
    if (isnan(s_deltaT)) { //suggests that the value is not initialized
        read_delta_t_from_efuse();
    }
    float result = (TSENS_ADC_FACTOR * (float)tsens_raw - TSENS_DAC_FACTOR * dac_offset - TSENS_SYS_OFFSET) - s_deltaT;
    return result;
}
```

</details>

<details>
  <summary>And here's v5.0 math:</summary>

```c++
static float parse_temp_sensor_raw_value(uint32_t tsens_raw, const int dac_offset)
{
    if (isnan(s_deltaT)) { //suggests that the value is not initialized
        read_delta_t_from_efuse();
    }
    float result = (TEMPERATURE_SENSOR_LL_ADC_FACTOR * (float)tsens_raw - TEMPERATURE_SENSOR_LL_DAC_FACTOR * dac_offset - TEMPERATURE_SENSOR_LL_OFFSET_FACTOR) - s_deltaT / 10.0;
    return result;
}
```

</details>

I verified that constants are indeed the same for both versions, and the equation seems to make sense. I also verified with a hot air gun that the values are changing with actual chip temperature, the scaling seems right, so the sensor is operational. The only suspicious part left is `s_deltaT` variable. It's read from an eFuse and constitutes a factory calibration offset. In my case, it's equal to `-67.0` (notice ). Let's do the calculation by hand (134 is the raw temp sensor value from debug log): `134×0.4386−27.88×0−20.52−(−67) = 105.2 C`. Indeed we get 100+C, so I don't think we have a software bug on our hands.

First thing that stand out is, v5.0 introduced division of this parameter by 10. Why is that? Did the factory calibration procedure change at some point, or something? Or did someone forget to implement this division during initial coding for in v4? Anyway, if we apply this division to the last calculation, we get around 45 degrees C, which is a bit too much in my opinion (my thermocouple meter shows exactly 38 degrees C for the shield and the PCB of the WROOM module, after I give them a couple minutes for the temperatures to equalize). Okay, even if 45C makes sense if we account for die-to-package thermal resistance, I don't see how the v4 code worked for some people if /10 stuff was just a coding oversight.

Notably, if we set `s_deltaT` to 0, we get around 38 degrees C, which is spot on. This begs the question. How come we have a factory-burnt eFuse with, what seems to be, a nonsensical value (at least for IDF v4 driver, for v5 /10 division it's just several degrees C out of spec)? Does the driver even check whether the eFuse is burnt or not (could this be in fact random garbage?)? Debug log has this to say, but I did not yet dive into eFuse documentation to make sense of it:
```
D (1395467) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (1395468) efuse: In EFUSE_BLK2__DATA4_REG is used 9 bits starting with 4 bit
```

And what's the best strategy to mitigate this issue? A could make a local patch right now and ignore this eFuse cal, but a local patch is a local patch, you have to remember to take it with you everywhere you go with this project, which is kinda inconvenient. And then comes the question whether all other ESP32S3-WROOM I buy next will be affected by this problem or is this going to be a random chance?

P.S. This issue may be related: #9617

mythbuster5, 2023-05-06:
Which version are you using? I remembered math on one version is wrong and we have already fixed it. The true functions is `float result = (TEMPERATURE_SENSOR_LL_ADC_FACTOR * (float)tsens_raw - TEMPERATURE_SENSOR_LL_DAC_FACTOR * dac_offset - TEMPERATURE_SENSOR_LL_OFFSET_FACTOR) - s_deltaT / 10.0;` Cloud you please check on your version?

kutukvpavel, 2023-05-06:
@mythbuster5 v4.4.4.
Okay, so v4.x raw to celsius conversion has a known bug? Looks like not too many people use temperature sensor, since I'm the first one to create an issue for it.
Only the last `/ 10` operation is missing from the formula for v4.4.4, everything else is the same, just refactored, so it's a simple fix. Since v4.4.4 is still maintained, should I make a pull request?

closed 2023-05-08 by kutukvpavel