espressif/esp-idf #11345: 添加zigbee AF层发送接收数据的例程以发送接收不定长数据

BanMian11, 2023-05-06:
### Is your feature request related to a problem?

zigbee没有自定义数据发送和接收的例程

### Describe the solution you'd like.

zigbee AF层发送接收数据的例程以发送接收不定长数据（长数据可以分包发送）

Alvin1Zhang, 2023-05-08:
Thanks for reporting.

BanMian11, 2023-05-09:
> @BanMian11 你是想要一些自定义的data, 通过APSDE service 发出去么

是的，在zigbee例程里边，我需要自定义data，data的长度可能会有点长，还要分包发送和接收

chshu, 2023-11-10:
@BanMian11 可以参考一下这里的 custom cluster example: https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_customized_devices

用户可以自定义应用层传输的数据类型和格式。

xieqinan, 2024-02-29:
@BanMian11 ，

最新的esp-zigbee-sdk 已经支持 AF层发送不定长的数据包的功能，相关的API可以参考：https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L118

closed 2024-03-08 by chshu