espressif/esp-idf #11348: Auto Download failed with USJ at ESP32-C3?

kaidegit, 2023-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-350-g56123c52aa 

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

合宙ESP32C3简约版（USB直连C3）

### Power Supply used.

USB

### What is the expected behavior?

auto download without press any key via usb

### What is the actual behavior?

cannot auto download. Even if force into bootloader by press boot and rst key, the `idf.py build flash monitor` will not let the chip go into normal mode, but still in bootloader and output these log:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x4 (DOWNLOAD(USB/UART0/1))
Saved PC:0x400462dc
waiting for download
```

### Steps to reproduce.

* open the blink example at pwsh
* long press boot and press rst, then release boot, to let the chip go into bl
* `idf.py set-target esp32c3`
* `idf.py build flash monitor`

### Debug Logs.

```plain
PS C:\Users\yekai\esp\blink> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-...rtition-table.bin C:/Users/yekai/esp/blink/build/blink.bin
blink.bin binary size 0x29190 bytes. Smallest app partition is 0x100000 bytes. 0xd6e70 bytes (84%) free.
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\boot...0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x5070 bytes. 0x2f90 bytes (37%) free.
Executing action: flash
Serial port COM15
Connecting...
Detecting chip type... ESP32-C3
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-idf\esptool_py && C:\Users\yekai\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe C:/Users/yekai/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/yekai/esp/blink/build/partition_table/partition-table.bin C:/Users/yekai/esp/blink/build/blink.bin"
blink.bin binary size 0x29190 bytes. Smallest app partition is 0x100000 bytes. 0xd6e70 bytes (84%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\bootloader\esp-idf\esptool_py && C:\Users\yekai\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe C:/Users/yekai/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x5070 bytes. 0x2f90 bytes (37%) free.
[2/3] cmd.exe /C "cd /D C:\Users\yekai\esp\esp-idf\components\esptool_py && C:\Users\yekai\.espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Users/yekai/esp/esp-idf -D SERIAL_TOOL=C:/Users/yekai/.espressif/python_env/idf5.1_py3.10_env/Scripts/python.exe;;C:/Users/yekai/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/yekai/esp/blink/build -P C:/Users/yekai/esp/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool esp32c3 -p COM15 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port COM15
Connecting...
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:79:b1:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00039fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20592 bytes to 12674...
Writing at 0x00000000... (100 %)
Wrote 20592 bytes (12674 compressed) at 0x00000000 in 0.2 seconds (effective 1045.9 kbit/s)...
Hash of data verified.
Compressed 168336 bytes to 89084...
Writing at 0x00010000... (16 %)
Writing at 0x0001a5e1... (33 %)
Writing at 0x00020f72... (50 %)
Writing at 0x00028443... (66 %)
Writing at 0x0002ed16... (83 %)
Writing at 0x00035eaa... (100 %)
Wrote 168336 bytes (89084 compressed) at 0x00010000 in 0.9 seconds (effective 1504.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 814.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory C:\Users\yekai\esp\blink
Executing "C:\Users\yekai\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe C:\Users\yekai\esp\esp-idf\tools/idf_monitor.py -p COM15 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace C:\Users\yekai\esp\blink\build\blink.elf --force-color -m 'C:\Users\yekai\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe' 'C:\Users\yekai\esp\esp-idf\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM15 instead...
--- idf_monitor on \\.\COM15 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x4 (DOWNLOAD(USB/UART0/1))
Saved PC:0x400462dc
waiting for download
```

### More Information.

OS:Windows 11 专业工作站版 22H2 22621.1555

kaidegit, 2023-05-07:
Here is the files that idf.py build output
[output_files.zip](https://github.com/espressif/esp-idf/files/11415175/output_files.zip)

igrr, 2023-05-07:
There is no way for software (e.g. esptool) to exit the ROM download mode once it has been entered due to strapping pin configuration. So the fact that the chip stays in download mode after flashing is unfortunately a limitation.

Do you have any issue with flashing using `idf.py build flash monitor ` without manually entering the download mode, though?

kaidegit, 2023-05-07:
thanks.
unfortunately,yes. And i seems find the problem. the chip that can auto download get the COM port lower than 10. after i change the port id to COM9 and replug the esp32c3, it can auto download.

kaidegit, 2023-05-07:
Here is some logs:
```
PS C:\Users\yekai\esp\blink> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-...rtition-table.bin C:/Users/yekai/esp/blink/build/blink.bin
blink.bin binary size 0x2e6b0 bytes. Smallest app partition is 0x100000 bytes. 0xd1950 bytes (82%) free.
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\boot...0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x5070 bytes. 0x2f90 bytes (37%) free.
Executing action: flash
Serial port COM14
Connecting....
COM14 failed to connect: Write timeout
Serial port COM1
Connecting.......................
COM1 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
PS C:\Users\yekai\esp\blink> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-...rtition-table.bin C:/Users/yekai/esp/blink/build/blink.bin
blink.bin binary size 0x2e6b0 bytes. Smallest app partition is 0x100000 bytes. 0xd1950 bytes (82%) free.
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\boot...0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x5070 bytes. 0x2f90 bytes (37%) free.
Executing action: flash
Serial port COM9
Connecting...
Detecting chip type... ESP32-C3
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-idf\esptool_py && C:\Users\yekai\.espressif\python_env\idf5.2_py3.10_env\Scripts\python.exe C:/Users/yekai/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/yekai/esp/blink/build/partition_table/partition-table.bin C:/Users/yekai/esp/blink/build/blink.bin"
blink.bin binary size 0x2e6b0 bytes. Smallest app partition is 0x100000 bytes. 0xd1950 bytes (82%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\bootloader\esp-idf\esptool_py && C:\Users\yekai\.espressif\python_env\idf5.2_py3.10_env\Scripts\python.exe C:/Users/yekai/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x5070 bytes. 0x2f90 bytes (37%) free.
[2/3] cmd.exe /C "cd /D C:\Users\yekai\esp\esp-idf\components\esptool_py && C:\Users\yekai\.espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Users/yekai/esp/esp-idf -D SERIAL_TOOL=C:/Users/yekai/.espressif/python_env/idf5.2_py3.10_env/Scripts/python.exe;;C:/Users/yekai/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/yekai/esp/blink/build -P C:/Users/yekai/esp/esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool esp32c3 -p COM9 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port COM9
Connecting...
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: a0:76:4e:ac:d8:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0003efff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20592 bytes to 12674...
Writing at 0x00000000... (100 %)
Wrote 20592 bytes (12674 compressed) at 0x00000000 in 0.4 seconds (effective 389.7 kbit/s)...
Hash of data verified.
Compressed 190128 bytes to 99171...
Writing at 0x00010000... (14 %)
Writing at 0x0001b223... (28 %)
Writing at 0x0002167a... (42 %)
Writing at 0x00028c65... (57 %)
Writing at 0x0002fc54... (71 %)
Writing at 0x00036c3b... (85 %)
Writing at 0x0003df77... (100 %)
Wrote 190128 bytes (99171 compressed) at 0x00010000 in 1.9 seconds (effective 793.9 kbit/s)...
Hash of data verified.
[... 89 lines cut ...]
  File "C:\Softwares\Python310\lib\subprocess.py", line 1209, in wait
    return self._wait(timeout=timeout)
  File "C:\Softwares\Python310\lib\subprocess.py", line 1490, in _wait
    result = _winapi.WaitForSingleObject(self._handle,
KeyboardInterrupt
idf_monitor failed with exit code 3221225786, output of the command is in the C:\Users\yekai\esp\blink\build\log\idf_py_stderr_output_12024 and C:\Users\yekai\esp\blink\build\log\idf_py_stdout_output_12024
PS C:\Users\yekai\esp\blink> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory C:\Users\yekai\esp\blink\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\esp-...rtition-table.bin C:/Users/yekai/esp/blink/build/blink.bin
blink.bin binary size 0x2e6b0 bytes. Smallest app partition is 0x100000 bytes. 0xd1950 bytes (82%) free.
[1/1] cmd.exe /C "cd /D C:\Users\yekai\esp\blink\build\boot...0 C:/Users/yekai/esp/blink/build/bootloader/bootloader.bin
[... cut ...] 070 bytes. 0x2f90 bytes (37%) free.
Executing action: flash
Serial port COM10
Connecting....
COM10 failed to connect: Write timeout
Serial port COM1
Connecting.......................
COM1 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
PS C:\Users\yekai\esp\blink>
```

dobairoland, 2023-05-09:
>  the chip that can auto download get the COM port lower than 10

Something similar is mentioned here: https://github.com/espressif/esptool/pull/880

closed 2023-05-10 by kaidegit

radimkarnis, 2023-05-29:
The fix is included in the [latest esptool release - v4.6](https://github.com/espressif/esptool/releases/tag/v4.6) and should propagate into ESP-IDF automatically. Thanks!