espressif/esp-idf #11350: periodic esp_timer callback misses to get triggered occasionally when trying to establish a TLS session

idea--list, 2023-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release V5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

AdafruitESP32S3 Feather

### Power Supply used.

USB

### What is the expected behavior?

I expect the callback function of a periodic esp_timer started by `esp_timer_start_periodic()` always being executed.
Because i can not make GPIO interrupts bullet-proof on long runs (not even with the proposed fix the bug described in 3.14 in [ESP32 errata](chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://www.espressif.com/sites/default/files/documentation/esp32_errata_en.pdf)), i went using a periodic timer. This way i am relying on timer_task running on core 0 with a priority of 22 to read out sensor data.

### What is the actual behavior?

Was pretty much happy with the reliability of the periodic timer until i added mqtt communication over tls (more specifically i integrated the latest release of [esp-aws-iot](https://github.com/espressif/esp-aws-iot) repo to my project).
Ever since adding that i notice the callback function sometimes simply does not get triggered, thus i am losing sensor data in a random pattern, but at least once every hour. Was not expecting this as i call the function containing the tls logic inside app_main task, which runs on core 0 at a priority of 1, so i wonder why it lets timer_task with a priority of 22 miss calling its callback.

Anyway i found if i  modify ssl_tls.c file of mbedtls component by including freertos/FreeRTOS.h, freertos/task.h and adding a vTaskDelay inside mbedtls_ssl_handshake() function around line 3095, then the timer becomes reliable again. I am just not feeling comfortable modifying any ESP-IDF core files. I wonder if there is a nicer workaround for this or this is a bug in mbedtls component

### Steps to reproduce.

1. enable auto light sleep feature of FreeRTOS
2. setup a periodic esp_timer to invoke a callback function that will read sensor data
3. inside app_main task's while loop call esp_timer_get_time() to store current time
4. if for example 1 minute passed by, call a function that will connect to an MQTT broker over a TLS session, subscribe, publish, unsubscribe and disconnect from TLS.
5. Let 3. and 4. run for about an hour as the timer's callback is missed only occasionally, not constantly.

idea--list, 2023-05-09:
Converted the function i was periodically calling by the esp_timer to a task now containing vTaskDelay. It does the same and seems to be more reliable.

Before doing so i also tried to enable CONFIG_ESP_TIMER_SUPPORTS_ISR_DISPATCH_METHOD in menuconfig, set dispatch method when configuring `esp_timer_create_args_t` and finally use `xTaskResumeFromISR()` inside the timer_callback function. All that did not make much difference. Was expecting a periodic HW timer more robust/reliable, but anyways there is a workaround.

KonstantinKondrashov, 2023-05-10:
Hi @idea--list!
I believe that an esp_timer callback just came later not missed. Core 0 is used by esp_timer by default. Probably when you enable some more tasks they are running on core 0 too (the core becomes busier) and it might delay esp_timer callbacks (a little bit postponed). To workaround this case I can suggest trying to use this option [ESP_TIMER_ISR_AFFINITY](https://github.com/espressif/esp-idf/blob/17451f1fb3d6485af5113d96b6836a7ce60efee8/components/esp_timer/Kconfig#L77-L105).

idea--list, 2023-05-10:
Hi @KonstantinKondrashov 
Thanks for the tip. I think those might be new options in the master branch, at least i can not find them in the 5.0.1 release i am working with.

KonstantinKondrashov, 2023-05-10:
@idea--list 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#handling-callbacks
"... If the timer is late then the callback will be called as soon as possible, it will not be lost. ..."
The callback must not be lost just to be a little bit late. The delay might happen in the next cases:
- esp_timer ISR doesn't run because a higher-priority ISR already runs, or because interrupts are disabled by a critical section
- esp_timer ISR runs, but esp_timer task doesn't run because there is a higher priority task running, or a flash operation is in progress
- esp_timer task runs, but another esp_timer callback is being executed for a long time

Make sure that you do not use `skip_unhandled_events` option (Timers with skip_unhandled_events option will not wake up the system.).

If you think that it is related to mbedtls, then you can try to use the software implementation (of AES MPI SHA) instead of hardware. It might call some critical sections that can delay callbacks.

(Top) → Component config → mbedTLS
[ ] Enable hardware AES acceleration
[ ] Enable hardware MPI (bignum) acceleration
[ ] Enable hardware SHA acceleration

idea--list, 2023-05-10:
@KonstantinKondrashov 
Thanks again for the replies, will use the task based workaround for now as that way the job is scheduled in a more monotonic pace.

closed 2023-05-10 by idea--list