espressif/esp-idf #11352: no more processes (EAGAIN) during https file download

ClaesIvarsson, 2023-05-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Successful file download from https every time!

### What is the actual behavior?

Sometimes download is stopped with (most frequent)
E (26140) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes

Sometimes download is stopped with
E (8953) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700 (MBEDTLS_ERR_X509_CERT_VERIFY_FAILED)

Sometimes the download is working just fine and succeeds until the whole file is downloaded.

### Steps to reproduce.

File is located on custom server, is ~3.5MB, web server is using custom tls cert from Let's Encrypt

Sample code
```
esp_http_client_config_t config;
memset(&config, 0, sizeof(config));
config.url            = download_url.c_str();
config.event_handler = _http_event_handler;
config.crt_bundle_attach = esp_crt_bundle_attach;
config.buffer_size = config.buffer_size_tx = 1024;
esp_http_client_handle_t client = esp_http_client_init(&config);
esp_err_t result = esp_http_client_perform(client);
```

### Debug Logs.

```plain
# most frequent
I (32548) [ota, 340]: HTTP_EVENT_ON_DATA, len=1024 of 15094/3453952
E (37641) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
E (37641) [ota, 411]: download not complete: 16118 != 3453952, errno: 11 (No more processes)
E (37651) [ota, 470]: Failed to download firmware

# sometimes
E (8953) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
E (8954) esp-tls: Failed to open new connection
E (8954) TRANSPORT_BASE: Failed to open a new connection
E (8962) HTTP_CLIENT: Connection failed, sock < 0
E (8962) [ota, 328]: HTTP_EVENT_ERROR
E (8962) [ota, 419]: Error perform http request ESP_ERR_HTTP_CONNECT
W (8966) [ota, 365]: Last esp error code: 0x801a
W (8968) [ota, 366]: Last mbedtls failure: 0x2700
E (8974) [ota, 470]: Failed to download firmware
```

### More Information.

* Custom board using ESP32-WROVER-IE
* root  node in mesh network
* using Ethernet to access internet
* dual core
* using mbetls
* not able to test on HEAD of branch due to #10755
* checked issue #4394 and #4241 without coming any closer

I have tried to check where in the code (esp-idf source) errno is set to EAGAIN and have found a couple of places. None of the places seems to be the source for this problem as I have added debug prints on those locations.

I have tried to use the following additional settings (individually and together) without any success:
config.timeout_ms = 30000;
config.keep_alive_enable = true;
config.skip_cert_common_name_check = true;

I have also added a retry (3 times, ) of the download process and the result is that sometimes it's successful on the second and third attempt.

ClaesIvarsson, 2023-05-09:
Found #4127, which seems to be somewhat relevant in the description.
In both cases problem is not consistent, resulting in different error message even though server is the same every time.

ClaesIvarsson, 2023-05-16:
I got this log:
```
E (106697) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
W (106697) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11 
E (110991) TRANSPORT_BASE: poll_read select error 104, errno = Connection reset by peer, fd = 56
W (110992) [ota, 393]: <0> Read data from http stream, errno: 0
```
I tried to print log from all occasions where return value is set to -26880 (MBEDTLS_ERR_SSL_WANT_READ). I didn't hit any of the places where this return value was hit.  Is it set somewhere in lower levels where functions are provided in library code ?

Any help tracing where in the code this happens and why it happens is much appreciated.

hmalpani, 2023-05-17:
Hello @ClaesIvarsson 
I tried reproducing the issue but was unable to recreate it. Can you share some sample code/application which can be used to reproduce the issue?
Thanks!

ClaesIvarsson, 2023-05-17:
I think I have found a change that solves the issue, I am currently trying to test and confirm.
sdkconfig has been changed like this:
CONFIG_LWIP_TCP_SYNMAXRTX=6 to 12
CONFIG_LWIP_TCP_MSS=624 to 1460
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534 to 5840
CONFIG_LWIP_TCP_WND_DEFAULT=131000 to 5840
CONFIG_LWIP_WND_SCALE=y to not set
CONFIG_LWIP_TCP_RCV_SCALE=1 to not set
I do not recall why the original value was set but I think know it is closer to default values.
Not sure which parameter of these actually helps, but this is as close as I have come so far.

hmalpani, 2023-05-23:
Hello @ClaesIvarsson 
I tried again to recreate the issue with the config options you were using as mentioned in the comment above. I am still not able to reproduce the issue.

Does changing the config helped you in resolving the issue?

ClaesIvarsson, 2023-05-23:
> Hello @ClaesIvarsson I tried again to recreate the issue with the config options you were using as mentioned in the comment above. I am still not able to reproduce the issue.
> 
> Does changing the config helped you in resolving the issue?

I have done numerous tests with the changes mentioned in [comment above](https://github.com/espressif/esp-idf/issues/11352#issuecomment-1551274007) and I have not been able to repeat the problem so far.
I will go for this solution and will not investigate it further on my side.

ClaesIvarsson, 2023-05-23:
Just realized where CONFIG_LWIP_TCP_MSS=624 came from in sdkconfig
it is the recommendation in the [ip_internal_network](https://github.com/espressif/esp-idf/blob/v4.4.3/examples/mesh/ip_internal_network/sdkconfig.defaults) example.
CONFIG_LWIP_TCP_MSS=1460 breaks the NAT functionality of the ip_internal_network, hence need to find an alternative value for CONFIG_LWIP_TCP_MSS where both problems are solved.
Is there a maximum value for CONFIG_LWIP_TCP_MSS which still not break NAT ?
I am currently running 1248 (2*624) and it seems to be working for both. Test in progress ...

ClaesIvarsson, 2023-05-30:
Ended up using CONFIG_LWIP_TCP_MSS=1400.
Case closed from my perspective.

closed 2023-05-30 by ClaesIvarsson