espressif/esp-idf #11353: Require option on web page to change IP, gateway, sabnet, dns, ssid and password

kapilt007, 2023-05-08:
### Is your feature request related to a problem?

no

### Describe the solution you'd like.

Dear Team,

I would like to receive solution for change IP after connecting Hotspot on web page (Request you to provide this option on web page to change IP)

For example 
My house router IP address is 192.168.1.1 and my ESP address is 192.168.4.1 

I want static ip and DHCP option to enter static IP, GATEWAY, subnet, DNS, SSID and password

mahavirj, 2023-06-07:
Please check out the example at https://github.com/espressif/esp-idf/tree/master/examples/protocols/static_ip, it should help you in this regard. Web page related changes can be done using `./examples/protocols/http_server/file_serving` example as a reference.

Alvin1Zhang, 2023-06-16:
Thanks for reporting, feel free to reopen.

closed 2023-06-16 by Alvin1Zhang

kapilt007, 2023-06-16:
Dear Team,

Thanks for your help and email sent to me

I have gone through code which was attached in this email

There is no HTML code which is require on web page to change network
credentials as per the subject line