espressif/esp-idf #11357: A file in sdcard , it's name is limited in 8 chars, and the  extension is limited in 3 chars

yizi15, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

 ESP32-LyraT V4.3

### Power Supply used.

USB

### What is the expected behavior?

the file name and extension could be longer

### What is the actual behavior?

when the name is "/sdcard/test11111.txt" or   "/sdcard/test.txt1" , it will be crashed;

the code:
` esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    audio_board_sdcard_init(set, SD_MODE_1_LINE);
    dir_list("/sdcard");
    FILE* f = fopen("/sdcard/test.txt1", "w");
    fwrite("1", 2, 1, f);
    fclose(f);`

### Steps to reproduce.

the code:
` esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    audio_board_sdcard_init(set, SD_MODE_1_LINE);
    dir_list("/sdcard");
    FILE* f = fopen("/sdcard/test.txt1", "w");

### Debug Logs.

```plain
log is:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400f6d55  PS      : 0x00060f30  A0      : 0x800f6dd8  A1      : 0x3ffb5c00  
0x400f6d55: _fwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c:135 (discriminator 14)

A2      : 0x3ffb5ec4  A3      : 0x00000002  A4      : 0x00000002  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xff000000  A8      : 0x800f6d55  A9      : 0x3ffb5be0
A10     : 0x00000000  A11     : 0x3ffb5c14  A12     : 0x00000601  A13     : 0x000001b6
A14     : 0x3ffb5ec4  A15     : 0x3ffc2c7c  SAR     : 0x00000006  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000064  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400f6d52:0x3ffb5c000x400f6dd5:0x3ffb5c40 0x400d6796:0x3ffb5c60 0x4010555c:0x3ffb5db0 0x4008b59d:0x3ffb5dd0
0x400f6d52: _fwrite_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fwrite.c:135 (discriminator 14)

0x400f6dd5: fwrite at /builds/idf/crosstool-NG/.build/HOST-x86_64C-lwe6a4r-CmoimnmgEwr3r2o/rx tfeanislae-de s(pP3e2r-meilsfs/isorncE/rrnoerw(l1i3b,/ n'e?w?l?i?b?/?l?i?b?c?/?s?t?d?i?o'/,f wNrointee,. c5:)1)77

Wa
iting for the device to reconnect0x400d6796: app_main at D:/Espressif4/frameworks/esp-adf/examples/get-started/play_mp3_control/main/play_mp3_control_example.c:185

0x4010555c: main_task at D:/Espressif4/frameworks/esp-idf-v4.4.2/components/freertos/port/port_common.c:141 (discriminator 2)

0x4008b59d: vPortTaskWrapper at D:/Espressif4/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131
```

### More Information.

when i try to readdir of /sdcard, 
DIR: /sdcard
        SYSTEM~1
this is dir:/sdcard/SYSTEM~1
ignore this dir
        TEST.TST
        TEST1.TXT
        TEST11.TXT
        TIANFU~1.MP3
        TEST1111.TXT
        TEST.TXT
        YUKEWE~1.MP3
        GEM_PA~1.MP3
        TEST.TX
all the file name will be cap, and more than 8 chars will not appear

yizi15, 2023-05-09:
Debug Logs. is not useable, in fact, the file handle f is 0 , FILE* f = fopen("/sdcard/test.txt1", "w");

yizi15, 2023-05-09:
AT idf v4.4.2, th chip esp32s3 is repetition, 
the code:
`    esp_err_t ret;

    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
#ifdef CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .format_if_mount_failed = true,
#else
        .format_if_mount_failed = false,
#endif // EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc/sdspi_mount is all-in-one convenience functions.
    // Please check its source code and implement error recovery when developing
    // production applications.

    ESP_LOGI(TAG, "Using SDMMC peripheral");
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    // Set bus width to use:
    slot_config.width = 4;
    // On chips where the GPIOs used for SD card can be configured, set them in
    // the slot_config structure:
    slot_config.clk = CONFIG_EXAMPLE_PIN_CLK;
    slot_config.cmd = CONFIG_EXAMPLE_PIN_CMD;
    slot_config.d0 = CONFIG_EXAMPLE_PIN_D0;
    slot_config.d1 = CONFIG_EXAMPLE_PIN_D1;
    slot_config.d2 = CONFIG_EXAMPLE_PIN_D2;
    slot_config.d3 = CONFIG_EXAMPLE_PIN_D3;

    // Enable internal pullups on enabled pins. The internal pullups
    // are insufficient however, please make sure 10k external pullups are
    // connected on the bus. This is for debug / example purpose only.
    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdmmc_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                     "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
    sd_card_test();

dir_list(MOUNT_POINT);
    // Use POSIX and C standard library functions to work with files:

    // First create a file.
    const char *file_hello = MOUNT_POINT"/hello.txt1";

    ESP_LOGI(TAG, "Opening file %s", file_hello);
    FILE *f = fopen(file_hello, "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }`

the log:
I (697) example: Filesystem mounted
Name: SDU1
Type: SDHC/SDXC
Speed: 20 MHz
Size: 15200MB
I (703) example: Opening dir /sdcard
max name len 255
DIR: /sdcard
        SYSTEM~1
this is dir:/sdcard/SYSTEM~1
ignore this dir
        TEST.TST
        TEST1.TXT
        TEST11.TXT
        TIANFU~1.MP3
        TEST1111.TXT
        TEST.TXT
        YUKEWE~1.MP3
        GEM_PA~1.MP3
        TEST.TX
        FOO.TXT
I (726) example: Opening file /sdcard/hello.txt1
E (731) example: Failed to open file for writing

0xjakob, 2023-05-09:
Regarding the "capping": Looks like you are running into the [8.3 filename limitation](https://en.wikipedia.org/wiki/8.3_filename). The [FATFS IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/fatfs.html#fat-filesystem-support) mentions the following:

> The filesystem uses [8.3 filenames](https://en.wikipedia.org/wiki/8.3_filename) format (SFN) by default. If you need to use long filenames (LFN), enable the [CONFIG_FATFS_LONG_FILENAMES](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-fatfs-long-filenames) option.

Have you tried the `CONFIG_FATFS_LONG_FILENAMES` option?

yizi15, 2023-05-09:
thanks, i will CONFIG_FATFS_LONG_FILENAMES .

closed 2023-05-09 by yizi15