espressif/esp-idf #11358: UART_SELECT_WRITE_NOTIF should be trigger when the uart is writable

XinyuKhan, 2023-05-09:
### Is your feature request related to a problem?

I was using vfs_uart module to manage uart io, but I found the behaviour of uart fd is not same as which of a socket fd. I never receive a writable event of a uart fd when I selecting it. I found the UART_SELECT_WRITE_NOTIF event was never triggered in the vfs implementation. I want to know why? 

### Describe the solution you'd like.

I hope the behaviour of the selecting of a uart fd is as same as which of a socket fd.

XinyuKhan, 2023-05-11:
Hi, thx for you rely, I found this problem in one of my big projects, so I need some time to extract the uart relative code out if it's necessary. Before that, you can search the UART_SELECT_WRITE_NOTIF macro in the esp-idf repro in version release 5.0, then you can find this macro is never actually used in the code. The implement of uart vfs shows clearly that the writable event will never be triggered if UART_SELECT_WRITE_NOTIF macro is not used in the code. That is the problem

XinyuKhan, 2023-05-22:
Dear dobairoland,

Thank you for your insightful comment on the UART_SELECT_WRITE_NOTIF issue. I appreciate your explanation and the historical context you provided.

I would like to share my perspective on a couple of points you raised:

The behavior where starting a select call and waiting for UART to be writable immediately returns from the call is, in my opinion, more intuitive and consistent with the behavior of TCP sockets. This consistency can help avoid confusion for developers who are working with both UART and TCP sockets.

While I understand that "one can always write to UART" is a low-level detail, the behavior of select is a higher-level convention. I believe that these higher-level conventions should not be compromised due to low-level details. The abstraction provided by these conventions is crucial for developers to write efficient and maintainable code without needing to understand all the low-level details.

I hope these points can contribute to the ongoing discussion and look forward to hearing your thoughts.

Best regards,
XinyuKhan

XinyuKhan, 2023-05-22:
> This was implemented years ago and introduced in v3.1. Write notification was never fired. If I remember the reason well then the assumption was that one can always write to UART. Starting a select call and waiting for UART to be writable would immediately return from the call.
> 
> Nobody missed this over the years. I guess implementing this would require to define the use case and create an example where the implementation could be tested.

Dear dobairoland,

Thank you for your insightful comment on the UART_SELECT_WRITE_NOTIF issue. I appreciate your explanation and the historical context you provided.

I would like to share my perspective on a couple of points you raised:

The behavior where starting a select call and waiting for UART to be writable immediately returns from the call is, in my opinion, more intuitive and consistent with the behavior of TCP sockets. This consistency can help avoid confusion for developers who are working with both UART and TCP sockets.

While I understand that "one can always write to UART" is a low-level detail, the behavior of select is a higher-level convention. I believe that these higher-level conventions should not be compromised due to low-level details. The abstraction provided by these conventions is crucial for developers to write efficient and maintainable code without needing to understand all the low-level details.

I hope these points can contribute to the ongoing discussion and look forward to hearing your thoughts.

Best regards,
XinyuKhan

RathiSonika, 2023-07-19:
Duplicates: https://github.com/espressif/esp-idf/issues/10986
Closed with commit [571831b](https://github.com/espressif/esp-idf/commit/571831b2eeb1c11569eb265f26f75138a485460c)

closed 2023-07-19 by RathiSonika