espressif/esp-idf #11359: Bluetooth playback exception

maplerian, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3839-gafbdb0f3ef

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32 custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Long periods of time are normal

### What is the actual behavior?

The connection to the mobile phone is disconnected due to an exception during playback

### Steps to reproduce.

1.  ...... sdkconfig ......
```
#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
# CONFIG_BT_NIMBLE_ENABLED is not set
# CONFIG_BT_CONTROLLER_ONLY is not set
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set

#
# Bluedroid Options
#
CONFIG_BT_BTC_TASK_STACK_SIZE=3072
# CONFIG_BT_BLUEDROID_PINNED_TO_CORE_0 is not set
CONFIG_BT_BLUEDROID_PINNED_TO_CORE_1=y
CONFIG_BT_BLUEDROID_PINNED_TO_CORE=1
CONFIG_BT_BTU_TASK_STACK_SIZE=3072
# CONFIG_BT_BLUEDROID_MEM_DEBUG is not set
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_A2DP_ENABLE=y
# CONFIG_BT_SPP_ENABLED is not set
# CONFIG_BT_L2CAP_ENABLED is not set
# CONFIG_BT_HFP_ENABLE is not set
# CONFIG_BT_HID_ENABLED is not set
CONFIG_BT_SSP_ENABLED=y
# CONFIG_BT_BLE_ENABLED is not set
# CONFIG_BT_STACK_NO_LOG is not set
[... 240 lines cut ...]
I/Audio/HW         [10:40 593] Audio status change : AUDIO_STATE_TIME_POS(170883)
I/stream/bt        [10:42 513] packet count: 20200
I/stream/bt        [10:44 833] packet count: 20300
I/stream/bt        [10:47 163] packet count: 20400
E: No elt when frame end
FEX 11 11
E (658603) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
E (658603) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7
I/stream/bt        [10:58 613] bt_av_hdl_a2d_evt event: 1
I/stream/bt        [10:58 613] A2DP audio state: Stopped
I/stream/bt        [10:58 613] bt_av_hdl_a2d_evt event: 0
I/stream/bt        [10:58 613] A2DP connection state: Disconnected, [9C:28:F7:71:AE:9D]
I/Audio/HW         [10:58 613] Audio status change : AUDIO_STATE_STOP(0)
```

### More Information.

```
IDF Version : v5.1-dev-3839-gafbdb0f3ef-dirty
Chip info   :
        model     : ESP32
        cores     : 2
        feature   : /802.11bgn/BLE/BT/External-Flash:16 MB
        revision number: 300
```

xiongweichao, 2023-05-09:
Hi @maplerian ,

Sorry, I didn't reproduce this problem.  May I ask if this problem happens every time?

BTW, it is recommended to enable [High level interrupt](https://github.com/espressif/esp-idf/blob/17451f1fb3d6485af5113d96b6836a7ce60efee8/components/bt/controller/esp32/Kconfig.in#LL436C8-L436C21).

Thanks

maplerian, 2023-05-09:
> Hi @maplerian ,
>
> Sorry, I didn't reproduce this problem. May I ask if this problem happens every time?
>
> BTW, it is recommended to enable [High level [...]

Okay, I'll try the high-priority interrupt test for a day, see if it repeats
Thanks

maplerian, 2023-05-11:
Yes, after one day of testing, this method is feasible

closed 2023-05-11 by maplerian