espressif/esp-idf #11360: esp32-s3 uart0 rts pin doesn't work (always high level)

Wanderer-web, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf version : v4.4.3
chip : esp32-s3r8
flash : W25Q128JVSIQ (16MB)
I use the following code to initialize UART0 of esp32-s3. Enable RTS hardware flow control, baud rate 3000000:

```c
#define TXD_PIN (43)
#define RXD_PIN (44)
#define RTS_PIN (11)
const uart_config_t uart_config = {
        .baud_rate = 3000000,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_RTS,
        .source_clk = UART_SCLK_APB,
};
uart_driver_install(UART_NUM_0, 8320, 0, 0, NULL, 0);
uart_param_config(UART_NUM_0, &uart_config);
uart_set_pin(UART_NUM_0, TXD_PIN, RXD_PIN, RTS_PIN, UART_PIN_NO_CHANGE);
```

But the rts pin seemed not to work correctly, I used an oscilloscope to measure the pin and found that it is always high (normally it should be low), and when I send data to the esp32-s3 UART, the rts pin level does not change, it is always high.
(This rts pin is not connected to a pull-up resistor)

songruo, 2023-05-12:
Hi @Wanderer-web,

According to our Technical Reference Manual,

> In this mode, rtsn_out is pulled high when data in Rx_FIFO exceeds UART_RX_FLOW_THRHD.

Therefore, you need to configure `rx_flow_ctrl_thresh` in the `uart_config_t`. Otherwise, 0 is write to the `UART_RX_FLOW_THRHD` register, and RTS line will be always high.

Wanderer-web, 2023-05-13:
> Hi @Wanderer-web,
>
> According to our Technical Reference Manual,
>
>
> Therefore, you need to configure `rx_flow_ctrl_thresh` in the `uart_config_t`. Otherwise, 0 is write to the `UART_RX_FLOW_THRHD` [...]

Thank you! It works now.

closed 2023-05-15 by songruo