espressif/esp-idf #11362: [ESP32S3] v4.4 devices not waking up from deep sleep.

KonssnoK, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, as usual, we pick one feature from the idf, we try and implement it, and on first deploy it goes south.

We are currently trying to use deep_sleep on our esp32s3n8r8.

The way we go into deepsleep is as follows
```
    mesh_stop();
    esp_wifi_stop();
    esp_deep_sleep(sleep_time_sec * SECOND_MS * 1000);
```

What happens:
sometimes, still not reproducible when, the device does not wake up until a forced reset is done to the device.

Our RTC settings are:
[image]

any help?
what could we instrument to debug?
Thanks

KonssnoK, 2023-05-09:
@esp-lis  
additional information
the issue seems to trigger only when using the 8MHz oscillator,
using the 150KHz internal one does not trigger the issue.

cbaechler, 2023-05-10:
I have been debugging this issue and found, that increasing the `rtc_dbias_slp` setting seems to help. 

First of all, I was able to reproduce this issue with a minimal code: 
```c
#include "esp_log.h"
#include "esp_sleep.h"

const static char* TAG = "main";

void app_main(void)
{
    ESP_LOGI(TAG, "Go to deep sleep");
    esp_deep_sleep(1 * 1000 * 1000);
    ESP_LOGE(TAG, "FAILED to go to deep sleep");

    while (true) {
        asm("nop");
    }
}
```

Setting RTC clock source in sdkconfig to `Internal 8MHz oscillator, ...` made the program stuck on _one_ of my ESP32-S3-DevKitC, while on others it looped as expected. (when I set it to `Internal 150kHz RC oscillator` it always worked) 

Furthermore I had one device which in the morning (after it was switched off for >12 hours) initially was also not restarting from deep sleep. After some time however, it started to work fine (however, I found that sleep time was longer compared to other devices). My assumption here is that the device getting warmer helped to make it work at the defined `rtc_dbias_slp` setting (i.e. bias can be lower when temperature is higher). 

`rtc_sleep_get_default_config` defaults `rtc_dbias_slp` to `RTC_CNTL_DBIAS_1V10`.  Increasing the RTC bias voltage in `esp_sleep_start` to `RTC_CNTL_DBIAS_1V15` made all devices restarting from deep sleep, while lowering it to `RTC_CNTL_DBIAS_1V05` made all devices stuck. 

```diff
// Enter sleep
rtc_sleep_config_t config;
rtc_sleep_get_default_config(sleep_flags, &config);
+ config.rtc_dbias_slp = RTC_CNTL_DBIAS_1V15;
rtc_sleep_init(config);
```

I therefore wonder whether the default value for `rtc_dbias_slp` is too optimistic? Unfortunately I was not able to find related information in the technical documents of the chipset. 

FYI @esp-lis @KonssnoK

cbaechler, 2023-05-26:
@esp-lis Any update on this issue? Thank you!

AxelLin, 2023-06-04:
Might be related to https://github.com/espressif/esp-idf/issues/11363 https://github.com/espressif/esp-idf/issues/9455

10086loutianhao, 2023-06-05:
I will take a look at this issue

10086loutianhao, 2023-06-05:
@KonssnoK  hello, I saw that you mentioned that you have also used the latest master to verify this issue and still reported an error. Please provide the latest version of commit you are using. I feel that maybe some of our changes were not synchronized to Github in time, thank you very much.

AxelLin, 2023-06-06:
@10086loutianhao 
Could you share the commit of the fix and backport the fix to all impacted branches?

10086loutianhao, 2023-06-06:
@AxelLin  emm, I am currently uncertain about the specific commit, but I noticed a disparity between the content of the latest 'master' branch and the commit being used by the client.

10086loutianhao, 2023-06-07:
I have noticed that other colleagues have already performed backports (MR23754, MR23758）

KonssnoK, 2023-06-09:
> I see that the latest master branch has resolved this issue. Indeed, when using 8Md256, the improper PMU parameters caused abnormal counting in the lp_timer. The more appropriate parameters have been implemented on the latest master branch.

hello @10086loutianhao , this issue has nothing to do with wrong counting, but seems related to CPU voltage levels.
We will check on master, but i think those changes are unrelated

10086loutianhao, 2023-06-09:
@KonssnoK Yes, your intuition is correct. Insufficient voltage during the sleep period leads to abnormal behavior of the clock used for counting. The overall clock count remains unchanged, but the clock cycle is altered. This alteration can result in an increase in the clock cycle or even the disappearance of the clock altogether. As a result, it may cause the system to either fail to wake up or wake up after an extended period of time due to timeouts.

AxelLin, 2023-07-10:
> I have noticed that other colleagues have already performed backports (MR23754, MR23758）

Are these backports available in github? Could you share the commit id of the fix?

10086loutianhao, 2023-07-11:
> > I have noticed that other colleagues have already performed backports (MR23754, MR23758）
> 
> Are these backports available in github? Could you share the commit id of the fix?

I saw that the above two merge requests have been merged, but when I checked on GitHub, only v5.0 has been synchronized, and v4.0 is still pending synchronization.

AxelLin, 2023-07-11:
@10086loutianhao 

Can you share the commit id for the fix? (in any branch)
If I know the commit id of the fix, I can trace the backport status by myself.

10086loutianhao, 2023-07-11:
> @10086loutianhao
> 
> Can you share the commit id for the fix? (in any branch) If I know the commit id of the fix, I can trace the backport status by myself.

in release/v5.0, commit id = d82af7f54a1bf582b850ade14281046aa17f7a29
in release/v4.4, commit id = 066a1ac0acc1f6d7e371399757480f5b17efc22e

Thanks

AxelLin, 2023-07-14:
> hello @10086loutianhao , this issue has nothing to do with wrong counting, but seems related to CPU voltage levels. We will check on master, but i think those changes are unrelated

@KonssnoK
066a1ac is available in v4.4 branch.
Could you check if it fixed your issue?

closed 2023-11-07 by KonssnoK