espressif/esp-idf #11363: [ESP32S3] v4.4 deep_sleep sleeping more than should

KonssnoK, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

issue is quite simple:
- we put 30 minutes sleep, device sleeps 40 minutes
- we put 1 hour sleep, device sleeps 1.5hours
- we put 1.5 hours sleep, device sleeps 3hours

Any idea on how to improve accuracy?

our RTC settings:
[image]

10086loutianhao, 2023-06-05:
I will take a look at this issue

10086loutianhao, 2023-06-06:
I see that the latest master branch has resolved this issue. Indeed, when using 8Md256, the improper PMU parameters caused abnormal counting in the lp_timer. The more appropriate parameters have been implemented on the latest master branch. (The cause of the malfunction is the same as in GH#11362)

10086loutianhao, 2023-11-07:
> @10086loutianhao was this merged in v4.4?
Yes, the solution here should be the same as the one in https://github.com/espressif/esp-idf/issues/11362. Please check it.

closed 2023-11-07 by KonssnoK