espressif/esp-idf #11364: I want to modify the sdkconfig configuration by automatically using a custom script (e.g. python) before compiling, how do I modify the cmake implementation? I tried to use add_custom_command PRE_BUILD stage, but it was still relatively backward

Dozingfiretruck, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to modify the sdkconfig configuration by automatically using a custom script (e.g. python) before compiling, how do I modify the cmake implementation? I tried to use add_custom_command PRE_BUILD stage, but it was still relatively backward

Alvin1Zhang, 2023-05-10:
Thanks for reporting, would you please provide details per the [issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

Dozingfiretruck, 2023-05-10:
> Thanks for reporting, would you please provide details per the [issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

I use idf5,only want to invoke custom scripts before compiling