espressif/esp-idf #11365: 蓝牙广播与蓝牙mesh 广播可以共存吗？

2815918642, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

为什么同时开启蓝牙 GATTS 服务与 蓝牙mesh 后无法找到蓝牙广播 只能搜索到mesh广播？ 关闭蓝牙mesh功能或者在使能mesh功能之前 蓝牙广播正常出现。使用的抓包软件：nRF connect 与EspBLEMesh

forx157, 2023-05-10:
Hi, @2815918642 ,

forx157, 2023-05-10:
Hi, @2815918642 ,You can try this API: esp_ble_mesh_start_ble_advertising.

2815918642, 2023-05-10:
> 嗨， 你可以试试这个API：esp_ble_mesh_start_ble_advertising。

好的 感谢

closed 2023-05-11 by 2815918642