espressif/esp-idf #11367: esp_hidd_dev_battery_set hangs forever if indicate not enabled

finger563, 2023-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-4665-gef64e4e5b3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

QtPy ESP32 PICO

### Power Supply used.

USB

### What is the expected behavior?

When calling `esp_hidd_dev_battery_set( ble_hid_dev, battery_level );`, the function should return instead of blocking forever.

### What is the actual behavior?

If the device (tested an iPhone 14 Plus and a Google Pixel 6) does not enable indication for the battery service, then calling esp_hidd_dev_battery_set will block forever (because the wait on the semaphore as part of `WAIT_CB(dev);` will never return).

Suggest wrapping the `WAIT_CB(dev);` within a check on `dev->bat_cc.indicate_eable == 1`.

### Steps to reproduce.

1. Pair and connect BLE HID demo to a device
2. Call `esp_hidd_dev_battery_set` after connection is established
3. See hang

zhp0406, 2023-05-14:
I have investigated the problem you reported and confirmed that the bug exists. I am currently working on resolving it as soon as possible.

Based on the "Battery Service Specification v1.1", the property of the "Battery Level" characteristic is "Notify". Therefore, the following update is proposed to solve this problem:

`static esp_err_t esp_ble_hidd_dev_battery_set(void *devp, uint8_t level)
{
    esp_err_t ret;
    esp_ble_hidd_dev_t *dev = (esp_ble_hidd_dev_t *)devp;
    if (!dev || s_dev != dev) {
        return ESP_FAIL;
    }
    dev->bat_level = level;

    if (!dev->connected || dev->bat_ccc.value == 0) {
        //if we are not yet connected, that is not an error
        return ESP_OK;
    }
    if(dev->bat_ccc.notify_enable){
        ret = esp_ble_gatts_send_indicate(dev->bat_svc.gatt_if, dev->conn_id, dev->bat_level_handle, 1, &dev->bat_level, false);
        if (ret) {
            ESP_LOGE(TAG, "esp_ble_gatts_send_notify failed: %d", ret);
            return ESP_FAIL;
        }
    }
    return ESP_OK;
}`

Alvin1Zhang, 2023-07-19:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2023-07-19 by Alvin1Zhang

finger563, 2023-07-19:
@Alvin1Zhang does that mean the fix will be in the next release?

Alvin1Zhang, 2023-07-19:
@finger563 Thanks for the updates, the fix on master branch has been available a while ago, https://github.com/espressif/esp-idf/commit/2e9669a814b3d5190df58bff61e8de2c7b875719 and fix on release/5.1 is at https://github.com/espressif/esp-idf/commit/56bb50b85e6ded2f04f045f3741c6cabde554d11. Also the fix has been back ported to other release branches. Thanks.