espressif/esp-idf #11373: dhcpserver.c `LWIP_HOOK_DHCPS_POST_APPEND_OPTS` invocation contains incorrect struct member

wuyuanyi135, 2023-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

N/A

### What is the actual behavior?

When trying to implement the hook as mentioned in https://github.com/espressif/esp-idf/issues/11372

The following line contains an error about the dereference to member `netif`
https://github.com/espressif/esp-idf/blob/17451f1fb3d6485af5113d96b6836a7ce60efee8/components/lwip/apps/dhcpserver/dhcpserver.c#L562

which should be correct to `LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->dhcps_netif, dhcps, DHCPOFFER, &end)`

There are three invocations that should be corrected.

In summary,

```diff
diff --git a/components/lwip/apps/dhcpserver/dhcpserver.c b/components/lwip/apps/dhcpserver/dhcpserver.c
index 4a65f86..ce7d39b 100644
--- a/components/lwip/apps/dhcpserver/dhcpserver.c
+++ b/components/lwip/apps/dhcpserver/dhcpserver.c
@@ -546,7 +557,7 @@ static void send_offer(dhcps_t *dhcps, struct dhcps_msg *m, u16_t len)

     end = add_msg_type(&m->options[4], DHCPOFFER);
     end = add_offer_options(dhcps, end);
-    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->netif, dhcps, DHCPOFFER, &end)
+    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->dhcps_netif, dhcps, DHCPOFFER, &end)
     end = add_end(end);

     p = dhcps_pbuf_alloc(len);
@@ -624,7 +635,7 @@ static void send_nak(dhcps_t *dhcps, struct dhcps_msg *m, u16_t len)
     create_msg(dhcps, m);

     end = add_msg_type(&m->options[4], DHCPNAK);
-    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->netif, dhcps, DHCPNAK, &end)
+    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->dhcps_netif, dhcps, DHCPNAK, &end)
     end = add_end(end);

     p = dhcps_pbuf_alloc(len);
@@ -701,7 +712,7 @@ static void send_ack(dhcps_t *dhcps, struct dhcps_msg *m, u16_t len)

     end = add_msg_type(&m->options[4], DHCPACK);
     end = add_offer_options(dhcps, end);
-    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->netif, dhcps, DHCPACK, &end)
+    LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->dhcps_netif, dhcps, DHCPACK, &end)
     end = add_end(end);

     p = dhcps_pbuf_alloc(len);
```

This compilation error exists in both v5.0.1 and master branch.

### Steps to reproduce.

-

### Build or installation Logs.

```plain
C:/Espressif/frameworks/esp-idf-v5.0.1/components/lwip/apps/dhcpserver/dhcpserver.c:560:43: error: 'dhcps_t' has no member named 'netif'
  560 |     LWIP_HOOK_DHCPS_POST_APPEND_OPTS(dhcps->netif, dhcps, DHCPOFFER, &end)
```
```

wuyuanyi135, 2023-05-16:
Anyone takes care of this issue?

david-cermak, 2023-05-17:
Thanks for reporting this bug and fixing it! If you feel like it you can open a PR with your correction in https://github.com/espressif/esp-idf/pulls

AxelLin, 2023-05-17:
@david-cermak 
FYI, https://github.com/espressif/esp-idf/pull/11374

closed 2023-06-14 by espressif-bot