espressif/esp-idf #11375: TASK_WDT triggered by wifi task causes crash

sxsx2yzyz, 2023-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-384-g3047225a9f

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-wrover-e

### Power Supply used.

External 5V

### What is the expected behavior?

Runs normally without crashing

### What is the actual behavior?

We have 10 sets of products with Wi-Fi and Bluetooth on at the same time, after running for about 20 hours, the first chip crashed, and after another 10 hours, the second crashed, and after 5 hours, the third had the same problem, we analyzed the coredump of the 3 chips, and they were all the same

### Steps to reproduce.

<details><summary>coredump</summary>
```
bash-3.2$ espcoredump.py info_corefile -t raw -c ./error88 ./build/led_ble_wifi.elf
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================
Crashed task handle: 0x3ffe8e68, name: 'wifi', GDB name: 'process 1073647208'
================== CURRENT THREAD REGISTERS ===================
exccause 0x1d (StoreProhibitedCause)
excvaddr 0x0
epc1 0x4013a8bf
epc2 0x0
epc3 0x4000bff0
epc4 0x4000921f
epc5 0x4000921a
epc6 0x0
eps2 0x0
eps3 0x60021
eps4 0x60023
eps5 0x60023
eps6 0x0
pc 0x40082250 0x40082250 <panic_abort+24>
lbeg 0x400934b1 1074345137
lend 0x400934d3 1074345171
lcount 0x0 0
sar 0x10 16
ps 0x60f21 397089
threadptr <unavailable>
br <unavailable>
scompare1 <unavailable>
[... 243 lines cut ...]
.coredump.tasks.data 0x3ffe5c2c 0x158 RW
.coredump.tasks.data 0x3ffe5a00 0x224 RW
.coredump.tasks.data 0x3ffba28c 0x158 RW
.coredump.tasks.data 0x3ffb7dd0 0x1b4 RW
.coredump.tasks.data 0x3fff3e20 0x158 RW
.coredump.tasks.data 0x3fff3bf0 0x228 RW
.coredump.tasks.data 0x3ffdc430 0x158 RW
.coredump.tasks.data 0x3ffdc230 0x1f8 RW
.coredump.tasks.data 0x3ffdf294 0x158 RW
.coredump.tasks.data 0x3ffdf090 0x1fc RW
.coredump.tasks.data 0x3ffd82f4 0x158 RW
.coredump.tasks.data 0x3ffdb270 0x1f4 RW
.coredump.tasks.data 0x3fff2cc0 0x158 RW
.coredump.tasks.data 0x3fff2990 0x328 RW
.coredump.tasks.data 0x3ffb7924 0x158 RW
.coredump.tasks.data 0x3ffb7740 0x1dc RW
.coredump.tasks.data 0x3ffe77ec 0x158 RW
.coredump.tasks.data 0x3ffe75b0 0x234 RW
.coredump.tasks.data 0x3ffd6758 0x158 RW
.coredump.tasks.data 0x3ffd65b0 0x1a0 RW
===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```
</details>

sxsx2yzyz, 2023-05-10:
sdkconfig.h
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000
#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration
#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type
#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager
#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_INFO is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=0
# CONFIG_BOOTLOADER_SPI_CUSTOM_WP_PIN is not set
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
# end of Bootloader config
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
[... 1600 lines cut ...]
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
# CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY_CPU1=y
CONFIG_TCPIP_TASK_AFFINITY=0x1
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY is not set
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=0
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

MaxwellAlan, 2023-09-15:
Hi @sxsx2yzyz 

How often does this issue reproduce in your test env?

Could you retest with the latest release/v4.4 ?

Thanks.

closed 2023-09-26 by sxsx2yzyz