espressif/esp-idf #11376: release 5.1 panics when attempt to connect using WiFi 6

roblatour, 2023-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4734-g8a08cfe7d1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32-c6

### Power Supply used.

USB

### What is the expected behavior?

I expect the esp32-c6 to be able to connect to a 2.4G WiFi 6 network, or if this is not yet implanted to see a graceful error reported.

### What is the actual behavior?

log reports:  ESP_ERR_INVALID_ARG
associated with the statement:
`ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA , WIFI_PROTOCOL_11AX)); `
and then the rtos panics

### Steps to reproduce.

```
static void start_wifi(void)
{

    // This subroutine starts the WiFi.

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();
    assert(sta_netif);

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL, NULL));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler, NULL, NULL));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = DEFAULT_SSID,
            .password = DEFAULT_PWD,
            .listen_interval = DEFAULT_LISTEN_INTERVAL,        
            .pmf_cfg = {                                       
                .capable = true,                            
                .required = false                          
            },
        },
[... 260 lines cut ...]
0x42000260: pthread_cleanup_thread_specific_data_callback at C:/esp-r-5-1/esp-idf/components/pthread/pthread_local_storage.c:126

4081e6c0: 0x00000001 0x00000000 0x00000000 0x4087d714 0x420ac478 0x00000000 0x00000000 0x00000000
0x420ac478: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

4081e6e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081e700: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081e720: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081e740: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081e760: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 9384f0afe15cb69a

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2b30
load:0x40875720,len:0x1860
entry 0x4086c410
```

xuxiao111, 2023-05-11:
@roblatour Hi, you can use "ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA , WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX));" instead of “ESP_ERROR_CHECK(esp_wifi_set_protocol(ESP_IF_WIFI_STA , WIFI_PROTOCOL_11AX));”

roblatour, 2023-05-11:
If I do what you suggest how to I tell it to prefer 11ax? 

Additionally, I found this code to check the connected protocols:
    if (esp_wifi_get_protocol(WIFI_IF_STA, &protocol_bitmap) == ESP_OK) {
    if (protocol_bitmap & 1) { ESP_LOGI(TAG, "Connected 11b"); }  // 11b
    if (protocol_bitmap & 2) { ESP_LOGI(TAG, "Connected 11g"); }  // 11g
    if (protocol_bitmap & 4) { ESP_LOGI(TAG, "Connected 11n");; }  // 11n
    if (protocol_bitmap & 8) { ESP_LOGI(TAG, "Connected low rate");}}  // Low rate

and with your suggestion it is reporting I am getting connection on 11b, 11g, 11n.
Not sure how to check if the connection on 11ax has been made or not?

xuxiao111, 2023-05-12:
Hi, "esp_wifi_set_protocol" is just setting the protocol that the STA can support. In fact, the protocol used after the connection is determined by the AP. You can use "esp_wifi_sta_get_negotiated_phymode" to get the negotiated phymode after the connection.

roblatour, 2023-05-12:
Perfect ... 

I just wrote this code (which I ran after the connect) 

void confirm_protocol_in_use()
{

    wifi_phy_mode_t WiFiMode;
    if (esp_wifi_sta_get_negotiated_phymode(&WiFiMode) == ESP_OK)
    {
        if (WiFiMode == WIFI_PHY_MODE_11B)
            ESP_LOGI(TAG, "11b");

        if (WiFiMode  == WIFI_PHY_MODE_11G)
            ESP_LOGI(TAG, "11g");

        if (WiFiMode == WIFI_PHY_MODE_LR)
            ESP_LOGI(TAG, "low rate");

        if (WiFiMode == WIFI_PHY_MODE_HT20)
            ESP_LOGI(TAG, "HT20");
            
        if (WiFiMode == WIFI_PHY_MODE_HT40)
            ESP_LOGI(TAG, "HT40");
            
        if (WiFiMode == WIFI_PHY_MODE_HE20)
            ESP_LOGI(TAG, "HE20");
    }
}

It reported 11b when connecting to my non-WiFi 6 network, and HE20 when connecting to my WiFi network.
Which is exactly what it should, and what I needed to confirm.

Regardless, I think the issue I originally reported is still something to look at; but with thanks to your help I don't need to approach it that way as the above code is able to confirm it is working as it should be.

Thanks again.

xuxiao111, 2023-05-12:
It's my pleasure and this issue we will take some time to look at it.

roblatour, 2023-05-22:
seems to be working now

closed 2023-05-22 by roblatour

kasul1, 2024-02-25:
I also get an error when I try this
esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11AX );
But
esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX);
But esp_wifi_sta_get_negotiated_phymode(&phymode);
The result is 
Only WIFI_PHY_MODE_HT20 comes out
WIFI_PHY_MODE_HE20 doesn't come out at all