espressif/esp-idf #11378: Failed to flash to ESP32-S3 via built-in USB serial on Linux but succeeded with Windows

lujinhao, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I'm using idf on Ubuntu 22. I can monitor my ESP32-S3 but failed to flash code via built-in USB serial.

The error message I got:
```
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/jinhao/esp/hello_world/build/hello_world.bin
[861/862] cd /home/jinhao/esp/hello_world/build/esp-idf/esptool_py && /home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/jinhao/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/jinhao/esp/hello_world/build/partition_table/partition-table.bin /home/jinhao/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x2fc80 bytes. Smallest app partition is 0x100000 bytes. 0xd0380 bytes (81%) free.
[861/862] cd /home/jinhao/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/jinhao/esp/esp-idf -D "SERIAL_TOOL=/home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/jinhao/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/jinhao/esp/hello_world/build -P /home/jinhao/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:4d:af:e8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x0003ffff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20304 bytes to 12889...
Writing at 0x00000000... (100 %)
Wrote 20304 bytes (12889 compressed) at 0x00000000 in 0.3 seconds (effective 471.5 kbit/s)...

A fatal error occurred: Packet content transfer stopped (received 25 bytes)
FAILED: CMakeFiles/flash /home/jinhao/esp/hello_world/build/CMakeFiles/flash 
cd /home/jinhao/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/jinhao/esp/esp-idf -D "SERIAL_TOOL=/home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/jinhao/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/jinhao/esp/hello_world/build -P /home/jinhao/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
```

However, I also tried on Windows with VScode setup and succeeded to flash code. I confirmed the menuconfig files are identical on Windows and Linux. 

The log on Windows
```
[860/860] cmd.exe /C "cd /D C:\Espressif\frameworks\hello_world\build\esp-idf\esptool_py && C:\Espress...d/build/partition_table/partition-table.bin C:/Espressif/frameworks/hello_world/build/hello_world.bin" 
hello_world.bin binary size 0x2fcb0 bytes. Smallest app partition is 0x100000 bytes. 0xd0350 bytes (81%) free.

 *  Executing task: C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\lujh8\esp\esp-idf\components\esptool_py\esptool\esptool.py -p COM4 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin 

esptool.py v4.5.1
Serial port COM4
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:4d:af:e8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x0003ffff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20304 bytes to 12889...
Wrote 20304 bytes (12889 compressed) at 0x00000000 in 0.3 seconds (effective 484.5 kbit/s)...
Hash of data verified.
Compressed 195760 bytes to 101682...
Wrote 195760 bytes (101682 compressed) at 0x00010000 in 1.9 seconds (effective 810.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 620.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

lujinhao, 2023-05-11:
I searched online for a while and found a solution that worked for me: `disable download stub`. https://blog.yelvlab.cn/archives/637/

I forgot to mention I'm running Ubuntu 22 on Window virtual machine. I found the same issue with the same setup I have https://github.com/espressif/openocd-esp32/issues/209

Without download stub, it becomes much slower. It would be great if it can flash successfully with stub enabled.

dobairoland, 2023-05-11:
There is also another option. You can start an RFC2217 server on your Windows machine and flash the device through that port from the virtual machine. The port needs be accessible from the virtual machine.

More on this topic:

https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/remote-serial-ports.html

radimkarnis, 2023-05-11:
> Probably the timing of the reset sequence is broken coming from the virtual machine.

Not the reset sequence (the failure doesn't happen while resetting), but mangled data transfer.

Have you tried lowering the baud rate?

lujinhao, 2023-05-12:
I just tried to lower baud rate down to 115200 and 9600, still no success. I noticed the writing speed shown is 479.2 kbit/s, certainly higher than 9600. I'm not sure if my baud rate command takes into effect.

```
jinhao@jinhao:~/esp/hello_world$ idf.py -p /dev/ttyACM0 -b 9600 flash
Executing action: flash
Running ninja in directory /home/jinhao/esp/hello_world/build
Executing "ninja flash"...
[1/5] cd /home/jinhao/esp/hello_world/build/esp-idf/esptool_py && /home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/jinhao/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/jinhao/esp/hello_world/build/partition_table/partition-table.bin /home/jinhao/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x2fc80 bytes. Smallest app partition is 0x100000 bytes. 0xd0380 bytes (81%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/jinhao/esp/hello_world/build/bootloader/esp-idf/esptool_py && /home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python /home/jinhao/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/jinhao/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x4f50 bytes. 0x30b0 bytes (38%) free.
[2/3] cd /home/jinhao/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/jinhao/esp/esp-idf -D "SERIAL_TOOL=/home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/jinhao/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/jinhao/esp/hello_world/build -P /home/jinhao/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/ttyACM0 -b 9600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:4d:af:e8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x0003ffff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 20304 bytes to 12888...
Writing at 0x00000000... (100 %)
Wrote 20304 bytes (12888 compressed) at 0x00000000 in 0.3 seconds (effective 479.2 kbit/s)...

A fatal error occurred: Packet content transfer stopped (received 25 bytes)
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/jinhao/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed

FAILED: CMakeFiles/flash /home/jinhao/esp/hello_world/build/CMakeFiles/flash 
cd /home/jinhao/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/jinhao/esp/esp-idf -D "SERIAL_TOOL=/home/jinhao/.espressif/python_env/idf5.0_py3.10_env/bin/python;;/home/jinhao/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/jinhao/esp/hello_world/build -P /home/jinhao/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jinhao/esp/hello_world/build/log/idf_py_stderr_output_7990 and /home/jinhao/esp/hello_world/build/log/idf_py_stdout_output_7990
```

radimkarnis, 2023-05-15:
Hi @lujinhao,
does your board have an integrated USB-to-UART bridge OR can you connect to the TX/RX pins with an external bridge?

If you can do that, we could also try to monitor this output to see the [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html#boot-log) and see the reset reason. Or if the chip crashes, we would be able to see the backtrace.
That means you would be trying to flash over the USB-Serial/JTAG peripheral (built-in USB) and monitoring over UART at the same time.

Also, @dobairoland suggested using RFC2217 as another possible solution, have you been able to try that? 
Thank you!

radimkarnis, 2023-06-26:
This issue [should be fixed](https://github.com/espressif/esptool/commit/6fd91afec42f7717f02bb41a4bb2789c73187c63) in the last release of esptool. This gets automatically installed if you run the install script again. Closing this issue - feel free to reopen if issues persist!

closed 2023-06-26 by radimkarnis

man715, 2024-01-17:
@radimkarnis,

FWIW, I am having the same issue. I am running Debian with ESP-IDF v5.3-dev-1353-gb3f7e2c8a4 in a VM. When I set Disable download stub, it works.