espressif/esp-idf #11379: ESP32 joined AP but cannot get IP address

AxelLin, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.5-83-g33b3db35d7

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

The device should be able to get IP address via dhcp after successfully join AP.

### What is the actual behavior?

One of my ESP32-WROOM-32E device cannot get IP address after join AP.

This issue happened with one of my testing AP.
Other ESP32-WROOM-32E devices do not have this issue, so I think AP is fine.
PS. This device works with other AP, so I think the hw is OK. 

I didn't see any error while testing.
However, when I set CONFIG_FREERTOS_UNICORE=y, I notice below error:
(It does not show below error if CONFIG_FREERTOS_UNICORE=n, just does not work)

dhcp_recv(pbuf = 0x3ff81a2c) from DHCP server 192.168.1.1 port 67
pbuf->len = 293
pbuf->tot_len = 293
netif->hwaddr[0]==a8 != reply_msg->chaddr[0]==a4

The device's MAC is: A8032AE6C4F4

### Steps to reproduce.

Just make device working in STA mode and join AP.

### Debug Logs.

```plain
Below log shows above hwaddr mismatch error.

FreeRTOS single-core:

D (11420) wpa: WPA: Key negotiation completed with ac:84:c6:32:1a:26 [PTK=CCMP GTK=CCMP]

I (11429) wifi:connected with IGS-MESH, aid = 1, channel 9, 40D, bssid = ac:84:c6:32:1a:26
I (11436) wifi:security: WPA2-PSK, phy: bgn, rssi: -44
I (11442) wifi:pm start, type: 1

D (11446) esp_netif_handlers: esp_netif action connected with netif0x3ffbf614 from event_id=4
D (11453) esp_netif_lwip: check: remote, if=0x3ffbf614 fn=0x40105584
0x40105584: esp_netif_up_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1276

D (11460) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffbf614
D (11466) esp_netif_lwip: check: local, if=0x3ffbf614 fn=0x40106380
0x40106380: esp_netif_update_default_netif_lwip at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:197

D (11473) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbf614
I (11474) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (11486) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (11491) esp_netif_lwip: check: remote, if=0x3ffbf614 fn=0x401052e0
0x401052e0: esp_netif_dhcpc_start_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1065

D (11498) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffbf614
D (11504) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf614
D (11512) esp_netif_lwip: if0x3ffbf614 start ip lost tmr: no need start because netif=0x3ffbf694 interval=120 ip=0
D (11521) esp_netif_lwip: starting dhcp client
dhcp_start(netif=0x3ffbf694) st1
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 500 msecs
D (11567) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (11593) wifi:<ba-add>idx:0 (ifx:0, ac:84:c6:32:1a:26), tid:0, ssn:0, winSize:64
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 1000 msecs
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 2000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 4000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 8000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_recv(pbuf = 0x3ffd405c) from DHCP server 192.168.1.1 port 67
pbuf->len = 293
pbuf->tot_len = 293
netif->hwaddr[0]==a8 != reply_msg->chaddr[0]==a4
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 15000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_recv(pbuf = 0x3ff81a2c) from DHCP server 192.168.1.1 port 67
pbuf->len = 293
pbuf->tot_len = 293
netif->hwaddr[0]==a8 != reply_msg->chaddr[0]==a4
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_recv(pbuf = 0x3ff81a2c) from DHCP server 192.168.1.1 port 67
pbuf->len = 293
pbuf->tot_len = 293
netif->hwaddr[0]==a8 != reply_msg->chaddr[0]==a4
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(658b293e)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 15000 msecs
dhcp_coarse_tmr()
```

### More Information.

Also tried v4.4.4-433-g3cec3a0026, the same symptom.

AxelLin, 2023-05-11:
I no longer observe hwaddr mismatch error after testing multiple times, however, the symptom is still the same, cannot get IP.

nishanth-radja, 2023-05-15:
@AxelLin Can we pls get a sniffer capture for the same, Also ssid and password used for connection is required to decode the packets. All of the 4 way handshake need to be present in the capture to decode the packets.

AxelLin, 2023-05-15:
@nishanth-radja 
There are many testing devices, do you have recommend filter setting to filter out unrelated packets?

nishanth-radja, 2023-05-16:
Yes, Pls provide the AP mac and STA mac,We will be able to filter out using them.

AxelLin, 2023-05-25:
@nishanth-radja 
I'm not allow to capture the log in this environment.
However, by comparing the failure device with normal device, I notice the wifi log is different.
Do you see anything wrong on the failure device boot log?

This time , I build with v5.2-dev-764-gd00e7b5af8

The failure device boot log: (cannot get IP address)
```plain
I (2077) wifi:set country: cc=US schan=1 nchan=11 policy=0

I (2086) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2181) wifi:mode : sta (a8:03:2a:e6:c4:f4)
I (2182) wifi:enable tsf
I (2183) net_utils: STA started
I (4596) wifi:new:<9,2>, old:<1,0>, ap:<255,255>, sta:<9,2>, prof:1
I (5092) wifi:state: init -> auth (b0)
I (5100) wifi:state: auth -> assoc (0)
I (5151) wifi:state: assoc -> run (10)
I (8255) wifi:state: run -> init (2c0)
I (8256) wifi:new:<9,0>, old:<9,2>, ap:<255,255>, sta:<9,2>, prof:1
I (8258) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (8261) wifi:state: init -> auth (b0)
I (8268) wifi:state: auth -> assoc (0)
I (8279) wifi:state: assoc -> run (10)
I (8327) wifi:connected with IGS-MESH, aid = 3, channel 9, 40D, bssid = ac:84:c6:32:1a:26
I (8328) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
I (8330) wifi:pm start, type: 1

I (8349) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```
The normal device boot log:
```plain
I (2064) wifi:set country: cc=US schan=1 nchan=11 policy=0

I (2073) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (2177) wifi:mode : sta (f0:08:d1:79:8c:60)
I (2178) wifi:enable tsf
I (2180) net_utils: STA started
I (4592) wifi:new:<9,2>, old:<1,0>, ap:<255,255>, sta:<9,2>, prof:1
I (5088) wifi:state: init -> auth (b0)
I (5102) wifi:state: auth -> init (8a0)
I (5103) wifi:new:<9,0>, old:<9,2>, ap:<255,255>, sta:<9,2>, prof:1
I (5104) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (5108) wifi:state: init -> auth (b0)
I (5116) wifi:state: auth -> assoc (0)
I (5126) wifi:state: assoc -> run (10)
I (5175) wifi:connected with IGS-MESH, aid = 1, channel 9, 40D, bssid = ac:84:c6:32:1a:26
I (5175) wifi:security: WPA2-PSK, phy: bgn, rssi: -46
I (5177) wifi:pm start, type: 1

I (5241) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5726) wifi:<ba-add>idx:0 (ifx:0, ac:84:c6:32:1a:26), tid:6, ssn:2, winSize:64
I (6685) esp_netif_handlers: sta ip: 192.168.1.108, mask: 255.255.255.0, gw: 192.168.1.1

```

AxelLin, 2023-05-25:
BTW, What does the number after the state machine switch in Wi-Fi log mean?
This document only shows an example https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/en/latest/software-framework/wifi.html#connect-what-does-the-number-after-the-state-machine-switch-in-wi-fi-log-mean
Where is the reference for all reasons?

aureliendeb34, 2023-10-19:
Hi , I have the issue that you show on you last post. My situation is the following:
I have a custom board with ESP32. We've updated our firmware from ESP-IDF 4 to ESP-IDF 5.0.1
On production devices, we have about 10% of them that wasn't able to reconnect to AP.
On the log below, it seems the device is trying to connect to AP several times but it can't.

I (2096021) wifi: Enabling WiFi
I (2096031) wifi: WiFi connect: SSID='XXXXXXX' ...
I (2096041) wifi:mode : sta (94:b9:7e:ea:e2:60)
I (2096051) wifi:enable tsf
I (2096091) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2096101) wifi:state: init -> auth (b0)
I (2097111) wifi:state: auth -> init (200)
I (2097111) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2097121) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2097131) wifi:state: init -> auth (b0)
I (2097151) wifi:state: auth -> assoc (0)
I (2097181) wifi:state: assoc -> run (10)
I (2100321) wifi:state: run -> init (2c0)
I (2100331) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2100341) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2100351) wifi:state: init -> auth (b0)
I (2100371) wifi:state: auth -> assoc (0)
I (2100431) wifi:state: assoc -> run (10)
I (2103531) wifi:state: run -> init (2c0)
I (2103541) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
E (2125491) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (2125491) task_wdt:  - IDLE (CPU 1)
E (2125491) task_wdt: Tasks currently running:
E (2125491) task_wdt: CPU 0: IDLE
E (2125491) task_wdt: CPU 1: tiT
E (2125491) task_wdt: Print CPU 1 backtrace

Backtrace: 0x4008E41E:0x3FFC0E80 0x4008341D:0x3FFC0EA0 0x40168621:0x3FFD1200 0x4016868E:0x3FFD1220 0x4016A968:0x3FFD1240 0x4016AA2B:0x3FFD1260 0x40168791:0x3FFD1280 0x40160E85:0x3FFD12A0
0x4008e41e: esp_ptr_in_dram at /home/aurelien/esp/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:48
 (inlined by) esp_flash_write at /home/aurelien/esp/esp-idf/components/spi_flash/esp_flash_api.c:881

0x4008341d: _xt_coproc_exc at /home/aurelien/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1012

0x40168621: udp_sendto_if at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:654

0x4016868e: udp_sendto_if at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:665

0x4016a968: icmp_input at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/ipv4/icmp.c:162

0x4016aa2b: icmp_input at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/ipv4/icmp.c:219

0x40168791: udp_sendto at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/udp.c:583

0x40160e85: dns_send at /home/aurelien/esp/esp-idf/components/lwip/lwip/src/core/dns.c:864

E (2145491) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (2145491) task_wdt:  - IDLE (CPU 1)
E (2145491) task_wdt: Tasks currently running:
E (2145491) task_wdt: CPU 0: IDLE
E (2145491) task_wdt: CPU 1: tiT
E (2145491) task_wdt: Print CPU 1 

then the watchdog task is trigged in loop. I can't identify why sometimes on some AP I've got this issue. The issue is really difficult to reproduce because it appears only sometimes.
Please could you tell me which problem occurs when we have: I (2103531) wifi:state: run -> init (2c0)

If you need more information to help, ask me. (for example the sdkconfig file or something else) thanks for your help. I continue to investigate on my issue. BR

AxelLin, 2023-10-19:
@aureliendeb34 Can you provide wireshark log?

AxelLin, 2024-07-01:
@nishanth-radja 
It has been in "In Progress" state for 3 Months, any update?

vik-gokhale, 2024-07-01:
Hi @AxelLin ,
We have not received any sniffer trace so far.
Without the necessary logs, we cannot debug the exact cause.
Is there a sniffer trace for this exact problem that you can share? Else we would have to close this GH Issue.

Regards,
Vikram

AxelLin, 2024-07-01:
> Hi @AxelLin , We have not received any sniffer trace so far. Without the necessary logs, we cannot debug the exact cause.

I have to point out the "In Progress" state is really misleading.
I thought you guys are in-progress to fix the issue, but it's not true. 

> Is there a sniffer trace for this exact problem that you can share? Else we would have to close this GH Issue.

I will check if it's possible to get the sniffer.

Regards,
Axel

AxelLin, 2024-07-01:
> Hi @AxelLin , We have not received any sniffer trace so far. Without the necessary logs, we cannot debug the exact cause. 

I don't get any feedback above the log I provided. (It's not sniffier, but do you see anything unusual from the log?)

AxelLin, 2024-07-09:
@vik-gokhale 

Attached the log, 

AP: ac:84:c6:32:1a:26
NG: sta: A8:03:2A:E6:C4:F4 (This device cannot get IP address)
OK: sta: F0:08:D1:79:8C:60 (This device can get IP address, for comparison)

[sta_get_ip_fail.pcapng.txt](https://github.com/user-attachments/files/16136137/sta_get_ip_fail.pcapng.txt)
[sta_get_ip_ok.pcapng.txt](https://github.com/user-attachments/files/16136140/sta_get_ip_ok.pcapng.txt)

vik-gokhale, 2024-07-10:
hi @AxelLin 
I see only 3 frames captured in the .pcapng.
This trace is not helpful.
[image]

Please try capturing the sniffer trace along with the Console logs with Debug enabled.
Without corresponding sniffer and console logs, also without data frames captured, analysis is impossible.

vik-gokhale, 2024-07-10:
@AxelLin 
IDF v4.x is EoL and no bug fixes would be entertained anymore.

AxelLin, 2024-07-10:
> @AxelLin IDF v4.x is EoL and no bug fixes would be entertained anymore.

FYI, My previous test was on v5.2 branch.

AxelLin, 2024-07-10:
> hi @AxelLin I see only 3 frames captured in the .pcapng. This trace is not helpful. 

But that is exactly what I got.
I also checked sniffer in the OK case, and it only shows "Probe Request".
I follow the comments in https://github.com/espressif/esp-idf/issues/8269#issuecomment-1034756929 for sniffer. Is there anything I did wrong? Or any additional
steps required for HT40?

vik-gokhale, 2024-07-10:
Yes, capturing HT data frames is not reliable on all network cards, HT40 is even more tricky.
With your network card, it leaves us with only one option - please set the STA in 11g mode, replicate and share all logs (Sniffer trace, Console Logs).
Also please the sniffer physically in between the Station and the AP, just to make the captures more reliable.

AxelLin, 2024-08-05:
@vik-gokhale 
I capture again (with a good one and a bad one), please take a look.
What's wrong with the bad device,

[a8_03_2a_e6_c4_f4_ng_log.pcapng.txt](https://github.com/user-attachments/files/16496320/a8_03_2a_e6_c4_f4_ng_log.pcapng.txt)
[f0_08_d1_79_8c_60_ok_log.pcapng.txt](https://github.com/user-attachments/files/16496321/f0_08_d1_79_8c_60_ok_log.pcapng.txt)

vik-gokhale, 2024-08-05:
Hi @AxelLin,
1. These traces are captured from a LAN interface and not a WLAN interface.
2. DHCP Discover and DHCP Request is not captured even in good example.

It is very important to get traces from a wireless interface and corresponding console logs with debug enabled.

AxelLin, 2024-08-06:
> Hi @AxelLin,
> 
> 1. These traces are captured from a LAN interface and not a WLAN interface.

My previous capture using WLAN interface does not capture all packets.
So this time I use the port mirror function on the router and capture packets
via the LAN port. I think this is all I can do now.
If it still does not help, I have no choice but give up on this issue.

closed 2024-08-07 by AxelLin

vik-gokhale, 2024-08-07:
Hi @AxelLin ,
Unfortunately, we have not been able to replicate this so far.
And without complete OTA traces and console logs, there is not much analysis we can do.

Also, If someone-else from the dev community has this problem, we would get another ticket.
Let us see if they could provide a complete trace.