espressif/esp-idf #11381: Wifi authentication with WIFI_ALL_CHANNEL_SCAN and multiple APs with same SSID leads to memory leak if PWD is wrong

RungeJan, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROVER-E on Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

## Setup
Multiple APs with the same ssid (wifi network with mutliple repeaters, security WPA2/WPA3-Personal)
Trying (unlimited) reconnects with wifi config containing invalid passwort
wifi_config_t: sta.scan_method is set to WIFI_ALL_CHANNEL_SCAN
## Expected behaviour
The program is running forever, trying to access the ssid with a wrong password, available heap is not decreasing constantly.

## Code
The following adjusted version of the wifi getting started example shows the problem:
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "esp_timer.h"
#include "esp_mac.h"

#include "lwip/err.h"
#include "lwip/sys.h"

#define EXAMPLE_ESP_WIFI_SSID      "phasepi"
#define EXAMPLE_ESP_WIFI_PASS      "wrongPwd"

static EventGroupHandle_t s_wifi_event_group;

#define WIFI_CONNECTED_BIT BIT0

static const char *TAG = "wifi station";

static void event_handler(void *arg, esp_event_base_t event_base,
	int32_t event_id, void *event_data)
{
	if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
[... 672 lines cut ...]

0x400fd184: cnx_start_handoff_cb at ??:?

0x400f6440: clear_bss_queue at ??:?

0x400f66ca: clear_bss_queue at ??:?

0x400f68d8: scan_inter_channel_timeout_process at ??:?

0x400fb7b1: chm_end_op at ??:?

0x400fb7d2: chm_end_op_timeout_process at ??:?

0x400fb310: esp_wifi_register_mgmt_frame_internal at ??:?

0x400fb599: ieee80211_timer_do_process at ??:?

0x40090db1: ppTask at ??:?

0x4008bccd: vPortTaskWrapper at E:/ESP_IDF_5.0.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

### More Information.

The issue was already present on v5.0.1, I haven`t tried earlier versions yet

kapilkedawat, 2023-05-22:
Hi @RungeJan ,

I tried this issue in our internal setup, I don't see memory going down after every iteration. It actually becomes constant after some iterations. Can you please share AP's config? Also how many APs with same SSID is present in the env?

I (85591) wifi station: [up:85] Heap: free: 232180(internal: 232180) (largest block: 110592), min free: 229520 / 301568 max
I (85891) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (85891) wifi:state: init -> auth (b0)
I (86551) wifi:state: auth -> assoc (0)
I (86861) wifi:state: assoc -> run (10)
I (89621) wifi:state: run -> init (fc0)
I (89621) wifi:new:<9,0>, old:<9,2>, ap:<255,255>, sta:<9,2>, prof:1
I (89621) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (89631) wifi:state: init -> auth (b0)
I (89651) wifi:state: auth -> assoc (0)
I (90591) wifi station: [up:90] Heap: free: 232156(internal: 232156) (largest block: 110592), min free: 229520 / 301568 max
I (90661) wifi:state: assoc -> init (2700)
I (90661) wifi:new:<9,0>, old:<9,2>, ap:<255,255>, sta:<9,2>, prof:1
I (90661) wifi station: connect to the AP fail
I (90661) wifi station: retry to connect to the AP
I (93081) wifi station: connect to the AP fail
I (93081) wifi station: retry to connect to the AP
I (95491) wifi:new:<9,2>, old:<9,0>, ap:<255,255>, sta:<9,2>, prof:1
I (95491) wifi:state: init -> auth (b0)
I (95591) wifi station: [up:95] Heap: free: 232168(internal: 232168) (largest block: 110592), min free: 229520 / 301568 max

RungeJan, 2023-05-22:
Hi,
I adjusted the code to print all aps that were found after startup.
<details> <summary>Code</summary>

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "esp_timer.h"
#include "esp_mac.h"

#include "lwip/err.h"
#include "lwip/sys.h"

#define EXAMPLE_ESP_WIFI_SSID      "our_ssid"
#define EXAMPLE_ESP_WIFI_PASS      "wrongPwd"

static EventGroupHandle_t s_wifi_event_group;

#define WIFI_CONNECTED_BIT BIT0
#define WIFI_STARTED_BIT BIT1

static const char *TAG = "wifi station";

static void event_handler(void *arg, esp_event_base_t event_base,
	int32_t event_id, void *event_data)
{
	if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
		xEventGroupSetBits(s_wifi_event_group, WIFI_STARTED_BIT);
	} else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
		ESP_LOGI(TAG, "connect to the AP fail");
		ESP_LOGI(TAG, "retry to connect to the AP");
		esp_wifi_connect();
	} else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
		ip_event_got_ip_t *event = (ip_event_got_ip_t *)event_data;
		ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
		xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
	}
}

void wifi_init_sta(void)
{
	s_wifi_event_group = xEventGroupCreate();

	ESP_ERROR_CHECK(esp_netif_init());

	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_netif_create_default_wifi_sta();

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));

	esp_event_handler_instance_t instance_any_id;
	esp_event_handler_instance_t instance_got_ip;
	ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
		ESP_EVENT_ANY_ID, &event_handler, NULL, &instance_any_id));
	ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
		IP_EVENT_STA_GOT_IP, &event_handler, NULL, &instance_got_ip));

	wifi_config_t wifi_config = {
		.sta = {
			.ssid = EXAMPLE_ESP_WIFI_SSID,
			.password = EXAMPLE_ESP_WIFI_PASS,
			.scan_method = WIFI_ALL_CHANNEL_SCAN,
[... 316 lines cut ...]

I (89631) wifi:state: init -> auth (b0)
I (89651) wifi:state: auth -> assoc (0)
I (90661) wifi:state: assoc -> init (2700)

My states are mostly only switching between auth and init
I (209614) wifi:state: init -> auth (b0)
I (210994) wifi:state: auth -> init (600)
I (211004) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (211004) wifi:new:<1,0>, old:<6,0>, ap:<255,255>, sta:<1,0>, prof:1
I (211004) wifi:state: init -> auth (b0)
I (212424) wifi:state: auth -> init (600)
I (212434) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (212434) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (212434) wifi:state: init -> auth (b0)
I (213824) wifi:state: auth -> init (600)
I (213834) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (213834) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (213834) wifi:state: init -> auth (b0)
I (215234) wifi:state: auth -> init (600)
I (215234) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1

I hope this helps.
Kind regards
Jan

AxelLin, 2023-06-25:
@kapilkedawat 
Any new finding in https://github.com/espressif/esp-idf/issues/11381#issuecomment-1556797857

masterxq, 2024-05-18:
Your and also my Problem is 95% related to WPA3 sae. And still present in latest Espressif:
```
ESP-IDF v4.4.7-197-g1b42fee4f0
```
I also have access points with the same name, so possible it is a combination ob both, but unsure there.

I did some test with heap trace. With Espressif ESP-IDF v4.4.7-197-g1b42fee4f0 and ESP-IDF v4.4.6-487-g2d60e58888. Both have the same result.

My experiment:
I started heap_trace and disconnected the client by the access point, so it needed to reconnect multiple times. My access point will block reconnect attempts about 1 minute. And after i disconnected multiple times i stopped heap trace and dumped its log. I did this multiple times with different amount of disconnects.

<details>
  <summary>Code</summary>

```
	while(true)
	{
		int len = uart_read_bytes(0, buffer, 128, 20 / portTICK_PERIOD_MS);
		if(len > 0)
		{
			buffer[len] = 0;
			if(buffer[0] == 'e')
			{
				if(!heaptrace_activated)
				{
					ESP_LOGW(TAG, "Heap trace allready 'E'nded");
				}
				else
				{
					wifi_manager_disable_wifi();
					vTaskDelay(10000/portTICK_RATE_MS);
					ESP_ERROR_CHECK(heap_trace_stop());
					heap_trace_dump();
					heaptrace_activated = false;
					ESP_LOGI(TAG, "Heap trace ended");
					wifi_manager_enable_wifi();
				}
			}
			else if(buffer[0] == 's')
			{
				if(heaptrace_activated)
				{
					ESP_LOGW(TAG, "Heap trace allready 'S'tarted");
				}
				else
				{
					ESP_ERROR_CHECK(heap_trace_start(HEAP_TRACE_LEAKS));
					heaptrace_activated = true;
					ESP_LOGI(TAG, "Heap trace started");
				}
			}
			ESP_LOGW(TAG, "Got %d bytes: %s", len, buffer);
		}
		wifi_manager_worker();
		vTaskDelay(10/portTICK_RATE_MS);
	}
```
</details>

<details>
<summary>Reconnect log</summary>

```
I (1270003) wifi:<ba-del>idx
I (1270013) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
W (1270033) httpd_txrx: httpd_sock_err: error in recv : 113
W (1280153) WifiMan: Wifi scan callback NULL connecting...
I (1280233) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1280243) wifi:state: init -> auth (b0)
I (1280243) wifi:state: auth -> init (1100)
[... 3028 lines cut ...]
0x400e9e80: mem_malloc at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x400e9ef6: do_memp_malloc_pool at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/memp.c:254

0x400e9f5c: memp_malloc at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400f0aec: sys_timeout_abs at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/timeouts.c:201

0x400f0b7c: lwip_cyclic_timer at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/timeouts.c:273

0x400f0ca0: sys_check_timeouts at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/core/timeouts.c:411

0x400e9acd: tcpip_timeouts_mbox_fetch at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/api/tcpip.c:117

0x400e9b85: tcpip_thread at /opt/htv-apps/espressif-4.4/esp-idf/components/lwip/lwip/src/api/tcpip.c:149

0x4008d908: vPortTaskWrapper at /opt/htv-apps/espressif-4.4/esp-idf/components/freertos/port/xtensa/port.c:144

728 bytes 'leaked' in trace (12 allocations)
```
</details>

@RungeJan It may be a workaround to disable WPA3 in your configuration if your environment supports this. If possible verify my tests.

@ espressif team. If my analysis is correct this may be a Hugh problem with ESP32 in a production environment. Please hurry up to fix this.

masterxq, 2024-05-22:
After some time of running my devices with WPA3 disabled one of my devices lost a big memory chunk again, about 100 KiByte. I'm unsure where the problem comes from. But it looks very much like the original bug report in this ticket.

I just got positive feedback to a feature that will help me to trace it. I will make a long time test with heap tracing as soon it is merged.

https://github.com/espressif/esp-idf/issues/13803

Never mind, it is possible or even likely that both problems exists, the WPA3 leak and the scan leak.

Should I create a second issue with the heap trace and possible WPA3 memory leak above? 

Best Regards

kapilkedawat, 2024-05-23:
Hi @masterxq, SAE leak issue was fixed in lDF v5.1 onward and we missed to backport it. This has been backported now and will be available in next sync. 

For the scan leak, are you using scan example or something else? We would appreciate if you can share more details on how to reproduce it.

masterxq, 2024-05-26:
@kapilkedawat 

Yes, thank you and I can confirm this, tested with:
`v5.4-dev-610-g003f3bb5dc`
The leak does not exist in this Version.

The new stop heap_trace_alloc_pause() feature was very helpfully on the analysis.

The other problem most likely still exists, but it is super hard to reproduce, it only appears very rare. And it definitely appears more often when there are problems with the AccessPoints, maybe because scanning is done more frequently in this case.

If it appears, big chunks of memory are being lost. Sadly I could not observe it on my test devices with heap trace. 

I will continue observing it for some days, and try to create Wi-Fi issues on my AccessPoints for trigger it, but possible I don't have the time to complete the analysis :/

Unfortunately, as long as the leak has not occurred on my test devices, I cannot even guarantee that it is not a leak in the user code. However, I always try to avoid allocating memory and if I do, I check and test intensively. Perhaps there are not that many code places where larger chunks are reserved and which are directly related and a static analysis of the code could be sufficient. Of course, I realize that the information base is limited and uncertain, I hope that I can contribute even more to the solution :)

masterxq, 2024-05-26:
> For the scan leak, are you using scan example or something else? We would appreciate if you can share more details on how to reproduce it.

Sorry did not completely answer your question:
Some day I forked from the scan example, but now it is a complete Wi-Fi manager component that is solving Wi-Fi task for many devices.
Sadly I can not share the code for it, as I lost the rights for doing that :(

The memory leak I observed could be in the user code or due to incorrect use of the espressif framework. But I really can't think of where else to look because I've checked everything several times and compared it with the documentation.
I'm sorry that I don't have a satisfactory answer at the moment, my test devices are much closer to a plain espressif, and they are tracking problems perhaps we are lucky there :)

kapilkedawat, 2024-05-27:
Hi @masterxq, user need to free the memory for scan after fetching the results. Please see https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv427esp_wifi_scan_get_ap_recordP16wifi_ap_record_t , I hope your code was doing that. Do let us know if you face the issue in future.

masterxq, 2024-05-27:
@kapilkedawat
This is respected everywhere. I did try to test all code pathes and made a static analysis... already before we talked :)

I gave a friend some test devices that also has the issue, maybe we can reproduce the problem there :)

kapilkedawat, 2024-06-03:
Hi @masterxq, SAE leak change is merged, feel free to try that. Can we close this issue and you can reopen/open a new issue if you encounter the leak again?

masterxq, 2024-06-10:
> Hi @masterxq, SAE leak change is merged, feel free to try that. Can we close this issue and you can reopen/open a new issue if you encounter the leak again?

I can confirm the sae leak is fixed.

I'm still losing big chunks of memory when devices are online for longer time. This happens every 3 weeks once in the average for me, with both espressif versions. 4.4.x and 5.4-dev. But until now I can not prove that this is an espressif bug and not happens in user code. An indicator that this is not user code is: I'm only allocating much memory at two places, when starting a wifi scan (indirectly) and another place where I hold the Wi-Fi results in another struct temporally (directly). Both have been checked by me and other persons to be leak free. There are also indicators that show up that the leak has something to do with wifi.

My work on the research is still active, but slow as I have a lot of immediate in other projects.

I did not open this issue, so I have not the rights to close it, but I would be fine with that.
If I find the root cause I can open a new Issue and if it is related I can refer this one.

If the sea leak was still in espressif v5.0.2, this ticket original issue is probably fixed in more recent versions, and the author would agree in closing this :)

Thanks for your assistance @kapilkedawat

masterxq, 2024-06-10:
> Hi @masterxq, you mean you are still able to reproduce this issue? If yes, could you please enable heap tracing and see where are you losing the memory?

Yes I'm still able to reproduce a problem, which could be the original problem in this issue :)
Yes that is what I'm trying, but this is not that simple, as I have the following problems:
- It only appears very rare, needs to wait 3 weeks with single device for each analysis
- It only appears in special environments
- The places where I can reproduce the issue best, I only have web access with OTA and network logs to the devices.
- The devices where the issue appears are used in production and handling real time task, I need to be carefully with the performance.
- The devices where this issue appears most recently have OTA Partitions with very few space left. (Bad partitioning :( )

So I need to write a firmware that is able to debug this over network without hitting performance limits and free some const space for being able to fit the partition max size. Or solve the problems on other ways.

I tried with test devices, but had several issues with these devices. As it is hard to reconstruct the environment and log over 3-9 weeks without touching the equipment. I would need to set up a dedicated test environment + equipment (computers, esp32, wifi access points...). And if this does not success, I have no new knowledge as I don't know what the problem was could be:
- not waited long enough
- wrong Wi-Fi environment
- Leak in unused user code
- Other work load
- Other thread behavior
- Cross component related issues (Other hardware connected) 
- ...

However, I decided that the best option is to analyze this in production, and you improved the tools to do this. But I need time to make this test, as it is really hard to reproduce.

What I did last week is to update my components to support espressif 5.x. So that I could make sure that the leak was not caused by a forgotten patch to old versions and that I had access to the latest version of the heap trace library. And patched some devices where the leak appears. I already have the result, the leak is still present when using 5.4-dev.

Ok, sorry, a lot of low relevance details, but possibly you know a more easy way to trace the bug :)

Best regards

masterxq, 2024-06-13:
I managed to complete my network heap trace tool and setup.

Started a Scan now, I hope I will soon come back with result...

masterxq, 2024-06-14:
Hi @kapilkedawat,
the issue I have is not related to Wi-Fi or scan, it is related to websockets.
It needs more research, if needed I will open a new issue.

Fell free to close this.

Thank you for your help :)

Alvin1Zhang, 2024-06-17:
Thanks for sharing the updates, feel free to reopen. Thanks.

closed 2024-06-17 by Alvin1Zhang