espressif/esp-idf #11383: ESP32-C6 second TWAI Controllers not supported

emotasHil, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The new ESP32-C6 features 2 TWAI Controllers.
The API in the ESP-IDF however does not cover support for a second TWAI controller at all.
Neither the master branch nor the development branches for 5.1 or 5.2
seems to have any indication for support for the second TWAI controller.

emotasHil, 2023-05-17:
The low level HAL seems to support the second TWAI controller already.

micbis, 2023-06-19:
Are there already plans to update the driver to support the second TWAI controller?

gpearcey, 2023-07-03:
Are the esp32-c6 twai controllers still compatible with SJA1000 as with esp32?

mlaz1, 2023-08-07:
@espressif-bot
Any status updates on this feature?

wanckl, 2023-08-08:
> Are the esp32-c6 twai controllers still compatible with SJA1000 as with esp32?

I think yeas.    

@mlaz1 Heard from internal information, 2nd TWAI will be supported at least end of this year in major release

closed 2023-10-25 by espressif-bot