espressif/esp-idf #11384: [ESP32C6] ESP-Nimble-CPP does not compile.

mickeyl, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4734-g8a08cfe7d1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have looked at the [ESP32C6 status](https://github.com/espressif/esp-idf/issues/10423) and found NimBLE to be already checked, so I tried compiling the excellent https://github.com/h2zero/esp-nimble-cpp after setting the target to `esp32c6`.

For some reason, it fails with:

```
/home/mickey/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++  -DESP_PLATFORM -DIDF_VER=\"v5.1-dev-4734-g8a08cfe7d1-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -Iconfig -I../components/esp-nimble-cpp/src -I../../../../../../components/newlib/platform_include -I../../../../../../components/freertos/FreeRTOS-Kernel/include -I../../../../../../components/freertos/FreeRTOS-Kernel/portable/riscv/include -I../../../../../../components/freertos/esp_additions/include/freertos -I../../../../../../components/freertos/esp_additions/include -I../../../../../../components/freertos/esp_additions/arch/riscv/include -I../../../../../../components/esp_hw_support/include -I../../../../../../components/esp_hw_support/include/soc -I../../../../../../components/esp_hw_support/include/soc/esp32c6 -I../../../../../../components/esp_hw_support/port/esp32c6/. -I../../../../../../components/esp_hw_support/port/esp32c6/private_include -I../../../../../../components/heap/include -I../../../../../../components/log/include -I../../../../../../components/soc/include -I../../../../../../components/soc/esp32c6 -I../../../../../../components/soc/esp32c6/include -I../../../../../../components/hal/esp32c6/include -I../../../../../../components/hal/include -I../../../../../../components/hal/platform_port/include -I../../../../../../components/esp_rom/include -I../../../../../../components/esp_rom/include/esp32c6 -I../../../../../../components/esp_rom/esp32c6 -I../../../../../../components/esp_common/include -I../../../../../../components/esp_system/include -I../../../../../../components/esp_system/port/soc -I../../../../../../components/esp_system/port/include/riscv -I../../../../../../components/esp_system/port/include/private -I../../../../../../components/riscv/include -I../../../../../../components/lwip/include -I../../../../../../components/lwip/include/apps -I../../../../../../components/lwip/include/apps/sntp -I../../../../../../components/lwip/lwip/src/include -I../../../../../../components/lwip/port/include -I../../../../../../components/lwip/port/freertos/include -I../../../../../../components/lwip/port/esp32xx/include -I../../../../../../components/lwip/port/esp32xx/include/arch -I../../../../../../components/bt/include/esp32c6/include -I../../../../../../components/bt/common/osi/include -I../../../../../../components/bt/common/api/include/api -I../../../../../../components/bt/common/btc/profile/esp/blufi/include -I../../../../../../components/bt/common/btc/profile/esp/include -I../../../../../../components/bt/porting/include -I../../../../../../components/bt/porting/nimble/include -I../../../../../../components/bt/porting/npl/freertos/include -I../../../../../../components/bt/porting/transport/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/ans/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/bas/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/dis/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/gap/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/hr/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/ias/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/lls/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/services/tps/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/util/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/store/ram/include -I../../../../../../components/bt/host/nimble/nimble/nimble/host/store/config/include -I../../../../../../components/bt/host/nimble/nimble/porting/nimble/include -I../../../../../../components/bt/host/nimble/port/include -I../../../../../../components/bt/host/nimble/nimble/nimble/transport/include -I../../../../../../components/esp_timer/include -I../../../../../../components/esp_wifi/include -I../../../../../../components/esp_wifi/wifi_apps/include -I../../../../../../components/esp_event/include -I../../../../../../components/esp_phy/include -I../../../../../../components/esp_phy/esp32c6/include -I../../../../../../components/esp_netif/include -I../../../../../../components/nvs_flash/include -I../../../../../../components/spi_flash/include -I../../../../../../components/esp_partition/include -march=rv32imac_zicsr_zifencei    -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fmacro-prefix-map=/home/mickey/Documents/late/misc/esp-idf/examples/bluetooth/nimble/ble_l2cap_coc/coc_blecent=. -fmacro-prefix-map=/home/mickey/Documents/late/misc/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu++20 -fno-exceptions -fno-rtti -MD -MT esp-idf/esp-nimble-cpp/CMakeFiles/__idf_esp-nimble-cpp.dir/src/NimBLEDevice.cpp.obj -MF esp-idf/esp-nimble-cpp/CMakeFiles/__idf_esp-nimble-cpp.dir/src/NimBLEDevice.cpp.obj.d -o esp-idf/esp-nimble-cpp/CMakeFiles/__idf_esp-nimble-cpp.dir/src/NimBLEDevice.cpp.obj -c ../components/esp-nimble-cpp/src/NimBLEDevice.cpp
../components/esp-nimble-cpp/src/NimBLEDevice.cpp:26:14: fatal error: esp_nimble_hci.h: No such file or directory
   26 | #    include "esp_nimble_hci.h"
      |              ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

When I set the target to `esp32c3`, it compiles. Is that a problem on my end or a bug in ESP-IDF?

### What is the actual behavior?

See above.

### Steps to reproduce.

1. Checkout ESP-IDF at the 5.1 release branch
2. Unzip the `esp-nimble-cpp` in the `components` folder of (e.g.,) `examples/bluetooth/nimble/ble_l2cap_coc/coc_blecent`
3. `idf.py set-target esp32c6`
4. `idf.py build`
5. Observe the compile error.

### Build or installation Logs.

```plain
See above.
```

rahult-github, 2023-06-22:
Hi @mickeyl , i think this issue should be on esp-nimble-cpp repo side, not esp-idf.  The support should be added in that codebase ( may be correct CMakelist ).   Also, currently esp32c6 is not supported on that repo , i believe , as i see below error while trying out the steps you mentioned. 

CMake Error at ../../../../../tools/cmake/component.cmake:323 (message):
  Component esp-nimble-cpp only supports targets: esp32;esp32s3;esp32c3
Call Stack (most recent call first):
  ../../../../../tools/cmake/component.cmake:442 (__component_check_target)
  components/esp-nimble-cpp/CMakeLists.txt:27 (idf_component_register)

mickeyl, 2023-06-28:
@rahult-github Thanks for answering. I don't think it's a problem with esp-nimble-cpp. If you apply the following diff

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index e23f111..508e933 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -25,6 +25,7 @@ idf_component_register(
     "esp32"
     "esp32s3"
     "esp32c3"
+    "esp32c6"
   INCLUDE_DIRS
     "src"
   SRCS
diff --git a/src/NimBLEDevice.cpp b/src/NimBLEDevice.cpp
index eda47ae..b466b38 100644
--- a/src/NimBLEDevice.cpp
+++ b/src/NimBLEDevice.cpp
@@ -85,8 +85,8 @@ std::list <NimBLEAddress>   NimBLEDevice::m_ignoreList;
 std::vector<NimBLEAddress>  NimBLEDevice::m_whiteList;
 uint8_t                     NimBLEDevice::m_own_addr_type = BLE_OWN_ADDR_PUBLIC;
 #ifdef ESP_PLATFORM
-uint16_t                    NimBLEDevice::m_scanDuplicateSize = CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
-uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
+uint16_t                    NimBLEDevice::m_scanDuplicateSize = 0; //CONFIG_BTDM_SCAN_DUPL_CACHE_SIZE;
+uint8_t                     NimBLEDevice::m_scanFilterMode = 0; //CONFIG_BTDM_SCAN_DUPL_TYPE;
 #endif
```

to accomodate for the missing target and the dramatically reduced Bluetooth Modem options for the C6, you
will still see that it doesn't compile due to a missing header. I can "fix" this by patching ESP-IDF like that:

```
diff --git a/components/bt/CMakeLists.txt b/components/bt/CMakeLists.txt
index f04ef4191d..6afc9e7723 100644
--- a/components/bt/CMakeLists.txt
+++ b/components/bt/CMakeLists.txt
@@ -24,6 +24,7 @@ if(CONFIG_BT_ENABLED)
     elseif(CONFIG_IDF_TARGET_ESP32C6)
         list(APPEND srcs "controller/esp32c6/bt.c")
         list(APPEND include_dirs include/esp32c6/include)
+        list(APPEND include_dirs host/nimble/esp-hci/include)
 
     elseif(CONFIG_IDF_TARGET_ESP32H2)
         list(APPEND srcs "controller/esp32h2/bt.c")
```

but I'm not sure whether this is the right way. For some reason, the other targets are fine without the extra `esp-hci` include.

rahult-github, 2023-06-28:
> ```
> diff --git a/CMakeLists.txt b/CMakeLists.txt
> index e23f111..508e933 100644
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -25,6 +25,7 @@ idf_component_register(
>      "esp32"
>      "esp32s3" [...]

So, the above files are NOT part of IDF , but part of the esp-nimble-cpp repo.  Hence, my comment, that these changes should be part of that repo and can be cloned into IDF correctly. 

> to accomodate for the missing target and the dramatically reduced Bluetooth Modem options for the C6, you will still see that it doesn't compile due to a missing header. I can "fix" this by patching [...]

No, esp32c6 doesn't use esp-hci part and hence IDF CMakelists don't include this for ESP32C6 target. So, if any compilation issue is observed , then i think, the porting is not done correctly.

rahult-github, 2023-11-06:
Closing this issue. Feel free to reopen in case of any ESP-IDF related files issue.

closed 2023-11-06 by rahult-github