espressif/esp-idf #11386: ESP32-S3 ULP ADC example does not work for espidf 5.0

patrickmagee, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom PCB ESP-32-WROOM-2

### Power Supply used.

External 3.3V

### What is the expected behavior?

Wake from deep sleep as per the Readme documentation . https://github.com/espressif/esp-idf/tree/f00c78b020/examples/system/ulp_fsm/ulp_adc

### What is the actual behavior?

The ESP32-S3 goes to sleep and does not wake.

### Steps to reproduce.

Code is as per the example : https://github.com/espressif/esp-idf/tree/f00c78b020/examples/system/ulp_fsm/ulp_adc
Only the ADC channel was changed (in both the adc.S and the ulp+adc_example_main.c.
Verified that the ADC is mapped and working by using it outside of sleep.
Verified the issue can be reproduced on a ESP32-S3-DevkitC-1

### More Information.

The project codebase uses the espidf 5.0 on the platformio extension of Vscode. 
I installed the espidf extension, updated to the master branch and ran the example from there. 
I could port the entire project over to the espidf extension but it would be a lot easier if the adc ulp would work in espidf 5.0 (which platformio uses).

Should this feature work on the ESP32-S3 using espidf 5.0? 

The relevant documentation here : https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32s3/api-reference/system/ulp-risc-v.html points to the example https://github.com/espressif/esp-idf/tree/f00c78b020/examples/system/ulp_fsm/ulp_adc. But in the "supported targets" is only lists ESP32 , not any other chips in the family.

ESP-Marius, 2023-05-12:
Hi @patrickmagee,

As mentioned in the [readme](77ba84eeb75bc978e765f65e48c0d36ba9c2e150), this example is esp32 only on v5.0.

We don't normally backport features, but I assume platformio will also update to 5.1 soon as we are getting ready to release it.

patrickmagee, 2023-05-12:
Yes they probably will, though they lag by a few weeks at least. 

Does the esp32-s3 have the ability to wake from sleep using the ulp adcs I’m espidf 5.0? Is it just this example that is broken?

If the api for ulp adc can work then I’ll figure it out, but I don’t want to spend time trying to implement a feature that won’t ever work.

 If that were the case I’d just ditch platformio and port over to the espidf official extension, though that’s not trivial at this stage of a very large codebase, so I’d only want to do that if there were no other option.

patrickmagee, 2023-05-13:
@ESP-Marius just checking if you can confirm whether 5.0 can perform a wake from sleep ulp adc? Or do we have to upgrade to 5.1 / master? 

I would suggest updating the ESP32-S3 5.0 documentation referenced above to remove the example, which doesn't work with the S3.

ESP-Marius, 2023-05-15:
> just checking if you can confirm whether 5.0 can perform a wake from sleep ulp adc? Or do we have to upgrade to 5.1 / master

It is missing the changes made in 77ba84eeb75bc978e765f65e48c0d36ba9c2e150, so if you either apply these changes, orr do what `ulp_adc_init()` does manually by doing the init yourself, then yes it should still work on 5.0 as far as I can tell.

Upgrading to 5.1/master is preferable though if that is an option.

> I would suggest updating the ESP32-S3 5.0 documentation referenced above to remove the example, which doesn't work with the S3

Good point, I'll make sure this is reflected in the docs.

closed 2023-05-15 by patrickmagee