espressif/esp-idf #11387: ESP BLE MESH Sensor Server model. Publish message with empty payload.

besbar, 2023-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have coded a ESP32 node (and some sensors on it) with a BT mesh sensor server model. I rely on the sensor server model example in this repo.

Whenever I send from another node a SENSOR GET message to the ESP32, this other node actually receives a SENSOR STATUS message in response with all the sensors data concatenated (marshalled sensor data).

But if I configure the ESP32 node to publish (on a regular basis, i.e. periodic publishing interval) to a group address. There is a log error raised by the node with the sensor client model (which subscribes to the same group address).
=> BLE_MESH: Too short SDU + MIC (len 4)
The server is actually publishing a message but with empty payload. Why it is not sending the sensor data?
Did I miss something on the configuration of the model?

Thank you for help

MalakasP, 2023-11-13:
Hi @besbar,
I've come across the same issue. The solution is to define and update the model publication context. In the `sensor_server` example the publication data is defined but is left empty.

To define model publication context:
`ESP_BLE_MESH_MODEL_PUB_DEFINE(sensor_pub, 20, ROLE_NODE);`

To update the publication context either publish the model or copy the data you want to be published to the buffer:

1. Model publish: ` esp_ble_mesh_model_publish(param->model, ESP_BLE_MESH_MODEL_OP_SENSOR_STATUS, length, status, ROLE_NODE);`
2. Copying to the buffer: `memcpy(&sensor_pub, status, length);`
 
You have to remember that the publication context must be updated constantly so that the published data will be up to date. For example, I'm publishing a model when the ESP_BLE_MESH_MODEL_OP_SENSOR_GET event is emitted. Did not find a way to automatically update the publication context after for example 2 mins.

Hopefully, this helps!

besbar, 2023-11-16:
Hello @MalakasP ,

It has been absolutely helpful as I manage to solve this problem thanks to your inputs.
You are right the model publication message ([esp_ble_mesh_model_pub_t::msg]) is empty and is never updated in sensor_server example. 

When sensor server receives a model_publication_set message with periodic publishing, then ESP_BLE_MESH_MODEL_PUBLISH_UPDATE_EVT will be triggered at this periodic time (as stated in documentation [see here](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-reference/bluetooth/esp-ble-mesh.html#_CPPv426esp_ble_mesh_model_publishP20esp_ble_mesh_model_t8uint32_t8uint16_tP7uint8_t23esp_ble_mesh_dev_role_t)
This event is a esp_ble_mesh_model_cb_event_t reported by this following callback esp_ble_mesh_register_custom_model_callback().

So after defining such a cb function, each time this event is reported (according to period that is defined in the model_pub_set), I call a function that updates the sensor values and call this API you mention `esp_ble_mesh_model_publish(param->model_publish_update.model, ESP_BLE_MESH_MODEL_OP_SENSOR_STATUS, length, sensor_data, ROLE_NODE);`=> **sensor server is publishing data correctly.** 

Alternatively as you mention, we need to populate esp_ble_mesh_model_pub_t::msg
So instead of using esp_ble_mesh_model_publish(), **I used that code and it works also.**
` uint8_t trans_data[length+1];
trans_data[0] = 0x52; 
for (i = 1; i <= length; i++) {
  trans_data[i] = sensor_data[i-1];
}
memcpy(sensor_pub.msg->data, trans_data, length+1);
sensor_pub.msg->len = length+1;  `

Thank you very much !