espressif/esp-idf #11389: sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107

MicroGrit, 2023-05-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.1 

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 WROOM-1-N8R8

### Power Supply used.

USB

### What is the expected behavior?

want to write data to SD card by SDIO

### What is the actual behavior?

can not write data to SD card by SDIO, report by 
```
E (2400) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
W (2400) example_host: slave init failed, retry...
```

### Steps to reproduce.

The electrical connections have been checked.

Modified the write mode for 4-wire and 1-wire.

### Debug Logs.

```plain
I (24) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (25) boot: compile time 21:21:20
I (25) boot: chip revision: v0.1
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b89ch ( 47260) map
I (100) esp_image: segment 1: paddr=0001b8c4 vaddr=3fc91300 size=04368h ( 17256) load
I (104) esp_image: segment 2: paddr=0001fc34 vaddr=40374000 size=003e4h (   996) load
I (109) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1f938h (129336) map
I (141) esp_image: segment 4: paddr=0003f960 vaddr=403743e4 size=0ce78h ( 52856) load
I (158) boot: Loaded app from partition at offset 0x10000
I (159) boot: Disabling RNG early entropy source...
I (170) cpu_start: Pro cpu up.
I (170) cpu_start: Starting app cpu, entry point is 0x40375178
0x40375178: call_start_cpu1 at C:/Users/zhaofeng/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (185) cpu_start: Pro cpu start user code
I (185) cpu_start: cpu freq: 160000000 Hz
I (185) cpu_start: Application information:
I (188) cpu_start: Project name:     SD_SDIO
I (193) cpu_start: App version:      1
I (197) cpu_start: Compile time:     May 11 2023 21:20:45
I (203) cpu_start: ELF file SHA256:  f0b102731355c9c8...
I (209) cpu_start: ESP-IDF:          v5.0.1
I (214) cpu_start: Min chip rev:     v0.0
I (219) cpu_start: Max chip rev:     v0.99 
I (223) cpu_start: Chip rev:         v0.1
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 3FC96100 len 00053610 (333 KiB): D/IRAM
I (242) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (248) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (255) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (262) spi_flash: detected chip: gd
I (265) spi_flash: flash io: dio
I (270) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (290) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2290) example_host: host ready, start initializing slave...
I (2290) example_host: Probe using SD 4-bit...

I (2290) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2300) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2300) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2310) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2320) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2330) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2340) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2350) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2360) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2370) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (2400) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
W (2400) example_host: slave init failed, retry...
E (3420) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
W (3420) example_host: slave init failed, retry...
E (4440) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
W (4440) example_host: slave init failed, retry...
E (5460) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
W (5460) example_host: slave init failed, retry...
```

AxelLin, 2023-06-04:
@MicroGrit 
Does it work if you upgrade to 5.0.2 which includes 1ebeea7763c180bf3046d01bcabee91bbc74ce83 .
https://github.com/espressif/esp-idf/issues/11438

pacucha42, 2023-11-03:
@MicroGrit - please, is the issue still valid? If so, could you retry with IDF release/v5.1? Thank you

suryasid09, 2023-12-04:
Hello, I have the same issue. But, the board I am using is ESP32-S3-USB-OTG with IDF release v5.2-dev-3775-gb4268c874a.

pacucha42, 2023-12-06:
Ok, thanks for retesting. I will take a look and update you ASAP

igrr, 2023-12-06:
@suryasid09 over here (https://github.com/espressif/esp-idf/issues/12719) you wrote "I changed the IDF version to 5.1. It works now." Can you please clarify which versions worked and which failed?

suryasid09, 2023-12-06:
@igrr sorry for the confusion.

There are two examples given in the idf:
1. examples -> peripherals -> sdio : Here doesn't matter which version of IDF(v5.2-dev-3775-gb4268c874a or  5.1). It fails in both case.
2. examples -> storage -> sd_card -> sdmmc -> (failed in v5.2-dev-3775-gb4268c874a) but works fine in 5.1

suryasid09, 2023-12-06:
> examples -> peripherals -> sdio is not related to SD cards. It is an example to communicate with SD**IO** devices, for example ESP32's SDIO slave peripheral.
> 
> > examples -> storage -> sd_card -> sdmmc -> (failed in v5.2-dev-3775-gb4268c874a) but works fine in 5.1
> 
> What about v5.1.2? You wrote "I verified with 5.1.2, still it has the same issue." above.

 my mistake. sorry for that! I understood earlier that both example offer different way to communicate with the SD card. When I said, I verified with version 5.1.2, it didn't work, I meant using peripheral example.

igrr, 2023-12-06:
The original issue from @MicroGrit is explained by the fact that we didn't support SDMMC on S3 well on v5.0.1. The related fix was merged in v5.0.2: https://github.com/espressif/esp-idf/commit/1ebeea7763c180bf3046d01bcabee91bbc74ce83. Here is the related issue: https://github.com/espressif/esp-idf/issues/8521.

@suryasid09 The summary for your issue is that sd_card example didn't work for you in IDF v5.2-dev-3775-gb4268c874a but did work with IDF v5.1.x. We'll try to reproduce the issue and get back to you.

I suggest that we close this issue (resolution: fixed in v5.0.2, verified that it works in v5.1.x)  and continue troubleshooting your problem with IDF v5.2-dev in the other issue you have opened.

closed 2023-12-06 by igrr