espressif/esp-idf #11391: esp_lcd: add IRAM attributes (configurable) to panel_io_spi callback functions

jrahlf, 2023-05-12:
### Is your feature request related to a problem?

`lcd_spi_pre_trans_cb` and `lcd_spi_post_trans_color_cb` have no IRAM attribute. When the spi master is configured with ESP_INTR_FLAG_IRAM, a cache exception might be thrown.
So currently one cannot use ESP_INTR_FLAG_IRAM together with esp_lcd (at least not with the spi variant).

### Describe the solution you'd like.

`lcd_spi_pre_trans_cb` and `lcd_spi_post_trans_color_cb` should have a configurable IRAM attribute, so that they can run when flash access is disabled.

kriegste, 2023-05-12:
https://github.com/espressif/esp-idf/issues/10698

In the mean time I found out that if I put these functions in IRAM, all works well:
spi_lcd_prepare_cmd_buffer
spi_lcd_prepare_param_buffer
panel_io_spi_tx_param
panel_io_spi_rx_param
lcd_spi_pre_trans_cb
lcd_spi_post_trans_color_cb
(all in components\esp_lcd\src\esp_lcd_panel_io_spi.c)

Yes, this costs some IRAM... but it is not that bad.

suda-morris, 2023-05-12:
@jrahlf May I know more background of your LCD application? Do you want to display something dynamically on the screen during your flash operation? If so, then I'm afraid you will need to put a lot of SPI functions in the IRAM. But if you're OK with a static screen during flash operations, then no SPI transaction is needed during that time.

Also, if you're using esp32s3, we can have other choices to not disable your cache while still keeping your SPI working.

Alvin1Zhang, 2025-03-10:
Thanks for reporting, will cause due to inactivity, feel free to reopen with more updates.

closed 2025-03-10 by Alvin1Zhang