espressif/esp-idf #11393: Memory problem when loading Wifi

jrHernandezSio, 2023-05-12:
Hi,

I got an error when running WIFI FOTA.

**EVIRONMENT:**
ESP32-D0WD
framework-espidf @ 3.40300.0 (4.3.0)
ESP-IDF : 4.3.0

**DESCRIPTION:**

```
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4008ffd8
file: "lib/WIFI/esp32_wifi.c" line 997
func: espwf_wifi_scan_init
expression: esp_wifi_init(&cfg)
```
It seems at least 1604 bytes are needed:

```
heap_caps_malloc was called but failed to allocate 1604 bytes with 0x80C capabilities.
```

However, it seems to have plenty of memory before running the wifi FOTA:

```
-DEBUG-: [1683894729,738 s] Starting FOTA...
------------ FREE mem: Total [ 206864 ] Max block [ 65536 ] ------------
------------ FREE mem: Total [ 206864 ] Max block [ 65536 ] ------------
Heap summary for capabilities 0x00000800:
  At 0x3ff80518 len 6872 free 176 allocated 5072 min_free 144
    largest_free_block 16 alloc_blocks 33 free_blocks 2 total_blocks 35
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 8128 allocated 6896 min_free 292
    largest_free_block 4096 alloc_blocks 9 free_blocks 1 total_blocks 10
  At 0x3ffce7b8 len 71752 free 70128 allocated 0 min_free 2468
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe0440 len 129984 free 128360 allocated 0 min_free 13920
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40078000 len 32768 free 31144 allocated 0 min_free 31132
    largest_free_block 16384 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x4009ac60 len 21408 free 19784 allocated 0 min_free 19784
    largest_free_block 16384 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 257792 allocated 17560 min_free 67812 largest_free_block 65536
Heap summary for capabilities 0x00000004:
  At 0x3ff80518 len 6872 free 176 allocated 5072 min_free 144
    largest_free_block 16 alloc_blocks 33 free_blocks 2 total_blocks 35
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 8128 allocated 6896 min_free 292
    largest_free_block 4096 alloc_blocks 9 free_blocks 1 total_blocks 10
  At 0x3ffce7b8 len 71752 free 70128 allocated 0 min_free 2468
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe0440 len 129984 free 128360 allocated 0 min_free 13920
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 206864 allocated 17560 min_free 16896 largest_free_block 65536
Heap summary for capabilities 0x00000008:
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 8128 allocated 6896 min_free 292
    largest_free_block 4096 alloc_blocks 9 free_blocks 1 total_blocks 10
  At 0x3ffce7b8 len 71752 free 70128 allocated 0 min_free 2468
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe0440 len 129984 free 128360 allocated 0 min_free 13920
    largest_free_block 65536 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 206688 allocated 12488 min_free 16752 largest_free_block 65536
------------ FREE mem: Total [ 206864 ] Max block [ 65536 ] ------------

// fota_wifi process starts here ->

heap_caps_malloc was called but failed to allocate 1604 bytes with 0x80C capabilities.
```

jrHernandezSio, 2023-05-16:
Hi,

Eventually I found out this is a specificid problem of a single ESP32 unit. 

It seems a part of the RAM is damaged or sth in the NVS is damaged and it hungers the RAM (> 200KB when initialized). It is also true I am using a huge NVS partition (5MB).

Have anyone detected a similar behaviour?
Any advice?

Regards, Ramon.

Sherry616, 2023-05-17:
Hi @jrHernandezSio, there is something to make sure:
1. what kind of flash do you use in the ESP32-D0WD？The official flash should be ok for 7000 NVS operation
2. could you remove the other code and only keep the wifi fota and do a test ?
3. when the log "heap_caps_malloc was called but failed to allocate 1604 bytes with 0x80C capabilities." is printed, could you check the heap info again? add some debug log before this log
4. if the issue only occurs in a single unit,  could you do some flash unity test for that unit?

jrHernandezSio, 2023-05-19:
Hi,

1. Yes, it is the internal flash:
```
PS C:\DATA\GIT\SAYME-FW\Labs\labs_ring_lrnb> esptool.py --port COM8 flash_id    
esptool.py v3.1
Serial port COM8
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:82:3d:24
Stub is already running. No upload is necessary.
Manufacturer: c8
Device: 4018
Detected flash size: 16MB
```
3.
```
heap_caps_malloc was called but failed to allocate 1604 bytes with 0x80C capabilities.

Heap summary for capabilities 0x00000004:
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 72 allocated 5592 min_free 72
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb9a20 len 16648 free 292 allocated 14732 min_free 292
    largest_free_block 0 alloc_blocks 72 free_blocks 0 total_blocks 72
  At 0x3ffce7b8 len 71752 free 2152 allocated 67976 min_free 2152
    largest_free_block 0 alloc_blocks 537 free_blocks 0 total_blocks 537
  At 0x3ffe0440 len 129984 free 1312 allocated 127048 min_free 1312
    largest_free_block 1024 alloc_blocks 14 free_blocks 1 total_blocks 15
  Totals:
    free 3972 allocated 220452 min_free 3972 largest_free_block 1024
W (37049) wifi:malloc buffer fail
E (37059) wifi:Expected to init 10 rx buffer, actual is 4
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4008ffd8
file: "lib/WIFI/esp32_wifi.c" line 997
func: espwf_wifi_scan_init
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x4008ffdb on core 0
```

4. Can you specify how to run such test.

Sherry616, 2023-05-19:
Hi @jrHernandezSio ,

From the mac of your chip, we have checked that the chip test in ATE is passed. But for this chip, does not import memory test

We have asked module team to check the flash type, it should have some conclusion. You can run the same code in other good chip, and compare the heap informations.

using spiffs instead nvs, both will operate spi flash. but just have a try

jrHernandezSio, 2023-05-23:
I was thinking of using wear_levelling, but as far as I see it is only available with FAT partitions, right?
I guess the wear_levelling also use some amount of RAM, but I assume it should be less than the RAM used by the NVS, right?

jrHernandezSio, 2023-06-22:
Hi,

I have detected a similar problem at another unit.

```
------------ FREE mem: Total [ 53228 ] Max block [ 16384 ] ------------

Task (AppMain) STACK min: 2148 bytes. Heap poisoned (0)
------------ FREE mem: Total [ 208356 ] Max block [ 65536 ] ------------

Task (AppMain) STACK min: 2548 bytes. Heap poisoned (0)

heap_caps_malloc was called but failed to allocate 2048 bytes with 0x1800 capabilities.

Heap summary for capabilities 0x00000004:
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 84 allocated 5580 min_free 68
    largest_free_block 0 alloc_blocks 20 free_blocks 0 total_blocks 20
  At 0x3ffb9a20 len 16648 free 292 allocated 14732 min_free 292
    largest_free_block 0 alloc_blocks 72 free_blocks 0 total_blocks 72
  At 0x3ffce7c0 len 71744 free 1552 allocated 68568 min_free 1544
    largest_free_block 0 alloc_blocks 387 free_blocks 0 total_blocks 387
  At 0x3ffe0440 len 129984 free 1124 allocated 127236 min_free 964
    largest_free_block 256 alloc_blocks 164 free_blocks 4 total_blocks 168
  Totals:
    free 3196 allocated 221220 min_free 3012 largest_free_block 256

heap_caps_malloc was called but failed to allocate 2048 bytes with 0x1800 capabilities.

Heap summary for capabilities 0x00000004:
  At 0x3ff80518 len 6872 free 144 allocated 5104 min_free 144
    largest_free_block 0 alloc_blocks 35 free_blocks 0 total_blocks 35
  At 0x3ffb6388 len 7288 free 84 allocated 5580 min_free 68
    largest_free_block 0 alloc_blocks 20 free_blocks 0 total_blocks 20
  At 0x3ffb9a20 len 16648 free 292 allocated 14732 min_free 292
    largest_free_block 0 alloc_blocks 72 free_blocks 0 total_blocks 72
  At 0x3ffce7c0 len 71744 free 1552 allocated 68568 min_free 1544
    largest_free_block 0 alloc_blocks 387 free_blocks 0 total_blocks 387
  At 0x3ffe0440 len 129984 free 1124 allocated 127236 min_free 964
    largest_free_block 256 alloc_blocks 164 free_blocks 4 total_blocks 168
  Totals:
    free 3196 allocated 221220 min_free 3012 largest_free_block 256
assertion "pxR
abort() was called at PC 0x4008405b on core 0

Backtrace:0x400923fb:0x3ffbe700 0x40092b6d:0x3ffbe720 0x40098fce:0x3ffbe740 0x4008405b:0x3ffbe7b0 0x400841a5:0x3ffbe7e0 0x401abca9:0x3ffbe800 0x401a4db1:0x3ffbeac0 0x401a4b39:0x3ffbeb10 0x40091f89:0x3ffbeb40 0x4012f0a0:0x3ffbeb60 0x40083aea:0x3ffbebb0 0x401521b1:0x3ffba9b0 0x40152ee2:0x3ffba9d0 0x40151da5:0x3ffba9f0 0x4000bd83:0x3ffbaa10 0x4000117d:0x3ffbaa30 0x400592fe:0x3ffbaa50 0x4005937a:0x3ffbaa70 0x40058b69:0x3ffbaa90 0x401abc1b:0x3ffbaac0 0x401adc67:0x3ffbaaf0 0x401a4db1:0x3ffbadb0 0x401a4b39:0x3ffbae00 0x40092041:0x3ffbae30 0x401305d5:0x3ffbae70 0x400db779:0x3ffbaec0 0x400e04a2:0x3ffbaf10 0x400e02ea:0x3ffbaf80 0x400e03fb:0x3ffbafa0 0x400d549f:0x3ffbafc0 0x400d67c7:0x3ffbafe0 0x400d6c55:0x3ffbb030 0x401c734f:0x3ffbb170 0x40095ead:0x3ffbb190
  #0  0x400923fb:0x3ffbe700 in panic_abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_system/panic.c:356
  #1  0x40092b6d:0x3ffbe720 in esp_system_abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_system/system_api.c:112
  #2  0x40098fce:0x3ffbe740 in abort at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\newlib/abort.c:46
  #3  0x4008405b:0x3ffbe7b0 in lock_acquire_generic at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\newlib/locks.c:138
  #4  0x400841a5:0x3ffbe7e0 in _lock_acquire_recursive at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\newlib/locks.c:166
  #5  0x401abca9:0x3ffbe800 in _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)  
  #6  0x401a4db1:0x3ffbeac0 in fiprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48
  #7  0x401a4b39:0x3ffbeb10 in __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)   
  #8  0x40091f89:0x3ffbeb40 in xRingbufferSendFromISR at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\esp_ringbuf/ringbuf.c:1061 (discriminator 1) 
  #9  0x4012f0a0:0x3ffbeb60 in uart_rx_intr_handler_default at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\driver/uart.c:883
  #10 0x40083aea:0x3ffbebb0 in _xt_lowint1 at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\freertos\port\xtensa/xtensa_vectors.S:1105
  #11 0x401521b1:0x3ffba9b0 in uart_tx_char at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\vfs/vfs_uart.c:159
  #12 0x40152ee2:0x3ffba9d0 in uart_write at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\vfs/vfs_uart.c:217
  #13 0x40151da5:0x3ffba9f0 in esp_vfs_write at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\vfs/vfs.c:423 (discriminator 4)
  #14 0x4000bd83:0x3ffbaa10 in ?? ??:0
  #15 0x4000117d:0x3ffbaa30 in ?? ??:0
  #16 0x400592fe:0x3ffbaa50 in ?? ??:0
  #17 0x4005937a:0x3ffbaa70 in ?? ??:0
  #18 0x40058b69:0x3ffbaa90 in ?? ??:0
[... cut ...]
  #31 0x400d6c55:0x3ffbb030 in app_main at src/main.c:6971
  #32 0x401c734f:0x3ffbb170 in main_task at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\freertos\port/port_common.c:133 (discriminator 2)
  #33 0x40095ead:0x3ffbb190 in vPortTaskWrapper at C:\Users\jr.hernandez\.platformio\packages\framework-espidf@3.40300.0\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 2de13e1d2006ab8e

Rebooting...
------------ FREE mem: Total [ 208356 ] Max block [ 65536 ] ------------

```

jrHernandezSio, 2023-06-22:
This issue seems to be related to this #9030. In the past I thought it was my fault by using too much RAM, but I now I think it is also a problem with some corrupte units.

jrHernandezSio, 2023-06-22:
Here you can see how both devices starts the process with the same memory and after BLE connection ends one has enough memory to continue anf the other does not.

[image]

jrHernandezSio, 2023-06-23:
I tried to move on to idf 5.0.2, but there are too many changes. I will try later on.

I have tried idf 4.4.3 and same result.

jrHernandezSio, 2023-06-26:
Hi,

I have found out the problem disappeared when the SPIFFS parittion is reflashed, which makes me think this is affecting to the RAM memory needed by the NVS. My guess is the nvs handlers are corrupted somehow so the NVS reuieres more memory than actually needed.

I have reviewed the partitions and seems to be a slight overlap between SPIFFS and NVS.

This is my desired partition table for a 16MB ESP32:
[image]

These are the actual partitioned addresses:

[image]

If the offset is not specified then it is assigned a wrong offset.

I have tried to reproduce it by loading both partitions with data but they never get fully full due to internal limitations of my program. Both partitions remains at 10% capacity and problemis not reproduced.

adokitkat, 2024-11-05:
It looks like your problem is fixed so I am closing this issue. If you need more help please reopen this or create a new issue. Thank you.

closed 2024-11-05 by adokitkat

adokitkat, 2025-01-29:
> I removed that from my partition table and run` idf.py erase-flash` to fix it.
> 
> My chip has 8MB flash, so perhaps I overfilled it.

If you previously made changes to the partition table, it may not have been updated - for example NVS partition is not reflashed automatically by default. You doing `idf.py erase-flash` erased all of your flash memory and the partition table and other partitions were recreated from scratch, which likely fixed your problem, not just removing the efuse partition.