espressif/esp-idf #11396: event-based printf/logging for ISRs

moefear85, 2023-05-13:
### Is your feature request related to a problem?

Any attempt to printf/log from an ISR crashes the system. This is however needed for debugging.

### Describe the solution you'd like.

It would be nice if there were queue-based versions of printf/ESP_LOGx(), such that when printing/logging, the statement doesn't get sent directly to the uart, rather is passed as a message on a queue to a dedicated handler, that runs using the idle task or its own task (whose priority is then configurable in sdkconfig, otherwise it might never execute, causing confusion), to actually print/log the message.

wuyuanyi135, 2023-05-13:
would `ESP_DRAM_LOGX` serve the purpose?

ESP-Marius, 2023-05-15:
Yeah, as @wuyuanyi135, `ESP_DRAM_LOGX` will probably be sufficient for what you are trying to achieve.

`ESP_DRAM_LOGX` and `ESP_EARLY_LOGx` both map to `esp_rom_printf`, but `ESP_DRAM_LOGX` also guarantees that the string is placed in DRAM (it uses `DRAM_STR`) .

` ESP_EARLY_LOGx` as the name implies is intended mostly for IDF internal code, before all the system support is up and running. So I recommend sticking with `ESP_DRAM_LOGX`

moefear85, 2023-05-31:
actually, i no longer consider this completed, because i keep running across cases where even esp_dram_logx & esp_early_logx crash the system.

ESP-Marius, 2023-06-01:
@moefear85 are you able to provide an example that shows this issue?

moefear85, 2023-06-02:
it happens in the uart isr

KonstantinKondrashov, 2023-10-06:
I guess that it might happen because you are trying to print a string that is not in DRAM. In this case, you need to wrap it by DRAM_STR, to place it in RAM which will be guaranteed available during ISR.

```c
const static char *MSG = DRAM_STR("I am a string stored in RAM");
ESP_DRAM_LOGI(TAG, "message text, %s", MSG);
```

closed 2023-10-11 by KonstantinKondrashov