espressif/esp-idf #11398: Make std::random_device generate true random numbers

Zhu-jiatong, 2023-05-14:
### Is your feature request related to a problem?

I intended to use a real random number generator for cryptographic purposes. So, I choose std::random_device, which should be a platform independent way that is able to return real random numbers.

However, the generator produces the same value `3499211612` every time the ESP32 is powered on, which is not random.

Sketch to reproduce problem (using Arduino core):
```
#include <random>
void setup()
{
 Serial.begin(115200);
 ... // wait until Serial is ready;
 std::cout << std::random_device()();
}
```

Original post: espressif/arduino-esp32#8201

### Describe the solution you'd like.

I do know that ESP32 support a hardware random number generator for producing true random numbers. Maybe using that to implement `std::random_device`?

0xjakob, 2023-05-15:
We'll take a look if we can integrate the RNG into `std::random_device` and keep you updated about the results.

If you currently want to use the random number generator of an ESP32 chip, e.g., you can refer to the [Random number generation documentation page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/random.html). It explains the API and the conditions that need to be met to produce real random numbers.

*Please note that `std::random_device` does not have to produce real (i.e., non-deterministic) random numbers:*

> std::random_device may be implemented in terms of an implementation-defined pseudo-random number engine if a non-deterministic source (e.g. a hardware device) is not available to the implementation. In this case each [std::random_device](https://en.cppreference.com/w/cpp/numeric/random/random_device#top) object may generate the same number sequence.

(from [std::random_device cppreference page](https://en.cppreference.com/w/cpp/numeric/random/random_device#top))

If you really want to use `std::random_device` for cryptographic purposes, I strongly recommend checking the entropy via [`std::random_device::entropy`](https://en.cppreference.com/w/cpp/numeric/random/random_device/entropy) first. Otherwise, you may end up with insecure RNGs depending on the platform you use. For ESP32, `std::random_device::entropy` currently returns 0.

zikalino, 2023-05-18:
@Zhu-jiatong please let us know if esp_random() works for you.

In a meanwhile we have decided improve std::random_device. This will require adding getentropy() implementation in ESP-IDF's newlib and then configuring the toolchain to use it. This patch however will be available later in Q3.

Zhu-jiatong, 2023-05-26:
Thank you for your information. The esp_random()​ does work. I am looking forward to the addition of std::random_device.