espressif/esp-idf #11399: E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure. Possible memory leak????

kotyara12, 2023-05-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf: 4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

PlatformIO, package Espressif 32

### Power Supply used.

External 5V

### What is the expected behavior?

Should work without errors

### What is the actual behavior?

When polling a sensor operating on the 485 modbus bus, an error is periodically generated:

```
E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure.
```

This behavior is often observed, then several different registers are polled in a row and for several sensors on the bus. Although the data was successfully received.

Possible memory leak????

```
/**
 * Read status register
 * */
esp_err_t reCWTSoilS::readModbusRegister(uint8_t cmd, uint16_t reg, uint16_t* value)
{
  mb_param_request_t _request = {
    .slave_addr = _address,
    .command    = cmd,
    .reg_start  = reg,
    .reg_size   = 1
  };
  return mbc_master_send_request(&_request, (void*)value);
}

/**
 * Read humidity and temperature data
 * */
sensor_status_t reCWTSoilS::readRawData()
{
  esp_err_t err = ESP_OK;
  uint16_t bufH, bufT, bufC, bufP = 0;

  // Humidity (moisture)
  if ((_item2) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_HUMIDITY, &bufH);
  };
  // Temperature
  if ((_item1) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_TEMPERATURE, &bufT);
  };
  // Conductivity
  if ((_item3) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_CONDUCTIVITY, &bufC);
  };
  // PH
  if ((_item4) && (err == ESP_OK)) {
    err = readModbusRegister(FUNCTION_CODE_STATUS_READ, REG_STATUS_PH, &bufP);
  };

  // Check exit code
  if (err != ESP_OK) {
    rlog_e(logTAG, RSENSOR_LOG_MSG_READ_DATA_FAILED, _name, err, esp_err_to_name(err));
    return convertEspError(err);
  };

  // Store values in sensors
  return setRawValues((float)bufT/10.0, (float)bufH/10.0, (float)bufC, (float)bufP/10.0);
};
```

### Steps to reproduce.

1. Connect two or more RS485 devices to the bus
2. Query registers with mbc_master_send_request() one at a time (not several in sequence, but one at a time)
3. Enjoy occasional error messages (not always)

### Debug Logs.

```plain
14:53:17 [I] OUTD :: Values raw: 24.40 °С / 44.60 % | out: 24.40 °С / 44.60 % | min: 24.37 °С / 44.47 % | max: 24.40 °С / 44.60 %
14:53:17 [I] INT1 :: Values raw: 24.40 °С / 44.00 % | out: 24.40 °С / 43.91 % | min: 24.40 °С / 43.90 % | max: 24.40 °С / 43.91 %
14:53:17 [I] OUTL :: Values raw: 35.83 lux | out: 35.83 lux | min: 35.00 lux | max: 36.67 lux

E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure.

14:53:18 [I] SOIL1 :: Values raw: 34.0 % / 21.9 °С | out: 33.1 % / 21.9 °С | min: 33.1 % / 21.9 °С | max: 33.1 % / 21.9 °С
14:53:18 [I] MQTT :: Publish to topic "local/sensors/greenhouse/outdoor/temperature": 24.40
14:53:18 [I] MQTT :: Publish to topic "local/sensors/greenhouse/outdoor/humidity": 44.60
14:53:18 [I] MQTT :: Publish to topic "local/sensors/greenhouse/outdoor/light": 36
```

alisitsyn, 2023-05-15:
Hi @kotyara12 ,

The `E (92525) MB_PORT_COMMON: vMBMasterRunResRelease(195): Resource release failure.` message means that the stack is trying to release the communication resource while it is already released. One of the possible reason when the stack releases the resource but the task waiting for it is unblocked. This is not an error but normal situation. In the latest esp-modbus this error is changed to debug message but in the v4.4.3 it still happens.  This is most possible cause in your scenario described above and you can migrate to [esp-modbus v1.0.10](https://github.com/espressif/esp-modbus) to avoid this message. Other reason related to incorrectly configured slave response time or/and other tasks with higher priority prevent to process request from slave during allowed slave response time. As per your description it is not your case.

kotyara12, 2023-05-15:
Ok, thanks for the info, I got you. I close the request

closed 2023-05-15 by kotyara12