espressif/esp-idf #11400: ESP-IDF执行./install出错

MrYanLK, 2023-05-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-541-g885e501d99

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

1

### What is the actual behavior?

我想编译AT固件，在esp-at目录下初次执行./build.py menuconfig时会自动下载一个对应的esp-idf仓库，在esp-idf下执行./install时报错如下，网上没有找到解决方法。

### Steps to reproduce.

1

### Build or installation Logs.

```plain
ylk@DESKTOP-6TBOMGV:~/esp-at/esp-idf$
ylk@DESKTOP-6TBOMGV:~/esp-at/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32s2, esp32s3, esp32c2, esp32, esp32h2, esp32c3
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20221026 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Installing Python environment and packages
Python 3.10.6
pip 23.1.2 from /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
Upgrading pip and setuptools...
Requirement already satisfied: pip in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (23.1.2)
Requirement already satisfied: setuptools in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (67.7.2)
Skipping the download of /home/ylk/.espressif/espidf.constraints.v5.0.txt because it was downloaded recently.
Installing Python packages
 Constraint file: /home/ylk/.espressif/espidf.constraints.v5.0.txt
 Requirement files:
  - /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt
  - /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
[... 117 lines cut ...]
      ../../subprojects/dbus-gmain/meson.build:107:11: ERROR: Dependency lookup for dbus-1 with method 'pkgconfig' failed: Pkg-config binary for machine 1 not found. Giving up.

      A full log can be found at /tmp/pip-install-l60nu1iy/dbus-python_3321b57663e444edb2f42aa252e803aa/.mesonpy-kyd3itze/build/meson-logs/meson-log.txt
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Traceback (most recent call last):
  File "/home/ylk/esp-at/esp-idf/tools/idf_tools.py", line 2610, in <module>
    main(sys.argv[1:])
  File "/home/ylk/esp-at/esp-idf/tools/idf_tools.py", line 2602, in main
    action_func(args)
  File "/home/ylk/esp-at/esp-idf/tools/idf_tools.py", line 2063, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr, env=env_copy)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/ylk/.espressif/python_env/idf5.0_py3.10_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt', '-r', '/home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt', '--upgrade', '--constraint', '/home/ylk/.espressif/espidf.constraints.v5.0.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
ylk@DESKTOP-6TBOMGV:~/esp-at/esp-idf$
```

gm-jiang, 2023-05-15:
To resolve this issue, maybe you can try the following steps:
Install pkg-config: Check if pkg-config is installed on your system. You can do this by running the command pkg-config --version in a terminal. If it's not installed, you will need to install it. The process varies depending on your operating system.

* For Debian-based systems (e.g., Ubuntu): Run sudo apt-get install pkg-config to install it.
* For Red Hat-based systems (e.g., Fedora): Run sudo dnf install pkg-config to install it.
* For macOS (using Homebrew): Run brew install pkg-config to install it.

MrYanLK, 2023-05-18:
请问这个问题怎么解决呢

ylk@DESKTOP-6TBOMGV:~/esp-at/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32c2, esp32, esp32s2, esp32c3, esp32s3, esp32h2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping riscv32-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2022r1-11.2.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2022r1-11.2.0 (already installed)
Skipping riscv32-esp-elf@esp-2022r1-11.2.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20221026 (already installed)
Skipping esp-rom-elfs@20220823 (already installed)
Installing Python environment and packages
Python 3.10.6
pip 23.1.2 from /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pip (python 3.10)
Upgrading pip and setuptools...
Requirement already satisfied: pip in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (23.1.2)
Requirement already satisfied: setuptools in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (67.7.2)
Skipping the download of /home/ylk/.espressif/espidf.constraints.v5.0.txt because it was downloaded recently.
Installing Python packages
 Constraint file: /home/ylk/.espressif/espidf.constraints.v5.0.txt
 Requirement files:
  - /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt
  - /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring windows-curses: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 3)) (67.7.2)
Requirement already satisfied: click in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 4)) (8.0.4)
Requirement already satisfied: pyserial in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 5)) (3.5)
Requirement already satisfied: future in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 6)) (0.18.2)
Requirement already satisfied: cryptography in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 7)) (36.0.2)
Requirement already satisfied: pyparsing in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 8)) (3.0.9)
Requirement already satisfied: pyelftools in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 9)) (0.27)
Requirement already satisfied: idf-component-manager in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 10)) (1.2.2)
Requirement already satisfied: esp-coredump in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 11)) (1.5.0)
Requirement already satisfied: esptool in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 12)) (4.5.1)
Requirement already satisfied: kconfiglib in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 15)) (14.1.0)
Requirement already satisfied: freertos_gdb in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt (line 19)) (1.0.2)
Collecting idf-build-apps (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt (line 5))
  Using cached idf_build_apps-0.6.1-py2.py3-none-any.whl (33 kB)
Requirement already satisfied: pyyaml in /home/ylk/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt (line 9)) (6.0)
Collecting junit_xml (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt (line 10))
  Using cached https://dl.espressif.cn/pypi/junit-xml/junit_xml-1.9-py2.py3-none-any.whl (7.1 kB)
Collecting netifaces (from -r /home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt (line 11))
  Using cached netifaces-0.11.0.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
[... cut ...]
subprocess.CalledProcessError: Command '['/home/ylk/.espressif/python_env/idf5.0_py3.10_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/ylk/esp-at/esp-idf/tools/requirements/requirements.core.txt', '-r', '/home/ylk/esp-at/esp-idf/tools/requirements/requirements.ttfw.txt', '--upgrade', '--constraint', '/home/ylk/.espressif/espidf.constraints.v5.0.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 1.
ylk@DESKTOP-6TBOMGV:~/esp-at/esp-idf$

wanckl, 2023-05-22:
你的这个依赖解决了吗
```
dbus-gmain| Run-time dependency dbus-1 found: NO (tried pkgconfig and cmake)
../../subprojects/dbus-gmain/meson.build:107:11: ERROR: Dependency "dbus-1" not found, tried pkgconfig and cmake
```

dobairoland, 2024-12-17:
I'm closing this one and half year old report. 

I think the issue was caused by trying to install the ttfw dependencies which is our old test infrastructure package and is deprecated now. I don't think esp-at would need it.

The issue keeps repeating because `./install.sh --enable-ttfw` was executed once. The issue can be resolved by removing the ttfw keyword by manually editing the `~/.espressif/idf-env.json` file. The file can be in other locations if it was changed,

closed 2024-12-17 by dobairoland