espressif/esp-idf #11401: hfp_ag can't connect hfp_hf

jjj19950710, 2023-05-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

I expect hfp_ ag can connect to hfp_ hf.

### What is the actual behavior?

When I was at Running the con command on hfp_ag, unable to connect hfp_hf.

### Steps to reproduce.

1. Get the HFP AG example from ESP IDF 5.0.1. && cd examples/bluetooth/bluedroid/classic_bt/hfp_ag
2. I use the [issius 11264](https://github.com/espressif/esp-idf/issues/11264) to solve the problem of HF AG not initialized
3. Enter the menuconfig and change the following:
Component config -> Bluetooth -> Bluedroid Options -> audio (SCO) data path -> HCI
Component config -> Bluetooth -> Controller Options -> BR/EDR Sync (SCO/eSCO) default data path -> HCI
4. Change the MAC address of the headset you want to connect to in app_hf_msg_set.c:18 (hfp_hf Mac address)
5. Build and run the example.
6. Try to connect to your headset using con command.

### Debug Logs.

```plain
hfp_ag log:

hfp_ag> con
Connect.
I (758604) BT_RFCOMM: RFCOMM_RemoveServer() handle:8
I (758604) BT_RFCOMM: RFCOMM_RemoveServer() handle:1
                                                                                                                       W (758914) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
I (759094) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: 94-b5-55-fc-21-6e
I (759094) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:256, p_mcb:0x0
I (759104) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:256, p_mcb:0x0, p_port:0x3ffc6588
I (759114) BT_RFCOMM: PORT_SetEventMask() handle:2 mask:0x1
I (759124) BT_RFCOMM: PORT_SetEventCallback() handle:2
hfp_ag> W (759164) BT_HCI: hci cmd send: disconnect: hdl 0x81, rsn:0x13
W (759164) BT_RFCOMM: PORT_StartCnf failed result:5
I (759164) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: ff-ff-ff-ff-ff-ff
I (759174) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:1 mtu:256, p_mcb:0x0
I (759174) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:1 mtu:256, p_mcb:0x0, p_port:0x3ffc662c
I (759194) BT_RFCOMM: PORT_SetEventMask() handle:3 mask:0x1
I (759204) BT_RFCOMM: PORT_SetEventCallback() handle:3
I (759204) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: ff-ff-ff-ff-ff-ff
I (759214) BT_RFCOMM: RFCOMM_CreateConnection(): scn:2, dlci:4, is_server:1 mtu:256, p_mcb:0x0
I (759214) BT_RFCOMM: RFCOMM_CreateConnection(): scn:2, dlci:4, is_server:1 mtu:256, p_mcb:0x0, p_port:0x3ffc66d0
I (759234) BT_RFCOMM: PORT_SetEventMask() handle:4 mask:0x1
I (759244) BT_RFCOMM: PORT_SetEventCallback() handle:4
W (759244) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x16
I (759244) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (759254) BT_APP_HF: --connection state DISCONNECTED, peer feats 0x0, chld_feats 0x0
```
hfp_hf log:
```
hfp_hf>
hfp_hf> W (1044736) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (1044746) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (1044746) BT_HF: event: 16
W (1045046) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x13
```
```

xiongweichao, 2023-05-15:
@jjj19950710  Sorry, I didn't reproduce this issue. Please try [v5.0.2](https://github.com/espressif/esp-idf/tree/v5.0.2) to see if this problem exists.

jjj19950710, 2023-05-15:
I use v5.0.2，Enter the menuconfig and change the following:
Component config -> Bluetooth -> Bluedroid Options -> audio (SCO) data path -> HCI
Component config -> Bluetooth -> Controller Options -> BR/EDR Sync (SCO/eSCO) default data path -> HCI
but it not useful.
hfp_hf run log:
```
entry 0x4008064c
I (27) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (27) boot: compile time 18:38:46
I (28) boot: chip revision: v3.0
I (31) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1f2c8h (127688) map
I (142) esp_image: segment 1: paddr=0002f2f0 vaddr=3ffbdb60 size=00d28h (  3368) load
I (144) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7ce64h (511588) map
I (333) esp_image: segment 3: paddr=000ace8c vaddr=3ffbe888 size=036e0h ( 14048) load
I (338) esp_image: segment 4: paddr=000b0574 vaddr=40080000 size=18984h (100740) load
I (392) boot: Loaded app from partition at offset 0x10000
I (392) boot: Disabling RNG early entropy source...
I (404) cpu_start: Pro cpu up.
I (404) cpu_start: Starting app cpu, entry point is 0x400813c8
0x400813c8: call_start_cpu1 at D:/esp/Espressif5.0.2/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (421) cpu_start: Pro cpu start user code
I (421) cpu_start: cpu freq: 160000000 Hz
I (421) cpu_start: Application information:
I (425) cpu_start: Project name:     hfp_hf
I (430) cpu_start: App version:      v5.0.2-dirty
I (436) cpu_start: Compile time:     May 15 2023 18:44:58
I (442) cpu_start: ELF file SHA256:  3b1b9ba9847a4250...
I (448) cpu_start: ESP-IDF:          v5.0.2-dirty
I (453) cpu_start: Min chip rev:     v0.0
I (458) cpu_start: Max chip rev:     v3.99
I (463) cpu_start: Chip rev:         v3.0
I (468) heap_init: Initializing. RAM available for dynamic allocation:
I (475) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (481) heap_init: At 3FFB6BF8 len 00001408 (5 KiB): DRAM
I (487) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (493) heap_init: At 3FFC8BE0 len 00017420 (93 KiB): DRAM
I (499) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (505) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (512) heap_init: At 40098984 len 0000767C (29 KiB): IRAM
I (520) spi_flash: detected chip: generic
I (523) spi_flash: flash io: dio
W (527) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (541) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (587) BTDM_INIT: BT controller compile version [2c56073]
I (587) system_api: Base MAC address is not set
I (587) system_api: read default base MAC address from EFUSE
I (597) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1307) BT_HF: Starting device discovery...
I (1327) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: ff-ff-ff-ff-ff-ff
I (1327) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:1 mtu:256, p_mcb:0x0
I (1327) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:1 mtu:256, p_mcb:0x0, p_port:0x3ffc629c
I (1337) BT_RFCOMM: PORT_SetEventMask() handle:2 mask:0x1
I (1347) BT_RFCOMM: PORT_SetEventCallback() handle:2
I (1357) BT_HF: event: 10
I (1357) BT_HF: event: 10
I (1367) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1367) BT_HF: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[... 134 lines cut ...]
 |  2. Setup a service level connection           |
 |  3. Run hfp_ag to test                         |
 |                                                |
 =================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
hfp_ag> W (51215) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (51215) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (51485) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x13

Unrecognized command
hfp_ag> W (84175) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (84175) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
W (84445) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13

hfp_ag>
```

xiongweichao, 2023-05-15:
Hi @jjj19950710 ,

Please try the following command to erase flash chip.
```
idf.py erase-flash -p port
```

Thanks

jjj19950710, 2023-05-16:
The problem has been resolved, and it is indeed caused by not erasing the chip,thanks!

Alvin1Zhang, 2023-05-17:
Good to know the issue is resolved, and thanks for sharing the updates.

closed 2023-05-17 by Alvin1Zhang