espressif/esp-idf #11403: ESP32-S3 freezes randomly

QuentinFarizon, 2023-05-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I expect the device to never freeze

### What is the actual behavior?

We have deployed our code on hundreds of customers, and for some of them the device sometimes freezes. The applicative code is basically a task with a blinking led, and doing http calls every minute. When frozen, there is no more communication with the server, and the LED is also frozen (with or without light), which indicates to me that the system is in a frozen state (rather than shutdown, or just somme taks being killed).

What we tried so far :
- Configuring interrupt and task watchdog
- rebooting on crashes/panic (CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT)
- custom watchdog, implemented as a ISR dispatched timer (see below)
- daily reboot (device reboots itself after 24h of uptime)
- reboot on low RAM from main loop
 
 We are out of ideas on what can cause the device to freeze, with all watchdogs activated.
 We do not have a scenario to reproduce, and never reproduce at our office.

Custom watchdog:
```
static void applicative_watchdog_handler()
{
    ESP_LOGI(TAG, "Handler applicative watchdog");
    esp_reset_reason_set_hint(ESP_RST_PANIC);
    software_reset();
}

void applicative_watchdog_init(void)
{
    esp_timer_create_args_t create_args = {
        .callback = applicative_watchdog_handler,
        .arg = NULL,
        .name = "applicative_watchdog",
        .dispatch_method = ESP_TIMER_ISR
    };
    esp_timer_create(&create_args, &s_applicative_watchdog_timer);
    esp_timer_start_once(s_applicative_watchdog_timer, TEN_MINUTES_IN_US);
    ESP_LOGI(TAG, "Started applicative watchdog");
}

void applicative_watchdog_feed(void)
{
    ESP_LOGD(TAG, "Feeding applicative watchdog");
    esp_timer_restart(s_applicative_watchdog_timer, TWO_MINUTES_IN_US);
}
```

### Steps to reproduce.

Device is simply polling http calls

CC-MASTER05, 2023-05-16:
Hello,
Your product is rebooting there are many case and check those points.

- Less current they get from power supply
- Very Fast communication doing
- flash size is may less for communication and store data
- check interrupt characteristics 

Every 24 hours rebooting will not help you to solve this issue so i don't think you need this logic you can try alternative path to solve this issue if you any debug log then share if we found some bug and anyway to help you.

Thanks,
Fenil

QuentinFarizon, 2023-05-16:
Hello, thanks for your input !

> Less current they get from power supply  

It's plugged on a AC-USB adapter so I doubt there can be any issues here

> Very Fast communication with your server happening  

I make a call every few tens of seconds, with very limited data

> check interrupt characteristics

Sorry, I'm not sure what you mean

> Monitor system temperatures and make sure they are within acceptable limits.

They are

> Check your code for any memory leaks or excessive memory usage

I have checked, no leak, and anyway I reboot in case of low memory before hitting any potential issue

CC-MASTER05, 2023-05-17:
Hi,
If your system used interrupt then you need to check and confirm one time when your interrupt is occur at a time if you changing any ISR variable value they make you in rebooting case so if your have interrupt based system then make interrupt secure.

Thanks

eiffel31, 2023-05-17:
Your description is not clear on a point: do you have some clients where it is working always fine and some where there is repeated problems?

If so, then I suggest you to investigate based on hardware exchange:
- swap the ESP32 unit
- swap the USB PSU

Other environment issues I can think of:
- WiFi router problems (collect data of hardware WiFi setup for example) 
- main power disturbances
- strong electromagnetic fields

Hope this helps

zikalino, 2023-05-18:
@QuentinFarizon I think we may need some more information to figure out what's going on here.
@eiffel31 has some valid questions, especially regarding "swapping the device"

I wonder if the customers usually have one or more devices, and if they have more devices, do they usually have problems with one of them or more?

Also do you have any estimate (%) on how many customers are experiencing problems?

Any additional information on particular customer configuration / environment would be very useful.

closed 2023-06-26 by QuentinFarizon