espressif/esp-idf #11406: esp32-c6 WiFi 6 TWT support

roblatour, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to  https://www.espressif.com/en/news/ESP32_C6

"One more important feature of the 802.11ax standard is the Target Wake Time (TWT), which allows the devices to sleep for an extended time-period, with assistance from an 802.11ax-capable, Wi-Fi access point. This feature in ESP32-C6 will allow our customers to build battery-operated, connected devices that can last for years on battery, while staying connected throughout."

and according to https://github.com/espressif/esp-idf/issues/10423
   TWT
    **Individual TWT – Supported**
    Broadcast TWT – Not supported (WIFI-5215)

However I can't find any documentation or examples on how to implement this using ESP-IDF.

ESP-Marius, 2023-05-16:
I think [this example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt) might be what you are looking for.

roblatour, 2023-05-16:
- but the code doesn't link ...

'''
Bootloader binary size 0x6820 bytes. 0x7e0 bytes (7%) free.
[2/4] Linking CXX executable itwt.elf
FAILED: itwt.elf 
cmd.exe /C "cd . && C:\esp-r-5-1\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=E:/Documents/esp32/itwt/build/itwt.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\itwt.elf.rsp -o itwt.elf  && cd ."
c:/esp-r-5-1/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(itwt.c.obj):(.literal.got_ip_handler+0x10): undefined reference to `esp_wifi_sta_itwt_setup'
c:/esp-r-5-1/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(itwt.c.obj): in function `got_ip_handler':
E:/Documents/esp32/itwt/main/itwt.c:121: undefined reference to `esp_wifi_sta_itwt_setup'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 
 * '''

closed 2023-05-16 by roblatour

lalves-isr, 2024-08-18:
> I think [this example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt) might be what you are looking for.

Trying to compile this example without success... What could be missing? Using ESP32C6 target.
"/itwt_main.c:187:17: error: 'wifi_event_sta_itwt_teardown_t' has no member named 'status'"