espressif/esp-idf #11407: ULP-RISC-V ADC Example "deep sleep" current consumption is > 2mA on ESP32-S3

patrickmagee, 2023-05-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-503-g17451f1fb3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Hardware

### Power Supply used.

External 3.3V

### What is the expected behavior?

Expected the processor to enter deep sleep with current consumption in the low 100's of uA. The lower the better but even 200-300uA would be acceptable in this application.

### What is the actual behavior?

The example operates as expected, i.e.  it goes to "deep sleep" and wakes when the ULP ADC measurement is above the threshold.
 However the current consumption in "deep sleep" remains 2000-2100uA.

### Steps to reproduce.

1. Create project from example https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_riscv/adc
2. Adjust the ADC channel to ADC_CHANNEL_8
3. Flash and Run
4. Observe current - >2mA
5. To check that the hardware (custom PCB) is not leaking current modify the code to wake from timer rather than ULP. This gives a deep sleep current of ~40uA (most of this will be from peripheral devices, quiescent regulator current etc).

```
void app_main(void)
    {
        esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

        /* not a wakeup from ULP, load the firmware */
        if ((cause != ESP_SLEEP_WAKEUP_ULP) && (cause != ESP_SLEEP_WAKEUP_TIMER))
        {
            printf("Not a ULP-RISC-V wakeup (cause = %d), initializing it! \n", cause);
            //      init_ulp_program();
        }

        /* ULP Risc-V read and detected a temperature above the limit */
        if (cause == ESP_SLEEP_WAKEUP_ULP)
        {
            printf("ULP-RISC-V woke up the main CPU\n");
            printf("Threshold: high = %" PRIu32 "\n", ulp_adc_threshold);
            printf("Value = %" PRIu32 " was above threshold\n", ulp_wakeup_result);
        }

        /* Go back to sleep, only the ULP Risc-V will run */
        printf("Entering in deep sleep\n\n");
        vTaskDelay(100 / portTICK_PERIOD_MS);

        /* RTC peripheral power domain needs to be kept on to keep SAR ADC related configs during sleep */
        //      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

        //      ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());

        esp_sleep_enable_timer_wakeup(1000 * 1000);

        esp_deep_sleep_start();
    }

6. Repeat the experiment with the similar example provided using fsm, noted the same behaviour. (https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_fsm/ulp_adc). Similar error observed with high current.
```

### Debug Logs.

```plain
Logs show normal operation, current consumption anomalies confirmed with 3x uA meters and 3x hardware.
```

### More Information.

While no current consumption figures are given for the examples, there are documented projects showing sub 20uA figures for similar code on ESP32 https://www.youtube.com/watch?v=n2uHENhxFZI. Even accounting for some variation between device family and peripheral current draw, I would expect to be in the low 100's of uA with this code. >2mA wont work for the application.

ESP-Marius, 2023-05-17:
Seems like a duplicate of https://github.com/espressif/esp-idf/issues/10595, we have a fix up for review internally.

patrickmagee, 2023-05-17:
Hi Marius, yes it does look like the same issue. I applied the patch to the master and I'm now getting "deep sleep" of 800uA.  I'll do more testing with the patch applied. I'll mark this issue as closed and add my findings to https://github.com/espressif/esp-idf/issues/10595 so as not to duplicate. Thanks.

closed 2023-05-17 by patrickmagee